espressif/arduino-esp32 #3656: HTTPClient - Chunked-Encoding

Manawyrm, 2020-01-22:
Hi,

### Description:
I'm using the HTTPClient library to download some data from a script.
When no Chunked-Encoding is used, everything works fine.
When the server uses HTTP Chunked-Encoding to send data, the resulting data in the buffer is corrupt.
This is because parts of the Chunked-Encoding itself are present in the output data.

Screenshot:
[image]
(1aca is the length for the Chunked-Encoding, which is present in the middle of this test output and shouldn't be)

### Sketch:  
https://github.com/espressif/arduino-esp32/blob/master/libraries/HTTPClient/examples/StreamHttpClient/StreamHttpClient.ino

Replace the URL with http://otaunsafe.tbspace.de/epaper/bugtest.php

On the serial port (115200 baud) you can now observe the corrupted data. 
The expected result are exactly 15000 bytes, going from 0x00 to 0xFF smoothly. 
The 1aca-hex length shouldn't be present.

### Additional info
Tested with newest master, built on Linux with PlatformIO (but I don't think this makes any difference here :smile: ) 

Thanks,
Tobias

a-c-sreedhar-reddy, 2020-01-23:
How can get the exact result other than browser to compare?

Manawyrm, 2020-01-23:
You could use `wget "http://otaunsafe.tbspace.de/epaper/bugtest.php"` on the command line.

Not sure about Windows, it could work like this (PowerShell):
``` 
new-object System.Net.WebClient).DownloadFile('http://otaunsafe.tbspace.de/epaper/bugtest.php','C:\bugtest.php')
```

Just use something that doesn't do HSTS preloading (a download manager like JDownloader or Aria is probably also fine).

closed 2020-04-06 by stale[bot]

Manawyrm, 2020-04-06:
This bug is unfortunately still present.