espressif/arduino-esp32 #3657: Potential deadlock in BluetoothSerial

jcdevel, 2020-01-23:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			current git master
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        Ubuntu 18.04 LTS

### Description:
I am sending timestamps every 100 ms from my ESP32 node via BluetoothSerial (do not ask why, just taking baby steps with Bluetooth and ESP32). I am using an Android Bluetooth Terminal app to print the timestamps on my phone. I have noticed that if I close the connection from the phone and open it again, after a couple times the ESP32 node freezes. A hard reset is needed to bring it back to life.

I have performed some analysis and debugging and I have traced the root cause to the blocking calls in the transmission task of the BluetoothSerial component. I have prepared a couple of patches that fix the issue. I will do a pull-request for revision.

### Sketch:
```cpp

#include <stdlib.h>
#include <string.h>
#include <BluetoothSerial.h>

BluetoothSerial SerialBT;

void setup() {
  Serial.begin(115200);
  SerialBT.begin("ESP32test"); //Bluetooth device name
  Serial.println("Start!");
}

void loop() {
  static unsigned long prev;
  unsigned long now;
  char out_buf[32];
  unsigned int len;

  now = millis();
  if ((now - prev) >= 100) {
    //periodic message
    itoa(now, out_buf, 10);
    len = strlen(out_buf);
    out_buf[len] = '\n';
    out_buf[len+1] = '0';
    SerialBT.write((uint8_t *)out_buf, len+1);
    prev = now;
  }
}

```

emmabeebe, 2020-01-27:
I am also having this issue. I am using serialBT to send data between two esp32 devices and when I manually pull the power to one of the devices the remaining device just hangs and freezes.

jcdevel, 2020-01-30:
@emmabeebe Could you verify if my PR fixes the issue on your side? Thanks!

Benik3, 2020-01-31:
I tested it with my GPS Serial to BT, where I encountered this problem pretty often and it looks, that it's fixed :)

jcdevel, 2020-04-14:
Could somebody review this issue? (PR #3658)

emmabeebe, 2020-04-22:
I have also reviewed it and the issue has been fixed.

max5555, 2020-05-27:
Make it faster

```
#include <Arduino.h>
#include "BluetoothSerial.h"

BluetoothSerial SerialBT;

void setup()
{
  Serial.begin(115200);
  SerialBT.begin("GPS"); 
}

void loop()
{
  delay(1);
  Serial.println(1);
  SerialBT.write('2');
  Serial.println(3);
}
```

And the sketch occasionally hangs on BT disconnect and continues on reconnect only.

closed 2020-08-09 by stale[bot]