espressif/arduino-esp32 #3660: Parallel use of WiFiClientSecure and WiFiClient. Non virtual methods in WiFiClient

maxanier, 2020-01-23:
### Description
In a project of mine, I want to perform the same HTTP request to either an online host via https or an local host (raspberry pi with WiFi AP) via http (switch between hosts depending on scenario).
I use `WiFiClient` (for http) resp. `WiFiClientSecure` (for https), since I would like to have a stream, so ArduinoJSON can directly write to it.
Because the requests are the same, I would like to use the same methods to prepare, send and process the requests for both http/https.

As `WiFiClientSecure` extends `WiFiClient` my idea is to create the `WiFiClient(Secure)` instance depending on the host, but then use type/cast it as `WiFiClient` for the common methods.

Unfortunately, with this setup does not work and results in strange crashes.

*This does not work*
```
WiFiClientSecure secureClient;
WiFiClient client = secureClient;
client.connect(...);
client.println(...); //send header
client.readBytesUntil(...); //crash here
```
There is no problem, if I just use the `secureClient` directly, or if I use a `WiFiClient` (for http).
I can provide the stacktrace later, but it is some buffer error in `WiFiClient#read`

### Assumed cause
I'm not an C++ expert, but from my investigation, I think it is caused by the following:
`WiFiClientSecure` overrides some methods (especially the read ones `read`) of `WiFiClient`, but because they are not declared `virtual` in WiFiClient they are not actually overridden, but merely hidden. This is fine as long as the secure client is accessed via a `WiFiClientSecure` reference. However, if accessing it via a `WiFiClient` reference it is does not properly use the new methods.
But maybe I am missing something.

Over at the ESP8266 libraries, most of the `WiFiClient` methods are `virtual` (https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/WiFiClient.h) and overridden in `WiFiClientSecure` (https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/WiFiClientSecureBearSSL.h). I did not try things on the ESP8266 though.

### Suggested solution
Is there a reason those methods are not `virtual` here? If not, could you make them (or I could make a PR and test if things work).

Or is there a better way to handle both HTTP and HTTPS?
(Yes I could always use a self signed certificate and an reverse proxy on the local network raspberry pi to exclusively use https)

eljarete, 2021-03-03:
Same situation here. We need to connect to different services as clients with different certificates, but if we run two or more WiFiClientSecure it crash when connecting second instance.

igolubic, 2021-03-06:
The same issue that we are trying to solve here: https://github.com/iotaap/iotaap-os/pull/13

We need WiFiClientSecure for two different things, but if used at the same time the system crashes. Even with the current implementation (stopping WiFiClientSecure between two usages) it causes a lot of issues.

me-no-dev, 2021-03-11:
Could you please provide some example code that I can use to reproduce the issue?

GiambaFr, 2021-03-13:
The same issue.

WifiClientSecure mqttSecure;
WificlientSecure netSecure;

When mqttSecure is connected and kept alive, send a post with httpclient and netSecure http.Post() return -1.

Like if mqttSecure and netSecure are the same object.

me-no-dev, 2021-03-17:
@GiambaFr I can run two secure clients together with current master. You mention http client but I see no example code. Please provide one :)

@maxanier `virtual` did not change anything. Something like this works:
```cpp
WifiClientSecure secClient;

void sendRequest(WiFiClient &client){
 client.println("GET / HTTP/1.0");
}

void someFn(){
  sendRequest(secClient);
}
```

maxanier, 2021-03-17:
Alright, thank you.
I can't remember if I actually tried to test the `virtual` annotation.

Luckily I know the connection type at compile time so as a workaround I am just using something like `#define WIFI_CLIENT WifiClientSecure`, but your suggested is better of course.
Although did you test the read/receive functionality as well? For me it only crashed during read
```
WiFiClientSecure secureClient;
WiFiClient client = secureClient;
client.connect(...);
client.println(...); //send header
client.readBytesUntil(...); //crash here <---
```

me-no-dev, 2021-03-17:
@maxanier your code can not work as-is, because you are not casting `WiFiClientSecure` to `WiFiClient`, but instead you are ending here: https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiClient.cpp#L188

Here is my test sketch:
```cpp
#include <WiFiClientSecure.h>

WiFiClientSecure client;
WiFiClientSecure client2;
std::unique_ptr<WiFiClient>client3 = std::unique_ptr<WiFiClient>(&client2);

bool clientStart(){
  Serial.println("client: connecting to server...");
  client.setInsecure();//skip verification
  if (client.connect("www.howsmyssl.com", 443)) {
    Serial.println("client: connected to server!");
    return true;
  }
  Serial.println("client: connection failed!");
  return false;
}

void clientMid(){
  Serial.println("client: sending request...");
  client.println("GET https://www.howsmyssl.com/a/check HTTP/1.0");
  client.println("Host: www.howsmyssl.com");
  client.println("Connection: close");
  client.println();
}

void clientEnd(){
  Serial.println("client: receiving headers...");
  while (client.connected()) {
    String line = client.readStringUntil('\n');
    if (line == "\r") {
      Serial.println("client: headers received");
      break;
    }
  }
  Serial.println("client: reading response...");
  while (client.available()) {
    Serial.write(client.read());
  }
  Serial.println();
}

bool client2Start(){
  Serial.println("client2: connecting to server...");
  client2.setInsecure();//skip verification
  if (client2.connect("www.google.com", 443)) {
    Serial.println("client2: connected to server!");
    return true;
  }
  Serial.println("client2: connection failed!");
  return false;
}

void client2Mid(WiFiClient &c){
  Serial.println("client2: sending request...");
  c.println("GET https://www.google.com/a/check HTTP/1.0");
  c.println("Host: www.google.com");
  c.println("Connection: close");
  c.println();
}

void client2End(WiFiClient &c){
  Serial.println("client2: receiving headers...");
  while (c.connected()) {
    String line = c.readStringUntil('\n');
    if (line == "\r") {
      Serial.println("client2: headers received");
      break;
    }
  }
  delay(100);//google responds with content in another package
  Serial.println("client2: reading response...");
  while (c.available()) {
    Serial.write(c.read());
  }
  Serial.println();
}

void setup() {
  Serial.begin(115200);
  delay(100);

  Serial.print("Connecting...");
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(100);
  }
  Serial.println("Connected");

  bool c1 = clientStart();
  if(!c1){
    return;
  }
  bool c2 = client2Start();
  clientMid();
  if(!c2){
    clientEnd();
    Serial.println("client: closing...");
    client.stop();
    return;
  }
  client2Mid(*client3);
  clientEnd();
  client2End(client2);
  Serial.println("client: closing...");
  client.stop();
  Serial.println("client2: closing...");
  client2.stop();
}

void loop() {
  delay(1000);
}
```

maxanier, 2021-03-17:
Oh, alright, I see, I did not know that. Thank you for providing your sketch! I should be able to create a better solution based on that.
I will close this issue then. I think the other problems mentioned in this thread are different as they deal with two concurrent clients apparently. Feel free to reopen, if you want to continue it in this issue

closed 2021-03-17 by maxanier

me-no-dev, 2021-03-17:
@maxanier sketch shows that I also use two clients at the same time, so I hope it will help the other users as well :)

snakeninny, 2023-01-29:
> I think the other problems mentioned in this thread are different as they deal with two concurrent clients apparently. Feel free to reopen, if you want to continue it in this issue

@maxanier I have a similar problem. In my code I have `WiFiClient espClient` and `WiFiClientSecure updateClient`, and they have irrelevant usage.
When `updateClient` is used by `ESPhttpUpdate.update`, `espClient` is disconnected in `ESPhttpUpdate.onProgress`, as you can test in the sketch:

```
#include "Arduino.h"
#include <ESP8266WiFi.h>
#include <ESP8266httpUpdate.h>
#include <PubSubClient.h>

#define ssid "ssid" // Change this
#define password "password" // Change this

String upUrl = "https://bin.bemfa.com/b/3BcN2Q1NGY4NWFmNDI5NzZlZTNjMjY5M2U2OTJhNmJiNTk=light002.bin";
char serverName[] = "baidu.com";

WiFiClient espClient;

void onStart()
{
  Serial.println("onStart");
  Serial.printf("WiFiClient is %sconnected.\r\n", espClient.connected() ? "" : "NOT ");
}

void onEnd()
{
  Serial.println("onEnd");
}

void onProgress(int cur, int total)
{
  Serial.printf("onProgress at %d of %d bytes.\r\n", cur, total);
  Serial.printf("WiFiClient is %sconnected.\r\n", espClient.connected() ? "" : "NOT ");
}

void onError(int err)
{
  Serial.printf("onError, err code = %d.\r\n", err);
}

void updateBin()
{
  WiFiClientSecure updateClient;
  updateClient.setInsecure();

  ESPhttpUpdate.onStart(onStart);
  ESPhttpUpdate.onEnd(onEnd);
  ESPhttpUpdate.onProgress(onProgress);
  ESPhttpUpdate.onError(onError);

  t_httpUpdate_return ret = ESPhttpUpdate.update(updateClient, upUrl);
  switch (ret)
  {
  case HTTP_UPDATE_FAILED:
    Serial.println("HTTP_UPDATE_FAILED");
    break;
  case HTTP_UPDATE_NO_UPDATES:
    Serial.println("HTTP_UPDATE_NO_UPDATES");
    break;
  case HTTP_UPDATE_OK:
    Serial.println("HTTP_UPDATE_OK");
    break;
  }
}

void setup()
{
  Serial.begin(115200);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");

  if (espClient.connect(serverName, 443))
  {
    if (espClient.connected())
    {
      Serial.println("Connected to https://baidu.com.");
    }
    updateBin();
  }
}

void loop()
{
}
```

```
...........
Connected to https://baidu.com.
onStart
WiFiClient is connected.
onProgress at 0 of 305120 bytes.
WiFiClient is NOT connected.
onProgress at 0 of 305120 bytes.
WiFiClient is NOT connected.
onProgress at 4096 of 305120 bytes.
WiFiClient is NOT connected.
...
```

Did I miss anything or this is normal?
How do I keep `espClient` connected during `ESPhttpUpdate.update`?

gsicilia82, 2024-03-10:
I was looking for a ready-to-use code snippet and could not find any "complete" solution. However, this was the first google hit, therefore, I will show my OOP solution here, maybe it helps other users:

```c++

#include <Arduino.h>
#include <HTTPClient.h>
#include <WiFiClientSecure.h>
#include <WiFi.h>

const char* test_root_ca= \
  "-----BEGIN CERTIFICATE-----\n" \
  "MIIFazCCA1OgAwIBAgIRAIIQz7DSQONZRGPgu2OCiwAwDQYJKoZIhvcNAQELBQAw\n" \
  "TzELMAkGA1UEBhMCVVMxKTAnBgNVBAoTIEludGVybmV0IFNlY3VyaXR5IFJlc2Vh\n" \
  "cmNoIEdyb3VwMRUwEwYDVQQDEwxJU1JHIFJvb3QgWDEwHhcNMTUwNjA0MTEwNDM4\n" \
  "WhcNMzUwNjA0MTEwNDM4WjBPMQswCQYDVQQGEwJVUzEpMCcGA1UEChMgSW50ZXJu\n" \
  "ZXQgU2VjdXJpdHkgUmVzZWFyY2ggR3JvdXAxFTATBgNVBAMTDElTUkcgUm9vdCBY\n" \
  "MTCCAiIwDQYJKoZIhvcNAQEBBQADggIPADCCAgoCggIBAK3oJHP0FDfzm54rVygc\n" \
  "h77ct984kIxuPOZXoHj3dcKi/vVqbvYATyjb3miGbESTtrFj/RQSa78f0uoxmyF+\n" \
  "0TM8ukj13Xnfs7j/EvEhmkvBioZxaUpmZmyPfjxwv60pIgbz5MDmgK7iS4+3mX6U\n" \
  "A5/TR5d8mUgjU+g4rk8Kb4Mu0UlXjIB0ttov0DiNewNwIRt18jA8+o+u3dpjq+sW\n" \
  "T8KOEUt+zwvo/7V3LvSye0rgTBIlDHCNAymg4VMk7BPZ7hm/ELNKjD+Jo2FR3qyH\n" \
  "B5T0Y3HsLuJvW5iB4YlcNHlsdu87kGJ55tukmi8mxdAQ4Q7e2RCOFvu396j3x+UC\n" \
  "B5iPNgiV5+I3lg02dZ77DnKxHZu8A/lJBdiB3QW0KtZB6awBdpUKD9jf1b0SHzUv\n" \
  "KBds0pjBqAlkd25HN7rOrFleaJ1/ctaJxQZBKT5ZPt0m9STJEadao0xAH0ahmbWn\n" \
  "OlFuhjuefXKnEgV4We0+UXgVCwOPjdAvBbI+e0ocS3MFEvzG6uBQE3xDk3SzynTn\n" \
  "jh8BCNAw1FtxNrQHusEwMFxIt4I7mKZ9YIqioymCzLq9gwQbooMDQaHWBfEbwrbw\n" \
  "qHyGO0aoSCqI3Haadr8faqU9GY/rOPNk3sgrDQoo//fb4hVC1CLQJ13hef4Y53CI\n" \
  "rU7m2Ys6xt0nUW7/vGT1M0NPAgMBAAGjQjBAMA4GA1UdDwEB/wQEAwIBBjAPBgNV\n" \
  "HRMBAf8EBTADAQH/MB0GA1UdDgQWBBR5tFnme7bl5AFzgAiIyBpY9umbbjANBgkq\n" \
  "hkiG9w0BAQsFAAOCAgEAVR9YqbyyqFDQDLHYGmkgJykIrGF1XIpu+ILlaS/V9lZL\n" \
  "ubhzEFnTIZd+50xx+7LSYK05qAvqFyFWhfFQDlnrzuBZ6brJFe+GnY+EgPbk6ZGQ\n" \
  "3BebYhtF8GaV0nxvwuo77x/Py9auJ/GpsMiu/X1+mvoiBOv/2X/qkSsisRcOj/KK\n" \
  "NFtY2PwByVS5uCbMiogziUwthDyC3+6WVwW6LLv3xLfHTjuCvjHIInNzktHCgKQ5\n" \
  "ORAzI4JMPJ+GslWYHb4phowim57iaztXOoJwTdwJx4nLCgdNbOhdjsnvzqvHu7Ur\n" \
  "TkXWStAmzOVyyghqpZXjFaH3pO3JLF+l+/+sKAIuvtd7u+Nxe5AW0wdeRlN8NwdC\n" \
  "jNPElpzVmbUq4JUagEiuTDkHzsxHpFKVK7q4+63SM1N95R1NbdWhscdCb+ZAJzVc\n" \
  "oyi3B43njTOQ5yOf+1CceWxG1bQVs5ZufpsMljq4Ui0/1lvh+wjChP4kqKOJ2qxq\n" \
  "4RgqsahDYVvTH9w7jXbyLeiNdd8XM2w9U/t7y0Ff/9yi0GE44Za4rF2LN9d11TPA\n" \
  "mRGunUHBcnWEvgJBQl9nJEiU0Zsnvgc/ubhPgXRR4Xq37Z0j4r7g1SgEEzwxA57d\n" \
  "emyPxgcYxn/eR44/KJ4EBs+lVDR3veyJm+kXQ99b21/+jh5Xos1AnX5iItreGCc=\n" \
  "-----END CERTIFICATE-----\n"; 

struct reqResponse_t {
    int code;
    String payload;
};

class CustomHTTPClient {
    private:
        WiFiClient *_client;
        WiFiClientSecure *_secureClient;
        HTTPClient _http;
        char* _certificate = NULL;
        bool _isHttps = false;
        reqResponse_t _response = {-1, ""};

    public:
        CustomHTTPClient() {
            _client = new WiFiClient();
            _secureClient = new WiFiClientSecure();
        }

        ~CustomHTTPClient() {
            delete _client;
            delete _secureClient;
        }

        void setSecurity(  const bool isHttps = false, const char* certificate = NULL) {
            _isHttps = isHttps;
            if ( _isHttps) _secureClient->setCACert( certificate);
            else _secureClient->setInsecure();
        }

        bool setBasicAuth( const char* user, const char* password) {
            if ( _isHttps) {
                _http.setAuthorization(user, password);
                return true;
            }
            else return false;
        }

        reqResponse_t sendRequest ( const char * type, const char * serverPath, const char * payload = NULL) {
            if(WiFi.status()== WL_CONNECTED){
                WiFiClient *client = _isHttps ? _secureClient : _client;
                if ( _http.begin( *client, serverPath)) {
                    int httpCode = 0;
                    if ( strcmp( type, "GET") == 0) httpCode = _http.GET();
                    else {
                        _http.addHeader("Content-Type", "application/json");
                        httpCode = _http.sendRequest( type, (uint8_t *) payload, strlen( payload));
                    }
                    _response.code = httpCode;
                    if ( httpCode > 0) _response.payload = _http.getString();
                    else _response.payload = "Error: " + _http.errorToString( httpCode);
                    _http.end();
                }
            }
            else {
                _response.code = -1;
                _response.payload = "WiFi Disconnected";
            }
            return _response;
        }
};

void setup() {
    Serial.begin(115200); 

    WiFi.begin("SSID", "WifiPassword");
    Serial.println("Connecting");
    while(WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
    }
    Serial.println("");
    Serial.print("Connected to WiFi network with IP Address: ");
    Serial.println(WiFi.localIP());
    
    reqResponse_t response;
    CustomHTTPClient httpClient;
    httpClient.setSecurity( true, test_root_ca);
//    httpClient.setBasicAuth( "user", "password");

    response = httpClient.sendRequest( "GET", "https://www.howsmyssl.com/a/check");

    Serial.printf("HTTP Response code: %d\n", response.code);
    Serial.println(response.payload);
}

void loop() {}

```

jhud, 2025-06-08:
To avoid digging into the intricacies of the class heirarchies, it seems you can call setPlainStart() on your WiFiClientSecure instance before your TCP/IP call, and it will skip all the SSL negations and work as a plain WiFiClient.

From the source:
``` // Certain protocols start in plain-text; and then have the client
  // give some STARTSSL command to `upgrade' the connection to TLS
  // or SSL. Setting PlainStart to true (the default is false) enables
  // this. It is up to the application code to then call 'startTLS()'
  // at the right point to initialize the SSL or TLS upgrade.```