espressif/arduino-esp32 #3661: Can Bus - ISR makes Guru Meditation Error: Core  1 panic'ed (Coprocessor exception)

Petros144, 2020-01-24:
### Hardware:
Board:							ESP32 Dev Module 
Core Installation version:			1.01
IDE name:						Arduino IDE? 
Flash Frequency:					80Mhz
PSRAM enabled:                                      no
Upload Speed:					115200
Computer OS:                                          Windows 8

 **Description:**

Hi! Im workin on a Can Bus Display right now, Im useing the on board Can Controler and external CAN transceiver.
TFT used: ili9341
Bluetooth running ANCS

For CAN Bus I Use this Library:
https://github.com/sandeepmistry/arduino-CAN

In Particular the Callback function : CAN.onReceive(onReceive);

I also send Frames in my loop() in 100ms Intervals to not block the Bus.

Everything works smooth but after a short while (sometimes 10sec /5sec) the Display shows pixel errors and the ESP restarts.

**This is my investigation so far:**

```
rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:10088
load:0x40080400,len:6380
entry 0x400806a4
**Guru Meditation Error: Core  1 panic'ed (Coprocessor exception)**
Core 1 register dump:
PC      : 0x400d1adb  PS      : 0x00060031  A0      : 0x800d7bb8  A1      : 0x3ffbe9b0  
A2      : 0x3ffc55b4  A3      : 0x00000001  A4      : 0x80093912  A5      : 0x00000000  
A6      : 0x00000001  A7      : 0x3ffbaca0  A8      : 0x00000000  A9      : 0x000b4103  
A10     : 0x000007e8  A11     : 0x000000fd  A12     : 0x0ffdac00  A13     : 0x0000abab  
A14     : 0x3ffb9a30  A15     : 0x00000000  SAR     : 0x00000019  EXCCAUSE: 0x00000004  
EXCVADDR: 0x00000000  LBEG    : 0x400df614  LEND    : 0x400df676  LCOUNT  : 0x00000003  
Core 1 was running in ISR context:
EPC1    : 0x400d1adb  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40082941

Backtrace: 0x400d1adb:0x3ffbe9b0 0x400d7bb5:0x3ffbea90 0x400d7bc1:0x3ffbeab0 0x40081819:0x3ffbead0 0x4000bfed:0x00000000

Rebooting...
```
**Exception Decoder spits this out:**

```
Decoding stack results
0x400d406f: onReceive(int) at C:\Users\PETROS~1\AppData\Local\Temp\arduino_modified_sketch_534488/ESP32_BT_6.1_OBDTIME_1.6.1_ANCS_v1.7.2.ino line 2788
0x400d7e51: ESP32SJA1000Class::handleInterrupt() at C:\Users\Petros T\Documents\Arduino\libraries\CAN\src\ESP32SJA1000.cpp line 383
0x400d7e5d: ESP32SJA1000Class::onInterrupt(void*) at C:\Users\Petros T\Documents\Arduino\libraries\CAN\src\ESP32SJA1000.cpp line 410
```

**Theese are the Lines inside the Lib:**

```
void ESP32SJA1000Class::handleInterrupt()
{
  uint8_t ir = readRegister(REG_IR);

  if (ir & 0x01) {
    // received packet, parse and call callback
    parsePacket();

383 >>>    _onReceive(available());
  }
}

void ESP32SJA1000Class::onInterrupt(void* arg)
{
410 >>>  ((ESP32SJA1000Class*)arg)->handleInterrupt();
}
```

**This is what I tried so far:**

1. Isolate the Can Recive task (without Callback) into separate RTOS Task, This works but slows the Whole Code down from 25FPS to 10FPS on ths display.
-> Not shure how to pass the callback function to a tasks, may this work? 

2. Tried to get other CAN Librarys to work -> also Too slow.

Hoping for help.

Petros144, 2020-01-24:
I found the issue,

It its not allowed to have a Float variable inside the Callback function.

After replacing this with a Double variable everything works!
found on ESP Forums:

> Floats use the FPU while doubles are calculated in software. For reasons, using the FPU inside an interrupt handler is currently not supported.

closed 2020-01-24 by Petros144

atanisoft, 2020-01-24:
@Petros144 any reason not to use the built-in CAN framework?

Petros144, 2020-02-19:
@atanisoft I need the speed provided via Interrrupt routine. now it works :)!

qcarver, 2020-03-30:
Thank you so much! I've got demo in 5 hours and had the exact same error with my code:

> static void IRAM_ATTR pps_isr_handler(void* arg)
> {
> 	//Q passes by copy not reference, make a container
> 	GPS_TIME::PACKET blessMe;
> 	//not used
> 	BaseType_t xxx = false;
> 
> 	//Move front of UART q into blessed Q
> 	if (xQueueReceiveFromISR(hUartRxdQ,(void *)&blessMe, &xxx)){
> 		//restamp with time of blessing
> 		blessMe.espReceiptTime += ONE_MILLION;
> 		//per 63530-10_Rev-B_Manual_Copernicus-II.pdf add 1 full second
> 		blessMe.gpsTimeOfWeek += 1.0f;
> 		//push to top of blessed Q
> 		xQueueOverwrite(hPpsBlessedQ,(void *)&blessMe);
> 	}
> }

I pulled the += statments out of the ISR and it worked perfectly!

miksumin, 2021-02-05:
I got the same issue in the interrupt handler procedure. Replacing float variable with double one solved the issue. I think it can be marked as a bug in working with float in ISR.

johnny1988, 2021-02-24:
Hello @Petros144 , I am also using saem code , similar CANbus tests
I also have same same problem, i want to know exactly where you chnaged from Float to Double, which call back function, can you copy paste the code function where it was , which library function is that?
Please help me , I need to solve same problem 
Thank you

Daniel-Chin, 2023-05-07:
> I found the issue,
>
> It its not allowed to have a Float variable inside the Callback function.
>
> After replacing this with a Double variable everything works! found on ESP Forums:

Just for convenience, the forum post in question is: https://esp32.com/viewtopic.php?t=1292

zalexzperez, 2023-07-18:
> @Petros144 any reason not to use the built-in CAN framework?

Does the TWAI driver support callback function so that every time a package is received, a function is executed?
The receive example only performs polling on the loop() to check if new packages arrived...

atanisoft, 2023-07-18:
> Does the TWAI driver (the built-in CAN framework) support callback function so that every time a package is received, a function is executed?

It only supports polled

zalexzperez, 2023-07-18:
> It only supports polled

That means some packages will be lost if our receive function is executed at the end of the void loop, right?
In that case, what is usually done to prevent it? Using a timer to execute the receive function every few milliseconds?

atanisoft, 2023-07-18:
> In that case, what is usually done to prevent it?

You can control how many frames will be buffered for reading when you configure the TWAI driver. You can also do as the examples do and run a higher priority task (not in `loop()`) which retrieves the frames from the RX queue as they become available and process them.

zalexzperez, 2023-07-18:
> You can also do as the examples

Do you mean this example? https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/TWAI/TWAIreceive/TWAIreceive.ino
I don't see the priority part.

And one final question, are you aware if it's actually possible to read a single-wire CAN? (33.3kbps GMLAN).
I tried doing it with Sandeep Mistry's library, but when connecting the transceiver's CANL pin to GND, my car's speedometer turns off and of course I'm not reading any packages. Unlike with the MCP2515(+TJA1050) and an Arduino.

atanisoft, 2023-07-18:
> Do you mean this example? https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/TWAI/TWAIreceive/TWAIreceive.ino
> I don't see the priority part.

That doesn't use a dedicated task, it uses polled. https://github.com/espressif/esp-idf/blob/release/v4.4/examples/peripherals/twai/twai_network/twai_network_slave/main/twai_network_example_slave_main.c shows using multiple tasks, you should focus primarily on calls involving `twai_receive_task`.