espressif/arduino-esp32 #3668: swithching off radio reduces available RAM

curly-brace, 2020-01-25:
Make your question, not a Statement, inclusive. Include all pertinent information:

i'm trying to disable radio completely with
```
  btStop();
  WiFi.mode(WIFI_OFF);
```
when using this functions they change nothing
```
  esp_bt_controller_disable();
  esp_bt_controller_deinit();
  esp_bt_controller_mem_release(ESP_BT_MODE_IDLE);
```
before i do it i have
```
heap size: 388252
heap free: 362576
heap max alloc: 124936
```
and after
```
heap size: 324496
heap free: 299220
heap max alloc: 113792
```
64k of RAM is gone!!! just two lines of code! nothing else is written in the project. magic!
I'm using lolin32 board and platformio, but in arduino ide it is exactly the same tho. running latest stable core version.

chegewara, 2020-01-26:
`esp_bt_controller_mem_release(ESP_BT_MODE_IDLE);` ???

Besides that, the magic is because when you use any bluetooth and/or wifi command then bluetooth and/or wifi driver will be merged to your app and driver will be loaded to memory. If you wont use then memory wont be allocated. So, if you dont want to use wifi and/or BT just dont try to turn off radio, it wont be used at all.

curly-brace, 2020-01-27:
@chegewara so you mean that if i don't use BT or WIFI the radio won't consume any current at all?

chegewara, 2020-01-27:
yes, thats correct

closed 2020-01-27 by curly-brace