espressif/arduino-esp32 #3669: BluetoothSerial not connect

peppeve, 2020-01-25:
I try to use the example  SerialtoserialBTM trying to connect a esp32 with an HC-05, set correct name, pin and address but no connection getted.
``The device started in master mode, make sure remote BT device is on!
Failed to connect. Make sure remote device is available and in range, then restart app.
Failed to connect. Make sure remote device is available and in range, then restart app.
Failed to connect. Make sure remote device is available and in range, then restart app.
Failed to connect. Make sure remote device is available and in range, then restart app.
Failed to connect. Make sure remote device is available and in range, then restart app.
Failed to connect. Make sure remote device is available and in range, then restart app.
Failed to connect. Make sure remote device is available and in range, then restart app.
Failed to connect. Make sure remote device is available and in range, then restart app.
Failed to connect. Make sure remote device is available and in range, then restart app.
Failed to connect. Make sure remote device is available and in range, then restart app.
Failed to connect. Make sure remote device is available and in range, then restart app.
Failed to connect. Make sure remote device is available and in range, then restart app.
Failed to connect. Make sure remote device is available and in range, then restart app.
Failed to connect. Make sure remote device is available and in range, then restart app.
Failed to connect. Make sure remote device is available and in range, then restart app.
Failed to connect. Make sure remote device is available and in range, then restart app.
Failed to connect. Make sure remote device is available and in range, then restart app.
Failed to connect. Make sure remote device is available and in range, then restart app.
Failed to connect. Make sure remote device is available and in range, then restart app.
``

GPSFan, 2020-01-25:
I have been experiencing the same problem for quite some time, however I have made some progress. IIRC the original BluetoothSerial code was tested with a Bluetooth GPS device. The ones I have, Qstarz BT-Q816 and BT-Q1000, use a BT module based on the RF Micro Devices SiW3500.
At first, I was only trying to get the BT Serial in Master mode to connect to an HC-05, lots of print statements later, I was at wits end so I tried esp32-IDF and after a long learning curve I got the HC-05 to connect. I then went back to the esp32-arduino source code and compared it to the IDF code.... too many differences for me to be sure of anything, ie 1.04 (3.2) vs the latest Github IDF master.
So then I dug out my BT GPS receivers and both would connect to the esp32 using the 1.04 based code.

I think that the HC-05 is different from other BT Classic serial adapters like the JDY-30 and those based on the SiW3500. Even though it advertises the same SPP capabilities, there is something different in how it responds to discovery requests.
I bought 3 JDY-30 modules and found I could connect to them just as to the SiW3500 based BT GPS devices. 
My conclusion is to wait for the IDF 4.0 based esp32-arduino code to be released and try again with the HC-05.

peppeve, 2020-01-25:
@GPSFan 
I updated to 1.04 and get connected.

``[I][BluetoothSerial.cpp:510] _init_bt(): device name set
The device started in master mode, make sure remote BT device is on!
[I][BluetoothSerial.cpp:225] esp_spp_cb(): ESP_SPP_INIT_EVT
[I][BluetoothSerial.cpp:722] connect(): master : remoteAddress
[I][BluetoothSerial.cpp:290] esp_spp_cb(): ESP_SPP_DISCOVERY_COMP_EVT
Connected Succesfully!
``
but my HC-05 send a lot of data out (i can check with my mobile connected to him), but nothing i see on the serial monitor. Then connection seem to be ok but no data exchanged.

GPSFan, 2020-01-25:
What version firmware is your HC-05?
AT+VERSION?\n\r

gogasan24ru, 2020-01-26:
Hello. I have same issue. 
My `AT+VERSION` returns `linvorV1.8`

peppeve, 2020-01-26:
@

> What version firmware is your HC-05?
> AT+VERSION?\n\r

I can't check, my hc-05 is installed inside an electrical box since 5 years.

GPSFan, 2020-01-26:
The HC-05's I have use: VERSION: 2.0-20100601
The Internet tells me that there are several more versions:
VERSION: 2.0-20100601
VERSION 3.0-20170601
VERSION:4.0-20190815 
Linvor version 1.5
linvor V1.8
I would expect differences and similarities between them all. I have no clue as to how the ones other than VERSION: 2.0-20100601 operate.
peppeve, if it's been there for 5 years, it is probably not VERSION:4.0-20190815.
All of these are based on the CSR (Cambridge Silicon Radio) BC417 chip.
The JDY-30 ones I have are based on the  Beken BK3231 chip, these work with the 1.04 esp32-arduino code. My HC-05's with firmware VERSION: 2.0-20100601 do not.
From the commit activity here lately I would expect a new release soon. Then, if it still doesn't work, will be the time to poke more at the problem.

peppeve, 2020-01-26:
@GPSFan I tested with the new version of the Bluetooth serial library coming from idf 4 repository, but nothing change.

GPSFan, 2020-01-26:
IMHO the BluetoothSerial.cpp and .h files are not the ones that matter, it's the esp-hal code that does the lower level interaction with the BlueDroid Classic stack that matters, there are so many changes between 3.2 (1.04) and the GitHub head that it's impossible for me to tell where the problem might lie.  I am perfectly willing to wait for the next release and start debugging from a fresh release.

GPSFan, 2020-01-26:
My application (which is working mostly) gets data from a 115200 UART port on a u-blox M9N connected to the JDY-30 then via Bluetooth Classic to a M5StickC (in SPP Master mode) where the UBX protocol messages are parsed and the Sat positions and SNR are graphed. I'm doing this in esp32-arduino now, but may move to esp32-IDF if the application gets too complex. I thought of using an esp32 on both ends, but I had the HC-05 modules and it should have worked, I believe it will with an HC-05 when the IDF 4.x is integrated into the esp32-arduino code. In the mean time, I got the JDY-30's to see if they would work and the did, so I continued development of my esp32 app/sketch/program.

peppeve, 2020-01-26:
My hc-05 in installed on my domotic house system, I have around 25 Arduino pro mini around my house made 5/6 years ago that communicate by a simple rs475 protocol. The hc-05 is used to bridge all data on the bus to my phone for debug or drive lights open door, Chech consumptions, etc. Now I upgrade my system with an esp32 emulating Philips hue light, for drive my system by Alexa. For do this I need to bridge my esp32 to my old bus rs485 by the hc-05 already installed on the system.

GPSFan, 2020-01-26:
Having HC-05's already installed in equipment is why I am interested in this problem. If I was starting from scratch I'd not use HC-05's as the JDY-30 is as cheap and works. Better to use an esp32 derivative and have full control of the firmware.

peppeve, 2020-02-08:
@GPSFan any news?

peppeve, 2020-04-06:
@GPSFan update? found a solution?

GPSFan, 2020-04-07:
I tried the idf-release/v4.0 and was able to get the hc-05 to connect, but no data transfer. I see that there has just been an update to that branch so I'll give it another try when I get another round tuit...

vanniaz, 2020-06-11:
I noticed that BluetoothSerial is reporting "connected" even when there is no device connected, so when you say that it connects but does not transfer data, the problem could be in the connection.
I opened a new issue for this:
#4081

closed 2020-08-25 by stale[bot]