espressif/arduino-esp32 #3671: undefined reference to `app_main'  OR undefined reference to `setup'

TLINDEN, 2020-01-25:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do?
Describe your system( Hardware, computer, O/S, core version, environment).
Describe what is failing.
Show the shortest possible code that will duplicate the error.
Show the EXACT error message(it doesn't work is not enough).
All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							Lolin esp32
Core Installation version:			n/a
IDE name:						emacs
Flash Frequency:					40Mhz
PSRAM enabled:                                    n/a
Upload Speed:					n/a
Computer OS:                                        Ubuntu

### Description:

I have a fresh installation and haven't done anything with esp32 before. I want to write my code with emacs and just use `make flash` to install firmwares.

I've installed 
`
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
`
I followed these insttructions: https://docs.espressif.com/projects/esp-idf/en/stable/get-started/linux-setup.html. I had to use the the release/v3.2.3 branch of esp-idf though because otherwise I couldn't get the whole thing to work.

Then I followed issue #1409. I created a new project folder with main/test.cpp containing the code posted in this issue. My Makefile has `test` set as project name.

When I compile with arduino autostart disabled I get:
```
LD build/test.elf
/home/scip/D/github/buswetter/esptest/build/esp32/libesp32.a(cpu_start.o):(.literal.main_task+0x18): undefined reference to `app_main'
/home/scip/D/github/buswetter/esptest/build/esp32/libesp32.a(cpu_start.o): In function `main_task':
/usr/local/esp32/esp-idf/components/esp32/cpu_start.c:512: undefined reference to `app_main'
collect2: error: ld returned 1 exit status
```
When I enable it, I get the same error as in #1409 (setup and loop undefined):
```
LD build/test.elf
/home/scip/D/github/buswetter/esptest/build/arduino/libarduino.a(main.o):(.literal._Z8loopTaskPv+0x8): undefined reference to `setup()'
/home/scip/D/github/buswetter/esptest/build/arduino/libarduino.a(main.o):(.literal._Z8loopTaskPv+0xc): undefined reference to `loop()'
/home/scip/D/github/buswetter/esptest/build/arduino/libarduino.a(main.o): In function `loopTask(void*)':
/home/scip/D/github/buswetter/esptest/components/arduino/cores/esp32/main.cpp:14: undefined reference to `setup()'
/home/scip/D/github/buswetter/esptest/components/arduino/cores/esp32/main.cpp:17: undefined reference to `loop()'
collect2: error: ld returned 1 exit status
```

It looks like the toolchain doesn't build main/test.cpp at all. In fact, I can even add any non-cpp rubbish in there and I get the same message as above, no syntax errors or the like.

What can I do to fix this weird behavior?

TLINDEN, 2020-01-26:
Ok, my bad. I had no `main/CMakeLists.txt` AND the sample code posted in #1409 is wrong (macros wrong). So, it works now.

closed 2020-01-26 by TLINDEN

Jobui21, 2020-08-29:
Hello I hope you are well. look I am also presenting this problem. since I had been working well with the Esp32 card for months and I even get those problems. Can you advise me how did you solve that?

lbernstone, 2020-08-29:
This is very outdated.  You can now use [lib-builder](https://github.com/espressif/arduino-esp32/blob/master/docs/lib_builder.md) to get everything  laid out properly, and then just add your main code. Choose either app_main or setup by adjusting the arduino settings in make menuconfig.

Sergio-Navarro-M, 2021-08-12:
> Ok, my bad. I had no `main/CMakeLists.txt` AND the sample code posted in #1409 is wrong (macros wrong). So, it works now.
Hello @TLINDEN ,

can you please share the code that did work for you? I am having the same issues while compiling test code.

Thanks in advance!
Sergio