espressif/arduino-esp32 #3674: How to add another arduino library?

TLINDEN, 2020-01-26:
I want to use the GxEDP library in my project. I cloned the directory into the libraries directory and added its *.cpp files to `CMakeLists.txt` and an entry into `Kconfig.projbuild`. It shows up in `make menuconfig` but when I try to build, I get:

```
Arduino libraries in build:       ESP32   FS GxEPD    Preferences    SPI      WiFi  Wire
make[1]: *** No rule to make target `libraries/GxEPD/src//GxEPD.o', needed by `libarduino.a'.  Stop.
make: *** [component-arduino-build] Error 2
```

Any idea?

lbernstone, 2020-01-26:
The sketchbook location is typically <home>/Arduino.  There is a libraries folder in there.  Drag and drop.

TLINDEN, 2020-01-27:
This directory doesn't exist since I'm not using arduino.

However, I managed to get it working, I had to add all *.c files to `CMakeLists.txt` as well.

closed 2020-04-10 by stale[bot]