espressif/arduino-esp32 #3676: something unknown happened about ESP32 heap

p2o5, 2020-01-27:
[ESP32-Cam_CameraWebServer.zip](https://github.com/espressif/arduino-esp32/files/4116182/ESP32-Cam_CameraWebServer.zip)
### Hardware:
Board:	ESP32-CAM  (ESP32 wrover model)
Core Installation/update date:			I don't know how to find
IDE name:							Arduino IDE 1.8.10
Flash Frequency:					80Mhz
PSRAM enabled:                                          yes
Upload Speed:						115200
Computer OS:                                            Windows 10

### Description:
I replace qr-code recognize with face recognize in example  "ESP32-camera-CameraWebServer"
something unknown happened about ESP32 heap

### Sketch
example "ESP32-camera-CameraWebServer"
replace qr code recognize with face recognize
"main.ino"=========
...
#define CAMERA_MODEL_AI_THINKER
...
"app_httpd.cpp"============
...
#include "quirc_internal.h"
#include <quirc.h>
static struct quirc *qr_recognizer;
static uint16_t old_width = 0;
static uint16_t old_height = 0;
static uint8_t *image = NULL;
static int id_count = 0;
static struct quirc_code code;
static struct quirc_data data;
...
static void rgb2grayscale(rgb_t *rgb, uint8_t *gray, uint16_t len) {
  uint32_t rgbv;
  for (int i = 0; i < len; i++) {
    rgbv = (rgb->r * 19595 + rgb->g * 38469 + rgb->b * 7472);
    *gray = rgbv >> 16;
    rgb = rgb + 1;
    gray = gray + 1;
  }
}

static int q_resize(struct quirc *q, int w, int h)
{
  int ret;
  uint8_t   *image  = NULL;
  quirc_pixel_t *pixels = NULL;
  size_t olddim, newdim, min;
  Serial.printf("q_resize w=%d,h=%d\n", w, h);
  /*
     XXX: w and h should be size_t (or at least unsigned) as negatives
     values would not make much sense. The downside is that it would break
     both the API and ABI. Thus, at the moment, let's just do a sanity
     check.
  */
  if (w < 0 || h < 0) {
    Serial.printf("q_resize2 w=%d,h=%d\n", w, h);
    ret = -8;
    goto fail;
  }

  /*
     alloc a new buffer for q->image. We avoid realloc(3) because we want
     on failure to be leave `q` in a consistant, unmodified state.
  */
  //image = (uint8_t*)calloc(w, h);
  image = (uint8_t*)heap_caps_malloc(w*h,MALLOC_CAP_SPIRAM);
  if (!image) {
    Serial.printf("largest free block=%d\n", heap_caps_get_largest_free_block(MALLOC_CAP_8BIT));
[... 182 lines cut ...]
A2      : 0x00000001  A3      : 0x0000cdcd  A4      : 0xb33fffff  A5      : 0x00000001  
A6      : 0x00060220  A7      : 0x0000abab  A8      : 0x0000cdcd  A9      : 0x3ffd8610  
A10     : 0x00000003  A11     : 0x00060223  A12     : 0x00060220  A13     : 0x00000001  
A14     : 0x00060020  A15     : 0x00000000  SAR     : 0x00000018  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000001  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x4008ebaa:0x3ffd8610 0x40090a33:0x3ffd8640 0x400911c3:0x3ffd8660 0x400872d8:0x3ffd8680 0x4008730d:0x3ffd86a0 0x40087891:0x3ffd86c0 0x4000beaf:0x3ffd86e0 0x400819b9:0x3ffd8700 0x40148593:0x3ffd8720 0x40152cd9:0x3ffd8760 0x400947c5:0x3ffd87e0 0x4008db91:0x3ffd8820

Rebooting...

=======================
ESP Exception Decoder:

PC: 0x4008ebaa: vTaskEnterCritical at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/include/freertos/portmacro.h line 285
EXCVADDR: 0x00000001

Decoding stack results
0x4008ebaa: vTaskEnterCritical at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/include/freertos/portmacro.h line 285
0x40090a33: multi_heap_internal_lock at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap.c line 375
0x400911c3: multi_heap_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c line 190
0x400872d8: heap_caps_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c line 111
0x4008730d: heap_caps_malloc_default at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c line 140
0x40087891: _malloc_r at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/syscalls.c line 37
0x400819b9: wifi_malloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/esp_adapter.c line 68
0x4008db91: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

p2o5, 2020-02-01:
@me-no-dev I did have post source code in my issue,top of it ,here is the link(https://github.com/espressif/arduino-esp32/files/4116182/ESP32-Cam_CameraWebServer.zip), Please try it

closed 2020-04-15 by stale[bot]

modiestas, 2023-04-03:
@p2o5 Did you find solution?