espressif/arduino-esp32 #3677: BLEScan first scan ok, next no name, no service UUID

Nagymadar, 2020-01-27:
I am using BLE_scan.ino example. Unfortunately only the first scan gives valid device serviceUUID and Name. From the second scan onward SENSOR_PRO device gives no serviceUUID the HMSoft device gives empty string as Name. It seems it is an old bug, but still no solution? Serial data from boot:

Scanning...
Advertised Device: Name: SENSOR_PRO, Address: ed:3d:9a:96:86:a3, serviceUUID: 6a800001-b5a3-f393-e0a9-e50e24dcca9e 
Advertised Device: Name: HMSoft, Address: 90:9a:77:27:65:15, manufacturer data: 484d909a77276515, serviceUUID: 0000ffe0-0000-1000-8000-00805f9b34fb, txPower: 0 
Devices found: 2
Scan done!

Advertised Device: Name: SENSOR_PRO, Address: ed:3d:9a:96:86:a3 
Advertised Device: Name: , Address: 90:9a:77:27:65:15, manufacturer data: 484d909a77276515, serviceUUID: 0000ffe0-0000-1000-8000-00805f9b34fb, txPower: 0 
Devices found: 2
Scan done!

Advertised Device: Name: SENSOR_PRO, Address: ed:3d:9a:96:86:a3 
Advertised Device: Name: , Address: 90:9a:77:27:65:15, manufacturer data: 484d909a77276515, serviceUUID: 0000ffe0-0000-1000-8000-00805f9b34fb, txPower: 0 
Devices found: 2
Scan done!

etc...

chegewara, 2020-01-29:
Its hard to do anything if i cant reproduce issue:
```
03:27:33.927 -> Device: 
03:27:33.927 -> TX power => -7
03:27:33.927 -> RSSI => -96
03:27:33.927 -> Name: Chege
03:27:34.226 -> Device: 
03:27:34.226 -> TX power => -7
03:27:34.226 -> RSSI => -85
03:27:34.226 -> Name: Chege
03:27:34.493 -> Device: 
03:27:34.493 -> TX power => -7
03:27:34.493 -> RSSI => -85
03:27:34.493 -> Name: Chege
03:27:34.792 -> Device: 
03:27:34.792 -> TX power => -7
03:27:34.792 -> RSSI => -86
03:27:34.792 -> Name: Chege
03:27:35.092 -> Device: 
03:27:35.092 -> TX power => -7
03:27:35.092 -> RSSI => -84
03:27:35.092 -> Name: Chege
03:27:35.658 -> Device: 
03:27:35.658 -> TX power => -7
03:27:35.658 -> RSSI => -89
03:27:35.658 -> Name: Chege
03:27:36.233 -> Device: 
03:27:36.233 -> TX power => -7
03:27:36.233 -> RSSI => -91
03:27:36.233 -> Name: Chege
03:27:36.529 -> Device: 
03:27:36.529 -> TX power => -7
03:27:36.529 -> RSSI => -81
03:27:36.529 -> Name: Chege
```
My peer device is android smartphone running nRF connect with advertising.
Here is scan callback:
```
class ScanResults: public BLEAdvertisedDeviceCallbacks {
  void onResult(BLEAdvertisedDevice device)
  {
    if(device.haveName())
    {
      Serial.printf("Device: \nTX power => %d\nRSSI => %d\n", device.getTXPower(), device.getRSSI());
      Serial.printf("Name: %s\n", device.getName().c_str());
    }
  }
};
```
I can only say that i am running this code on platformio with arduino + esp-idf, wifi and mqtt (code omitted).

Nagymadar, 2020-01-29:
With the original arduino BLE_scan.ino, just modified the callback as below. Still no luck. I have tried haveName() too in the callback, but then I got back nothing after the first scan. 

Hardware: ESP32 Wrover Module, default settings.

```
class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
    void onResult(BLEAdvertisedDevice device) {
      //Serial.printf("Advertised Device: %s \n", device.toString().c_str());
      Serial.printf("Device: \nTX power => %d\nRSSI => %d\n", device.getTXPower(), device.getRSSI());
      Serial.printf("Name: %s\n", device.getName().c_str());
    }
};
```

Results:

```
Scanning...
Device: 
TX power => 0
RSSI => -91
Name: Comet Blue
Device: 
TX power => 12
RSSI => -92
Name: 
Device: 
TX power => 0
RSSI => -78
Name: Comet Blue

Devices found: 3
Scan done!
Device: 
TX power => 0
RSSI => -95
Name: 
Device: 
TX power => 0
RSSI => -79
Name: 
Device: 
TX power => 12
RSSI => -92
Name: 

Devices found: 3
Scan done!
Device: 
TX power => 0
RSSI => -90
Name: 
Device: 
TX power => 12
RSSI => -94
Name: 
Device: 
TX power => 0
RSSI => -86
Name: 
```

Nagymadar, 2020-01-29:
Some devices loose their name, some their UUID. See results and full code below:

Results:

```
Scanning...
Device Name: SENSOR_PRO, TX: 0, RSSI: -70, UUID: 6a800001-b5a3-f393-e0a9-e50e24dcca9e
Device Name: Comet Blue, TX: 0, RSSI: -90, UUID: 47e9ee00-47e9-11e4-8939-164230d1df67
Devices found: 2
Scan done!

Device Name: SENSOR_PRO, TX: 0, RSSI: -58
Device Name: , TX: 0, RSSI: -89, UUID: 47e9ee00-47e9-11e4-8939-164230d1df67
Devices found: 2
Scan done!

Device Name: , TX: 0, RSSI: -91, UUID: 47e9ee00-47e9-11e4-8939-164230d1df67
Device Name: SENSOR_PRO, TX: 0, RSSI: -58
Devices found: 2
Scan done!

Device Name: SENSOR_PRO, TX: 0, RSSI: -58
Device Name: , TX: 0, RSSI: -91, UUID: 47e9ee00-47e9-11e4-8939-164230d1df67
Devices found: 2
Scan done!
```

Code:

```
#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEScan.h>
#include <BLEAdvertisedDevice.h>

int scanTime = 5; //In seconds
BLEScan* pBLEScan;

class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
    void onResult(BLEAdvertisedDevice device) {
      Serial.printf("Device Name: %s, TX: %d, RSSI: %d", device.getName().c_str(), device.getTXPower(), device.getRSSI());
      if (device.haveServiceUUID()) Serial.printf(", UUID: %s", device.getServiceUUID().toString().c_str());
      Serial.println("");
    }
};

void setup() {
  Serial.begin(115200);
  Serial.println("Scanning...");

  BLEDevice::init("");
  pBLEScan = BLEDevice::getScan(); //create new scan
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setActiveScan(true); //active scan uses more power, but get results faster
  pBLEScan->setInterval(100);
  pBLEScan->setWindow(99);  // less or equal setInterval value
}

void loop() {
  // put your main code here, to run repeatedly:
  BLEScanResults foundDevices = pBLEScan->start(scanTime, false);
  Serial.print("Devices found: ");
  Serial.println(foundDevices.getCount());
  Serial.println("Scan done!\n");
  pBLEScan->clearResults();   // delete results fromBLEScan buffer to release memory
  delay(2000);
}
```

Nagymadar, 2020-01-30:
Can I produce some debug info for you? (raw data print or something)

chegewara, 2020-01-31:
You can print raw advertising data for each device in onResult() callback with:
getPayload() and getPayloadLength() and Serial functions.

This way we can investigate it. Remember to print hex values.

Nagymadar, 2020-01-31:
I switched also on Warnings and printed heap size. Here it is:

```
[W][esp32-hal-psram.c:30] psramInit(): PSRAM init failed!
Scanning...
Device: Name: SENSOR_PRO, Address: e5:89:06:ef:6d:e8, serviceUUID: 6a800001-b5a3-f393-e0a9-e50e24dcca9e 
Payload [33] = { 0B 09 53 45 4E 53 4F 52 5F 50 52 4F 02 01 06 11 07 9E CA DC 24 0E E5 A9 E0 93 F3 A3 B5 01 00 80 6A }
[W][BLEScan.cpp:69] handleGAPEvent(): ESP_GAP_SEARCH_INQ_CMPL_EVT
Devices found: 1
Heap: 178128
Scan done!

Device: Name: SENSOR_PRO, Address: e5:89:06:ef:6d:e8 
Payload [15] = { 0B 09 53 45 4E 53 4F 52 5F 50 52 4F 02 01 06 }
[W][BLEScan.cpp:69] handleGAPEvent(): ESP_GAP_SEARCH_INQ_CMPL_EVT
Devices found: 1
Heap: 178168
Scan done!

Device: Name: SENSOR_PRO, Address: e5:89:06:ef:6d:e8 
Payload [15] = { 0B 09 53 45 4E 53 4F 52 5F 50 52 4F 02 01 06 }
[W][BLEScan.cpp:69] handleGAPEvent(): ESP_GAP_SEARCH_INQ_CMPL_EVT
Devices found: 1
Heap: 178168
Scan done!
```

Nagymadar, 2020-01-31:
Don't ask me how, but one version of the code worked, but only once. Then I changed one character in the code (printf format), uploaded it and failed again. Change back to the working one, fail again.

Now I changed the hardware to a completely different board: NodeMcu ESP32. Same error. I switced on verbose level debug. Does this help you? 

What is your ESP board settings? I updated every board and library. You?

```
[W][esp32-hal-psram.c:30] psramInit(): PSRAM init failed!
*** Restart finished. Scanning...
[V][BLEScan.cpp:195] start(): >> start(duration=5)
[D][FreeRTOS.cpp:189] take(): Semaphore taking: name: ScanEnd (0x3ffdcfc4), owner: <N/A> for start
[D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: ScanEnd (0x3ffdcfc4), owner: start
[V][BLEScan.cpp:227] start(): << start()
[V][FreeRTOS.cpp:63] wait(): >> wait: Semaphore waiting: name: ScanEnd (0x3ffdcfc4), owner: start for start
[V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 2 0x02
[V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 7 0x07
[V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[D][BLEAdvertisedDevice.cpp:418] setRSSI(): - setRSSI(): rssi: -49
[V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0x9
[D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0x09 (), length: 10, data: 53454e534f525f50524f
[D][BLEAdvertisedDevice.cpp:407] setName(): - setName(): name: SENSOR_PRO
[V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0x1
[D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0x01 (), length: 1, data: 06
[V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0x7
[D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0x07 (), length: 16, data: 9ecadc240ee5a9e093f3a3b50100806a
[D][BLEAdvertisedDevice.cpp:447] setServiceUUID(): - addServiceUUID(): serviceUUID: 6a800001-b5a3-f393-e0a9-e50e24dcca9e
***   Device: Name: SENSOR_PRO, Address: e5:89:06:ef:6d:e8, serviceUUID: 6a800001-b5a3-f393-e0a9-e50e24dcca9e 
***   Payload [33] = { 0B 09 53 45 4E 53 4F 52 5F 50 52 4F 02 01 06 11 07 9E CA DC 24 0E E5 A9 E0 93 F3 A3 B5 01 00 80 6A }
[V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[D][BLEAdvertisedDevice.cpp:418] setRSSI(): - setRSSI(): rssi: -89
[V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0x1
[D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0x01 (), length: 1, data: 06
[V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0x7
[D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0x07 (), length: 16, data: 67dfd13042163989e411e94700eee947
[D][BLEAdvertisedDevice.cpp:447] setServiceUUID(): - addServiceUUID(): serviceUUID: 47e9ee00-47e9-11e4-8939-164230d1df67
[V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0xff
[D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0xff (), length: 8, data: 2d2a00000000b665
[D][BLEAdvertisedDevice.cpp:395] setManufacturerData(): - manufacturer data: 2d2a00000000b665
[V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0x9
[D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0x09 (), length: 10, data: 436f6d657420426c7565
[D][BLEAdvertisedDevice.cpp:407] setName(): - setName(): name: Comet Blue
***   Device: Name: Comet Blue, Address: 1a:06:f8:4e:e1:d6, manufacturer data: 2d2a00000000b665, serviceUUID: 47e9ee00-47e9-11e4-8939-164230d1df67 
***   Payload [43] = { 02 01 06 11 07 67 DF D1 30 42 16 39 89 E4 11 E9 47 00 EE E9 47 09 FF 2D 2A 00 00 00 00 B6 65 0B 09 43 6F 6D 65 74 20 42 6C 75 65 }
[V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[D][BLEAdvertisedDevice.cpp:418] setRSSI(): - setRSSI(): rssi: -87
[V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0x1
[D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0x01 (), length: 1, data: 06
[V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0x7
[D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0x07 (), length: 16, data: 67dfd13042163989e411e94700eee947
[D][BLEAdvertisedDevice.cpp:447] setServiceUUID(): - addServiceUUID(): serviceUUID: 47e9ee00-47e9-11e4-8939-164230d1df67
[V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0xff
[D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0xff (), length: 8, data: 2c22000000004d9f
[D][BLEAdvertisedDevice.cpp:395] setManufacturerData(): - manufacturer data: 2c22000000004d9f
[V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0x9
[D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0x09 (), length: 10, data: 436f6d657420426c7565
[D][BLEAdvertisedDevice.cpp:407] setName(): - setName(): name: Comet Blue
[... cut ...]
[V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[W][BLEScan.cpp:69] handleGAPEvent(): ESP_GAP_SEARCH_INQ_CMPL_EVT
[V][FreeRTOS.cpp:143] give(): Semaphore giving: name: ScanEnd (0x3ffdcfc4), owner: start
[V][FreeRTOS.cpp:77] wait(): << wait: Semaphore released: name: ScanEnd (0x3ffdcfc4), owner: <N/A>
*** Devices found: 3
*** Heap: 177736
*** Scan done!
```

chegewara, 2020-02-01:
Can you try this before scan start, with value true:
https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/src/BLEScan.h#L51

Nagymadar, 2020-02-02:
I have tried. No change. :-( Any other what I can try?

chegewara, 2020-02-02:
Only thing i can say is that i never found similar issue and it is very strange.
It is obviously ignoring scan response packet which is received only during active scan.
Could you try with other esp32?

Nagymadar, 2020-02-02:
I am using 2 boards:
1. Waveshare e-paper ESP32
2. NodeMCU ESP32S

I am getting close... I used to use Arduino's **ESP32 Wrover Module** board settings in tools menu. Changing to **ESP32 Dev Module** solved the problem on board 1, but not on board 2. Confusing... Any idea?

I've got this warning for both of them:
```
Multiple libraries were found for "BLEDevice.h"
 Used: C:\Users\detre\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\BLE
Using library BLE at version 1.0.1 in folder: C:\Users\detre\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\BLE 
```
But I do not find any other BLEDevice.h anywhere. Can be a problem?

Nagymadar, 2020-02-02:
I scanned my whole c drive. No other **BLEDevice.h**.

chegewara, 2020-02-03:
If i remember this board is using XTAL 26MHz:
`NodeMCU ESP32S`
Try to pick NodeMCU from boards.

Nagymadar, 2020-02-03:
I've tried `NodeMCU-32S` board. No change. But I've looked it up: Clock Speed: 240 Mhz.

Interestingly when I changed back to ESP Dev board, it worked. Then put in 2 Serial.println("Done") lines just to change the code, and fails again! I removed them, works. Changed back and forth, not it does not work anyome. I am really confused. Once one compilation works, even after reset and 20+ scans. But some compilation just does not work. I 've checked the Arduino comp messages. The BLE library compilation sequence is a bit different. That is all. Any idea?

chegewara, 2020-02-03:
I dont have explanation for that. What you can do is to see if you are not short on heap, or experiment with different boards selected in arduino.
Thats good you found it somehow related to hardware and board selected, but heap shortage is last thing i can think of.

Nagymadar, 2020-02-05:
I have checked 3 other boards:
1. Heltec Lora V2
2. Ai Thinker ESP32-CAM
3. TTGO T-Journal ESP32 CAM/OLED

All worked with ESP Dev board settings. But... 
Interestingly the Heltec did not work with its own board settings, only with ESP Dev.
The TTGO failed after programming, but after a manual reset it started working:

```
*** Restart finished. Scanning...
***   Device: Name: SENSOR_PRO, Address: ed:3d:9a:96:86:a3, serviceUUID: 6a800001-b5a3-f393-e0a9-e50e24dcca9e 
***   Payload [33] = { 0B 09 53 45 4E 53 4F 52 5F 50 52 4F 02 01 06 11 07 9E CA DC 24 0E E5 A9 E0 93 F3 A3 B5 01 00 80 6A }
*** Devices found: 1
*** Heap: 187860
*** Scan done!

***   Device: Name: SENSOR_PRO, Address: e5:89:06:ef:6d:e8 
***   Payload [15] = { 0B 09 53 45 4E 53 4F 52 5F 50 52 4F 02 01 06 }
*** Devices found: 1
*** Heap: 187728
*** Scan done!

***   Device: Name: SENSOR_PRO, Address: ed:3d:9a:96:86:a3 
***   Payload [15] = { 0B 09 53 45 4E 53 4F 52 5F 50 52 4F 02 01 06 }
*** Devices found: 1
*** Heap: 187728
*** Scan done!

_(Reset Button pressed...)_
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
*** Restart finished. Scanning...
***   Device: Name: SENSOR_PRO, Address: ed:3d:9a:96:86:a3, serviceUUID: 6a800001-b5a3-f393-e0a9-e50e24dcca9e 
***   Payload [33] = { 0B 09 53 45 4E 53 4F 52 5F 50 52 4F 02 01 06 11 07 9E CA DC 24 0E E5 A9 E0 93 F3 A3 B5 01 00 80 6A }
*** Devices found: 1
*** Heap: 187860
*** Scan done!

***   Device: Name: SENSOR_PRO, Address: ed:3d:9a:96:86:a3, serviceUUID: 6a800001-b5a3-f393-e0a9-e50e24dcca9e 
***   Payload [33] = { 0B 09 53 45 4E 53 4F 52 5F 50 52 4F 02 01 06 11 07 9E CA DC 24 0E E5 A9 E0 93 F3 A3 B5 01 00 80 6A }
*** Devices found: 1
*** Heap: 187860
*** Scan done!
```

chegewara, 2020-02-05:
So, this looks like esp-idf or in arduino-esp32 boards.txt problem.
@me-no-dev

Nagymadar, 2020-02-06:
Might be. I still not understand why it does work in some cases and don't in the other. As always, this is the worst type of bug ever. :-D I will keep you informed when I find out something.

OersHunorDetre, 2020-02-10:
I found out that all of the can boards can fail, even in ESP Dev board settings. When I restart the ESP it can fail from or work randomly. In other words, when it works for 2 sans, then it keeps working forever until I reboot. When it fails the second scan, it keeps failing until I reboot. I cannot see any pattern unfortunately.

PatrikOO, 2020-02-25:
Hi guys,
does anybody has datasheet for SENSOR_PRO device? I am trying to read out temperature from it but no luck. IT has newer BMP280 chip in difference to older version with BMP180.

Looks that I am able to receive temperature data with this settings:
GATT_SERVICE = "6a800001-b5a3-f393-e0a9-e50e24dcca9e" in old sensor it was  "6e400001-b5a3-f393-e0a9-e50e24dcca9e"

    GATT_TEMP_CHARACTERISTICS = "6a80b280-b5a3-f393-e0a9-e50e24dcca9e" in old version it was "6e400005-b5a3-f393-e0a9-e50e24dcca9e"
    SENSOR_DESCRIPTOR = "00002902-0000-1000-8000-00805f9b34fb"

but no idea how to decode it.

Nagymadar, 2020-02-25:
I found out that all of the can boards can fail, even in ESP Dev board settings. When I restart the ESP it can fail from or work randomly. In other words, when it works for 2 sans, then it keeps working forever until I reboot. When it fails the second scan, it keeps failing until I reboot. I cannot see any pattern unfortunately.

Nagymadar, 2020-02-25:
Hi Patrik, this is what I found out on my device:

```
/* nRF51 Sensor tag */
//BLEUUID SENSOR_PRO_serviceUUID("6a800001-b5a3-f393-e0a9-e50e24dcca9e"); // The remote service we wish to connect to.
//BLEUUID SENSOR_PRO_tempUUID   ("6a80b280-b5a3-f393-e0a9-e50e24dcca9e"); // The characteristic of the remote service we are interested in.

/*
                           00 01 137  85 255 255 255 254  | -2C, at 80m height
                           00 01 137  72   0   0   0   4  | +4C, at 80m height
SENSOR_PRO_tempUUID value: 00 01 109 195   0   0   0  25  | +25C, ~1001,8 hPa at 560m height
                            |  |   |   |               |
                            |  |   |   |               temperature (*1 = C)
                            |  |   |   |
                            |  |   |   |
                            |  |   |   | 
                            |  |   |   humidity/4 in %?
                            |  |   pressure
                            |  ?
                            ?
 */

//Temperature conversion:
int temperature = (inputbyte[4] << 24) | (inputbyte[5] << 16) | (inputbyte[6] << 8) | (inputbyte[7]);
```

Let me know if you find out more.

closed 2020-05-09 by stale[bot]

luisgodelmo, 2020-06-01:
Hi, I think I had exactly the same problem: first scan was always different from following scans (from the 2nd scan it ommited name & manufacturer), which made me impossible to get desired data from 3 temp/hum IBS TH1 Plus Inkbird sensors. 
I thought it might be related to the fact that, after resetting ESP32, the sensors were unknown for the board, but for the 2nd scan something changed so that the data sent by sensor/received by ESP32 was different --> maybe if I restarted BLE at each loop, every scan would be "from scratch" so that sensors would be always "new". I'm not sure if that was actually the problem or what, but SOLVED IT. Only added these init & deinit statements in void loop. 

void loop() {
  BLEDevice::init("myBLEScan");
  client.loop();
  if (cnt == 0) { bleScan(); }

  cnt=cnt+1;
  
  if (cnt >= 5) { cnt=0; }

  client.loop();
  delay(5000); // or 5s
  BLEDevice::deinit(false);
}

Now some expert could tell me why was this happening and why this solved things out... 
Also maybe this is not the most elegant solution once we know the problem source.

Nagymadar, 2020-06-16:
I use this BLE fork instead. Does not solve this problem, but generally more stable:
https://github.com/wakwak-koba/arduino-esp32/tree/master/libraries/BLE

Might this be a solution? Can you try @luisgodelmo ?
https://github.com/wakwak-koba/NimBLE-Arduino

unisim1983, 2020-12-04:
For me the solution was:
in the loop:

void loop() {
  BLEDevice::init("abc");
  BLEScanResults foundDevices = pBLEScan->start(scanTime, false);
  pBLEScan->clearResults();   // delete results fromBLEScan buffer to release memory
  BLEDevice::deinit(false);
}

hohmannc85, 2024-06-26:
> I use this BLE fork instead. Does not solve this problem, but generally more stable: https://github.com/wakwak-koba/arduino-esp32/tree/master/libraries/BLE
> 
> Might this be a solution? Can you try @luisgodelmo ? https://github.com/wakwak-koba/NimBLE-Arduino

@Nagymadar ist getName there working?