espressif/arduino-esp32 #3678: Zero Values from MPU6050

Sebastiangka, 2020-01-27:
### Hardware:
Board:							NodeMCU-32s
IDE name:						Arduino IDE 1.8.10
Flash Frequency:					80MHz and 40Mhz
PSRAM enabled:                                      no
Upload Speed:					115200
Computer OS:                                         Windows 10

### Description:
After having programmed the MPU6050 on the Arduino Leonardo, I changed my board to a NodeMCU 32S. Surprise...my Sketch didn't work anymore. I made some changes and it runs, but not perfectly.

My problem is that I only get zero Values from the registers (ACC X, Y, Z). For the moment I only want to get the raw data from the accelerometer.
I checked the code but I don't get wants wrong or what am I missing. I would be very grateful if someone could give some hint/advice.

### What I tried:
Baudrate 921600  and 115200
Flash frequency 40 and 80 MHZ
Different scale for accelerometer
Wire.begin with 4Hz, 1kHz, 100kHz

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Wire.h>
#define BaudRate 115200
#define Acc_Sensitivity 16384.00 
#define LED_B   0
#define LED_G   2 
#define LED_R   4
#define SDApin  21    //IO 21 SDA
#define SCLpin  22    //IO 22 SCL

long accl_X;
long accl_Y; 
long accl_Z;

bool blinkState = false;
uint32_t tStart;
float Timer;
uint32_t period = 360*1000L; // timer = 1 Stunde für die Daten lesen. Kann geändert werden.
void setup() {
  
  setupMPU6050(); // Aufrufen der Funktion setupMPU6050 
  Serial.begin(BaudRate); 
  while (!Serial)
    {
    ; // warten auf Connection. Nur USB-Port 
    }
  pinMode(LED_R, OUTPUT);
  pinMode(LED_G, OUTPUT);
  pinMode(LED_B, OUTPUT);
  Wire.begin(SDApin, SCLpin);//, 100000);
  
}

void loop() {
  recordRegister();
  blinkState = !blinkState;
  digitalWrite(LED_G, blinkState);
  PrintData();
  delay(50);

}

void setupMPU6050()
{
  Wire.beginTransmission(0b1101000);
  Wire.write(0x6B);
  Wire.write(0b00000000);           //MPU wake up
  Wire.endTransmission();
  
  Wire.beginTransmission(0b1101000);
  Wire.write(0x1B);                 //Gyro
  Wire.write(0b00000000);
  Wire.endTransmission();
  
  Wire.beginTransmission (0b1101000); 
  Wire.write(0x1C);                 //Acc
  Wire.write(0b00000000); // setzen Beschleunigungssensor auf Skalierung +-2g 
  //Wire.write(0b00000010);   //Skalierung des Sensor auf +- 8g 
  //Wire.write(0b00011000); // voll Skalierung +-16g 
  Wire.endTransmission ();
}
void recordRegister()
{
  Wire.beginTransmission (0b1101000); // I2C Adresse 
  Wire.write(0x3B); // setzen Register 3B für erste Beschleunigung auf Lesen  
  Wire.endTransmission ();
  Wire.requestFrom(0b1101000,6); //Beschleunigung 6 Register (von 3B bis 40) 

  //Rohdaten Beschleunigungssensor 
  while (Wire.available ()<6);
    accl_X = Wire.read()<<8 | Wire.read(); // 0x3B (ACCEL_XOUT_H) und 0x3C (ACCEL_XOUT_L) 
    accl_Y = Wire.read()<<8 | Wire.read(); // 0x3D (ACCEL_YOUT_H) und 0x3E (ACCEL_YOUT_L) 
    accl_Z = Wire.read()<<8 | Wire.read(); // 0x3F (ACCEL_ZOUT_H) und 0x40 (ACCEL_ZOUT_L)

  accl_X = accl_X/Acc_Sensitivity;
  accl_Y = accl_Y/Acc_Sensitivity;
  accl_Z = accl_Z/Acc_Sensitivity;
  delay(50);
}

void PrintData()
{
  Serial.print("| accl_X: ");
  Serial.print(accl_X);
  Serial.print(" g");
  Serial.print("\t");

  Serial.print("| accl_Y ");
  Serial.print(accl_Y);
  Serial.print(" g");
  Serial.print("\t");

  Serial.print("| accl_Z ");
  Serial.print(accl_Z);
  Serial.print(" g");
  Serial.print("\n");
}
```
[image]

stickbreaker, 2020-01-27:
Check the `Wire()` error return code.
Also, you need to configure `Wire()` before you use it.
```c++
//Change the code below by your sketch
#include <Wire.h>
#define BaudRate 115200
#define Acc_Sensitivity 16384.00 
#define LED_B   0
#define LED_G   2 
#define LED_R   4
#define SDApin  21    //IO 21 SDA
#define SCLpin  22    //IO 22 SCL

long accl_X;
long accl_Y; 
long accl_Z;

bool blinkState = false;
uint32_t tStart;
float Timer;
uint32_t period = 360*1000L; // timer = 1 Stunde für die Daten lesen. Kann geändert werden.
void setup() {
  Wire.begin(SDApin, SCLpin);//, 100000); // configure before use.
  
  setupMPU6050(); // Aufrufen der Funktion setupMPU6050 
  Serial.begin(BaudRate); 
  while (!Serial)
    {
    ; // warten auf Connection. Nur USB-Port 
    }
  pinMode(LED_R, OUTPUT);
  pinMode(LED_G, OUTPUT);
  pinMode(LED_B, OUTPUT);
}

void loop() {
  recordRegister();
  blinkState = !blinkState;
  digitalWrite(LED_G, blinkState);
  PrintData();
  delay(50);
}

void setupMPU6050()
{
  Wire.beginTransmission(0b1101000);
  Wire.write(0x6B);
  Wire.write(0b00000000);           //MPU wake up
  uint8_t err= Wire.endTransmission();
  if (err != 0) {
     Serial.printf("MPU wakeup Fail, err=%d(%s)", err, Wire.getErrorText(err));
	 }
  Wire.beginTransmission(0b1101000);
  Wire.write(0x1B);                 //Gyro
  Wire.write(0b00000000);
  err = Wire.endTransmission();
  if (err != 0) {
     Serial.printf("Gyro wakeup Fail, err=%d(%s)", err, Wire.getErrorText(err));
	 }
  
  Wire.beginTransmission (0b1101000); 
  Wire.write(0x1C);                 //Acc
  Wire.write(0b00000000); // setzen Beschleunigungssensor auf Skalierung +-2g 
  //Wire.write(0b00000010);   //Skalierung des Sensor auf +- 8g 
  //Wire.write(0b00011000); // voll Skalierung +-16g 
  err = Wire.endTransmission ();
  if (err != 0) {
     Serial.printf("Acc configure Fail, err=%d(%s)", err, Wire.getErrorText(err));
	 }
 
}
void recordRegister()
{
  Wire.beginTransmission (0b1101000); // I2C Adresse 
  Wire.write(0x3B); // setzen Register 3B für erste Beschleunigung auf Lesen  
  uint8_t err =Wire.endTransmission ();
  if(err != 0) {
     Serial.printf("set Accel register fail, err=%d(%s)", err ,Wire.getErrorText(err));
	 }
 
  Wire.requestFrom(0b1101000,6); //Beschleunigung 6 Register (von 3B bis 40) 
  err = Wire.lastError();
  if( err != 0 ){
     Serial.printf("Read accel registers Fail, err=%d(%s)", err, Wire.getErrorText(err));
	 }
 
  //Rohdaten Beschleunigungssensor 
  // while (Wire.available ()<6); // available() will never increase, requestfrom() sets it on exit
  
  if(Wire.available() == 6) {
    accl_X = Wire.read()<<8 | Wire.read(); // 0x3B (ACCEL_XOUT_H) und 0x3C (ACCEL_XOUT_L) 
    accl_Y = Wire.read()<<8 | Wire.read(); // 0x3D (ACCEL_YOUT_H) und 0x3E (ACCEL_YOUT_L) 
    accl_Z = Wire.read()<<8 | Wire.read(); // 0x3F (ACCEL_ZOUT_H) und 0x40 (ACCEL_ZOUT_L)
  }

  accl_X = accl_X/Acc_Sensitivity;
  accl_Y = accl_Y/Acc_Sensitivity;
  accl_Z = accl_Z/Acc_Sensitivity;
  delay(50);
}

void PrintData()
{
  Serial.print("| accl_X: ");
  Serial.print(accl_X);
  Serial.print(" g");
  Serial.print("\t");

  Serial.print("| accl_Y ");
  Serial.print(accl_Y);
  Serial.print(" g");
  Serial.print("\t");

  Serial.print("| accl_Z ");
  Serial.print(accl_Z);
  Serial.print(" g");
  Serial.print("\n");
}
```

Chuck.

Sebastiangka, 2020-01-28:
There is no error return code. I got already some values but they are not right. Here a pic when I leave the MPU laying on a surface.
Thanks for your time Chuck
Greetings
Sebastian
[image]

Sebastiangka, 2020-01-28:
Ok I changed the print-function and looks much better. But Im still afraid about the wire() configuration. 

[image]

stickbreaker, 2020-01-28:
Wire config is simple. `Wire.begin(SDApin,SCLpin, busSpeed);`
`busSpeed` is limited by CPUclock speed, with a 240MHz CPUclock, minimum bus is about 10k, maximum is about 24MHz, but the I2C bus's max speed is hardware dependent, I2C is spec'd at 400KHz, to get the bus to operate correctly at 400KHz you need to use 2.4k ohm pullups or perhaps 1.2K.  The pullup resistance value is dependent on bus capacitance. See this wiki entry on my repo [Pullup considerations](https://github.com/stickbreaker/arduino-esp32/wiki/Pullup-Values)

A limitation with the ESP32 on I2C is that the maximum SCL clock stretch interval is 13.1ms with a 240MHz cpu clock them MPU6050 can exceed this, the `Wire()` object will return error code 2:Timeout. you have to work around this.

Chuck.

Sebastiangka, 2020-01-28:
That wiki helped me a lot to understand whats going on. Tomorrow i will have the chance to take those pullups in my circuit. I´m excited about the results to compare the value at max bus speed with/without the pullups.
Thanks

Sebastiangka, 2020-01-31:
Chuck, I saw some improvements! had to change other stuffs in my code and now it´s working. Thanks!

closed 2020-01-31 by stickbreaker

HkRoni, 2022-08-17:
What finally worked? Can you share the final code??

Shashaank1453, 2022-12-27:
> What finally worked? Can you share the final code??

yes please share the final code