espressif/arduino-esp32 #3685: what is meant by Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited) ?

yogeshwaran774, 2020-01-29:
### Hardware:
Board:							?ESP32 Dev Module? ?node32? ?ttgo_lora?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
PSRAM enabled:                                          ?no?
Upload Speed:						?115200?
Computer OS:                                            ?Windows 7

### Description:
I am working in Decawave localisation project and my whole project was working fine before i add an timer and call a function to transmit data.

after transmitting i am getting Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited) .
basically i am not able to understand what is InstrFetchProhibited and how to find which instruction is causing the issue by using Backtrace values of ESP32

### Sketch:  
void transmit(uint8_t packt_type, uint8_t pack_id, uint32_t dest_id, uint8_t payload[], uint8_t payload_size)
{

// it is executing all instruction inside this funciton
}

void Client_indicate()
{
  // transmit() fucntion called
}
void handleRange()
{
  clientRequest = timerBegin(1, 80, true);                  //timer 0, div 80
  timerAttachInterrupt(clientRequest, &Client_indicate, true);  //attach callback
  timerAlarmWrite(clientRequest, 5000 * 1000, true); //set time in us
  timerAlarmEnable(clientRequest);
}

void setup()
{

}
void loop()
{
 handleRange();
}
### Debug Messages:
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x00000004  PS      : 0x00050233  A0      : 0x00000004  A1      : 0x3ffbe700  
A2      : 0x3ffbe740  A3      : 0x3ffc1000  A4      : 0x3f401585  A5      : 0x00000001  
A6      : 0x00000000  A7      : 0x00000008  A8      : 0x00000001  A9      : 0x800811f0  
A10     : 0x3ffbe760  A11     : 0x3ffc1000  A12     : 0x00000003  A13     : 0x00000001  
A14     : 0x00000000  A15     : 0x00000008  SAR     : 0x00000001  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000004  LBEG    : 0x3ffc1000  LEND    : 0x20000000  LCOUNT  : 0x3ffc0fdc  

Backtrace: 0x00000004:0x3ffbe700 0x00000001:0x3ffbe720 0x400d350c:0x3ffbe740 0x400d9a59:0x3ffbe760 0x400811ed:0x3ffbe780 0x40081595:0x3ffbe7a0 0x40081abd:0x3ffbe7c0 0x40083895:0x3ffb1f40 0x400814af:0x3ffb1f60 0x400d4205:0x3ffb1f80 0x400da1e5:0x3ffb1fb0 0x40088ea5:0x3ffb1fd0

Decoding stack results
0x400d350c: transmit(unsigned char, unsigned char, unsigned int, unsigned char*, unsigned char) at C:\Users\jilka\AppData\Local\Temp\arduino_build_432229\sketch/transmit.h line 148

0x400d9a59: Print::println(char const*) at C:\Users\jilka\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\cores\esp32\Print.cpp line 176

0x400811ed: Client_indicate() at C:\Users\jilka\AppData\Local\Temp\arduino_build_432229\sketch/receive.h line 117

0x40081595: __timerISR at C:\Users\jilka\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\cores\esp32\esp32-hal-timer.c line 88

0x40083895: timer_alarm_isr at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/esp_timer_esp32.c line 274

0x400814af: millis at C:\Users\jilka\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\cores\esp32\esp32-hal-misc.c line 139

0x400d4205: loop() at \\JILKA-PC\Users\jilka\Hacklab-master\Hacklab-master\Products\P003-F-MTMC\Firmware\CAS\CAS_PAS_UPGRADE_V1_ESP/CAS_PAS_UPGRADE_V1_ESP.ino line 392

0x400da1e5: loopTask(void*) at C:\Users\jilka\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\cores\esp32\main.cpp line 19

0x40088ea5: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143

me-no-dev, 2020-01-29:
Please take some care on how you format your issue. There are particular ways to paste code and text blocks. With that said, you are constantly calling the timer init functions in the loop... over and over. Those should be in `setup()` not in `loop()`.

closed 2020-01-29 by me-no-dev

yogeshwaran774, 2020-01-29:
issue is not solved yet and i am calling only once at a time in loop by putting one bool flag. sorry i didn't mention that part in the above code. So i am pretty sure i am initializing timer only once.

can tell where my code got crashed by using Backtrace.

me-no-dev, 2020-01-29:
so you posted partial code and the error is in the missing part :) last called function was transmit

yogeshwaran774, 2020-01-29:
> so you posted partial code and the error is in the missing part :) last called function was transmit

YES id didn't and i can send my arduino code to your mail id just send a mail to yogeshwaran774@gmail.com or i can explain once again here, the handleRange() is called whenever range packet comes and inside i am initiating timer of around 5 sec after that my device will again start communicate and gets the range packet again in between 5 sec but here i am not initiating the timer again.

whenever the timer shoots Client_indicate() is called and inside transmit() function is called. The transmit function dose it task properly and getting out. As soon as the transmit function ends ESP32 is throwing an error Guru Meditation Error: Core 1 panic'ed (InstrFetchProhibited). I decoded the Backtrace and i found Decoding stack results
0x400d350c: transmit(unsigned char, unsigned char, unsigned int, unsigned char*, unsigned char) at C:\Users\jilka\AppData\Local\Temp\arduino_build_432229\sketch/transmit.h line 148

As far us my understanding the decode trace says that at transmit function esp32 faced some error or exception but by looking at my serial print the transmit function was already completed. I am sorry if my understanding was wrong and if possible tell me how to understand the backtrace of ESP32 or any link which related to backtrace also ok for me.

yogeshwaran774, 2020-01-29:
I think the issue resolved. How it was you know i removed yield() which is present in the transmit() function. I don't what exactly happening and how the error related to yield(). Right now i kept for testing soon i tell you the exact solution and i try to find some resources regarding yield() and interrupts of ESP32.