espressif/arduino-esp32 #3686: How to recover from BEACON_TIMEOUT, STA_DISCONNECTED and STA_LOST_IP

GeorgeFlorian, 2020-01-29:
OS: Linux Mint 19.1
Board: ESP32 EVB

I have a server that sends data through a POST request to the ESP32. I am using ESPAsyncWebServer to handle it:
```
  server.on("/post", HTTP_POST, [](AsyncWebServerRequest * request){
      if(request->hasArg("number")){
          String arg = request->arg("number");
          Serial.print("The number is: ");
          Serial.println(arg);
      } else {
          Serial.println("Post did not have a 'number' field.");
      }
      request->send(200);
  });
``` 
The issue is that if the server disconnects the ESP hangs, then disconnects from the WiFi and returns the following:
```
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 200 - BEACON_TIMEOUT
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 4 - ASSOC_EXPIRE
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 8 - STA_LOST_IP
```
and stays like this. I have seen it reconnect only one time.  
How can I force the ESP to reconnect to the WiFi ?

CelliesProjects, 2020-01-29:
Reconnect does not work properly.
There are work-arounds.

I use something like
````c++
WiFi.onEvent( WiFiEvent );
WiFi.begin();
...

void WiFiEvent( WiFiEvent_t event ) {
  switch ( event ) {
    case SYSTEM_EVENT_AP_START:
      ESP_LOGI( TAG, "AP Started");
      //WiFi.softAPsetHostname(AP_SSID);
      break;
    case SYSTEM_EVENT_AP_STOP:
      ESP_LOGI( TAG, "AP Stopped");
      break;
    case SYSTEM_EVENT_STA_START:
      ESP_LOGI( TAG, "STA Started");
      //WiFi.setHostname( DEFAULT_HOSTNAME_PREFIX.c_str( );
      break;
    case SYSTEM_EVENT_STA_CONNECTED:
      ESP_LOGI( TAG, "STA Connected");
      //WiFi.enableIpV6();
      break;
    case SYSTEM_EVENT_AP_STA_GOT_IP6:
      ESP_LOGI( TAG, "STA IPv6: ");
      //ESP_LOGI( TAG, "%s", WiFi.localIPv6().toString());
      break;
    case SYSTEM_EVENT_STA_GOT_IP:
      //ESP_LOGI( TAG, "STA IPv4: ");
      //ESP_LOGI( TAG, "%s", WiFi.localIP());
      break;
    case SYSTEM_EVENT_STA_DISCONNECTED:
      ESP_LOGI( TAG, "STA Disconnected");
      WiFi.begin();
      break;
    case SYSTEM_EVENT_STA_STOP:
      ESP_LOGI( TAG, "STA Stopped");
      break;
    default:
      break;
  }
}
````

GeorgeFlorian, 2020-01-29:
@CelliesProjects What is `ESP_LOGI` ?

CelliesProjects, 2020-01-29:
>@CelliesProjects What is ESP_LOGI ? 

https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/log.html

Control this by setting `Tools > Core Debug Level` in Arduino IDE.
The advantage over `Serial.print()` is that if you compile for log level `None` there will be no logging code added to your binary. 
`Serial.print` statements produce code and bloat your binary.
Afaik not all esp32 boards support setting the loglevel.

>Doesn't this bit do nothing ?

It shows connection is lost on serial output. ( if loglevel is set to `Info` or higher )

CelliesProjects, 2020-01-29:
Alernative solution could be something like
````c++
if ( !WiFi.isConnected() ) WiFi.begin();
````

GeorgeFlorian, 2020-01-29:
So the ESP can't recover from a `SYSTEM_EVENT_STA_STOP`  ?

I think that this `if ( !WiFi.isConnected() ) WiFi.begin();` in `void loop()` can do wonders.

CelliesProjects, 2020-01-29:
The first suggestion does its work automagically in the background.
Second one has to be in a loop/task somewhere.

me-no-dev, 2020-01-29:
```ESP_LOGI``` is IDF function. It's not controlled by Arduino. What is controlled is ```log_i``

simap, 2020-02-22:
I think automatic reconnects for several causes were broken recently from this change:
https://github.com/espressif/arduino-esp32/pull/3085

simap, 2020-04-22:
@me-no-dev thoughts on this?

OmarAlkassab, 2020-05-28:
I have faced the same problem, any solution?

simap, 2020-08-20:
Is there anything I can do to help further this along or get a maintainer to take a look?

simap, 2020-10-20:
poke

Duolabs, 2020-11-01:
Hallo everyone.
I got stuck in the same point as you all.
After I created and closed 200 sockets the ESP32 Wroom looses the WiFi Connection and gets stuck there.
The log:
**[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED [W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 200 - BEACON_TIMEOUT abort() was called at PC 0x4015489b on core 0**

This happens every around 200-210 sockets connections:
Here is my simple routines which I call all of times:

```
int TCPClient_Send (const char * Domain, const char * Port, const char * Timeout, const char * Payload, char * TCPResult)
{
	int iResult = -1;
        struct addrinfo     hints;
	hints.ai_family     = AF_INET;
	//hints.ai_socktype   = SOCK_STREAM;
    	hints.ai_socktype   = SOCK_RAW;
    
    	struct addrinfo     *res;
    	struct in_addr      *addr;
    
    	iResult = getaddrinfo(Domain, Port, &hints, &res);
    	if((iResult != 0) || (res == NULL)) {
                        return -1;
        } 
        
        int s = socket(res->ai_family, res->ai_socktype, IP_PROTO_ICMP);
    	if(s < 0) {
      		Serial.println ("TCPClient_Send::TCPCLIENT_SOCKET_CREATION_ERROR");
      		return TCPCLIENT_SOCKET_CREATION_ERROR;
    	}

    	Serial.println ("TCPClient_Send::TCPSocket Created with fd= " + String(s));
          
    	// connect to the specified server
    	iResult = connect(s, res->ai_addr, res->ai_addrlen);
    	if(iResult != 0) {
      		Serial.println ("TCPClient_Send::TCPCLIENT_FAILED_TO_CONNECT");
      		closesocket(s);
      		return TCPCLIENT_FAILED_TO_CONNECT;
    	}

    	Serial.println ("TCPClient_Send::Closing Socket: " + String(s));
    	closesocket(s);
    	return 1;
}

```

Any help would be highly appreciated

simap, 2021-06-23:
[image]

VojtechBartoska, 2022-04-06:
Hello @simap and @GeorgeFlorian  , sorry for late reply. Are you able to test your issue on development version [2.0.3-RC1](https://github.com/espressif/arduino-esp32/releases/tag/2.0.3-RC1) to check if this is still valid? You can take a look on [Docs](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html) where is explained how to choose development release version in Arduino IDE.

closed 2022-04-21 by VojtechBartoska