espressif/arduino-esp32 #3696: ubuntu 20.04 Arduino IDE python error

minzukas, 2020-01-31:
So after upgrading from Ubuntu 18.04 LTS to 20.04 both Arduino IDE and Arduino Pro gives me:
"exec: "python": executable file not found in $PATH
Error compiling for board ESP32 Wrover Module" - same with all ESP32.

Tried reinstalling, currently  on 1.8.11. 
Read for two days and tried tons of different things, including guide:
https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/debian_ubuntu.md
https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/boards_manager.md

Python3 and bunch others installed.

Nothing helps...

WereCatf, 2020-01-31:
> So after upgrading from Ubuntu 18.04 LTS to 20.04 

Ubuntu 20.04 is not even in beta-phase yet, it's pretty pointless to be making bug-reports about arduino-esp32 against an early alpha-release of Ubuntu.

lbernstone, 2020-01-31:
What do you get for `which python`?

minzukas, 2020-02-01:
> What do you get for `which python`?

for sudo apt install python I get:

sudo apt install python
[sudo] password for mindaugas: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package python is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  python2-minimal:i386 python2:i386 python2-minimal python2 2to3

E: Package 'python' has no installation candidate

me-no-dev, 2020-02-01:
look for python3

lbernstone, 2020-02-01:
sounds like an ubuntu issue.  python2 is end of life.  So, they should be setting python3 as the default alternative.  At the command line, run `which python`.  If there is none, make a symlink /usr/bin/python -> /usr/bin/python3.

minzukas, 2020-02-01:
> sounds like an ubuntu issue. python2 is end of life. So, they should be setting python3 as the default alternative. At the command line, run `which python`. If there is none, make a symlink /usr/bin/python -> /usr/bin/python3.

Thank You! The problem was indeed with symlink, just had to move to local...
`mindaugas@mindaugas-pc:~$ which python3
/usr/bin/python3
mindaugas@mindaugas-pc:~$  ls -l /usr/local/bin/py*
-rwxr-xr-x 1 root root 2438 liep.  15  2019 /usr/local/bin/pydisasm.py
-rwxr-xr-x 1 root root  230 liep.  15  2019 /usr/local/bin/pydisassemble
mindaugas@mindaugas-pc:~$ ^C
mindaugas@mindaugas-pc:~$ sudo ln -sf /usr/bin/python3 /usr/local/bin/python
mindaugas@mindaugas-pc:~$ ls -l /usr/local/bin/py*
-rwxr-xr-x 1 root root 2438 liep.  15  2019 /usr/local/bin/pydisasm.py
-rwxr-xr-x 1 root root  230 liep.  15  2019 /usr/local/bin/pydisassemble
lrwxrwxrwx 1 root root   16 vas.    2 00:18 /usr/local/bin/python -> /usr/bin/python3
`

closed 2020-04-16 by stale[bot]

tbaran, 2020-06-24:
Had the same issue moving to Ubuntu 20.04. It looks like python2 support has been dropped and you cannot install python-pip anymore from apt sources. The fix is easy:

`$ sudo apt remove python-is-python2 
$ sudo apt install python-is-python3`

That will relink the default python version to python3

chepo92, 2020-09-14:
> Had the same issue moving to Ubuntu 20.04. It looks like python2 support has been dropped and you cannot install python-pip anymore from apt sources. The fix is easy:
>
> `$ sudo apt remove [...]

while that fixes the original error, now it throws:
`
ModuleNotFoundError: No module named 'serial'`

lbernstone, 2020-09-14:
sudo pip3 install pyserial

chegewara, 2020-09-14:
From experience my suggestion is to not use snap with arduino on ubuntu 20.04. I had so many problems, including not working exception decoder and eventually arduino ide stopped to work. When i switched to portable version i am happy now.