espressif/arduino-esp32 #3702: Sketch too big  -- Error compiling for board SparkFun ESP32 Thing.

drmpf, 2020-02-03:
Why does the Arduino ESP32 support not allow the default partition size to be increased

What you are trying to do?   Increase the available program space
Describe your system( Hardware, computer, O/S, core version, environment).  Windows 7, Arduino 1.8.9,  ESP32 support 1.0.2
Describe what is failing.  Upload not run when programs exceeds 1310720 bytes and the default partition selected.
Show the shortest possible code that will duplicate the error.  Write a large program say WiFiSecureClient and AsyncWebServer so that program exceeds 1310720 bytes.
Use the Partition editor to remove app1 and increase size of app0
Genarate csv/bin  (updates default.csv and default.bin)
Show the EXACT error message(it doesn't work is not enough).
Sketch uses 1337334 bytes (102%) of program storage space. Maximum is 1310720 bytes.
Global variables use 61608 bytes (18%) of dynamic memory, leaving 266072 bytes for local variables. Maximum is 327680 bytes.
Sketch too big; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing it.
Error compiling for board SparkFun ESP32 Thing.

### Hardware:
Board:							SparkFun ESP32 Thing.
Core Installation version:			1.0.2
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    ?
Upload Speed:					921600
Computer OS:                                        Windows 7

### Description:
Describe your problem here
see above

A work around is to select a non-default partition setting like "no OTA", the build/upload process then completes.
I noticed a possible commit to allow partition.csv file in the sketch dir
9ef3e2d
but that did not work for me using version 1.0.2

Tip. You need to reload you SPIFFS after a partition change.

chegewara, 2020-02-04:
You can also have partitions.csv in your project folder with customized partitions. This will be used instead.

drmpf, 2020-02-04:
**partitions.csv not honored in Arduino ESP32 package V1.0.4**

Test.  (using ESP32 arduino addon V1.0.4)
Select no OTA schema in Arduino tools menu, check compile works
copy no OTA.csv partition schema from  ...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\tools  to the sketch directory  (i.e. where the .ino file is located)
rename to partitions.csv
Select 'default' schema in Arduino tools menu 
Recompile
Expected result  --  partitions.csv to override any selected schema in tools menu
Actual result 
_Sketch uses 1335557 bytes (101%) of program storage space. Maximum is 1310720 bytes.
Global variables use 62424 bytes (19%) of dynamic memory, leaving 265256 bytes for local variables. Maximum is 327680 bytes.
Sketch too big; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing it.
Error compiling for board SparkFun ESP32 Thing._

atanisoft, 2020-02-05:
> partitions.csv not honored in Arduino ESP32 package V1.0.4

Can you confirm that your copy of platform.txt for your "1.0.4" install has [these](https://github.com/espressif/arduino-esp32/blob/master/platform.txt#L66-L69) lines in it? If these lines are present then I suspect something else is preventing your Arduino IDE from picking up this updated version. You might need to purge all versions of arduino-esp32 from your AppData\Local\Arduino15\ tree and reinstall via boards manager.

drmpf, 2020-02-05:
Still fails see detailed tests below.  Latest platform.txt is not installed by installing V1.0.4 and using the platform.txt referenced above fails with esp32.project.ld but not found at the link step

==================
Tests in detail

Open Arduino IDE 1.8.9,  remove ESP32 support, close Arduino IDE  (deleted esp32 dir)
Open Arduino IDE, install ESP32 support V1.0.4
the above lines are still missing from platform.txt
first line of platform.txt installed for V1.0.4 is
```
name=ESP32 Arduino
version=

```your platform.txt has 
```
name=ESP32 Arduino
version=0.0.1
```
Copied your platform.txt over the installed on and compiled with a partition.csv in the sketch dir
got
```
cmd /c if exist "....\\Documents\\Arduino\\ESP32_Goblot_bb\\partitions.csv" copy /y "....\\Documents\\Arduino\\ESP32_Goblot_bb\\partitions.csv" "....\\AppData\\Local\\Temp\\arduino_build_7448\\partitions.csv"
cmd /c if not exist "....\\AppData\\Local\\Temp\\arduino_build_7448\\partitions.csv" copy "....\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4\\tools\\partitions\\default.csv" "....\\AppData\\Local\\Temp\\arduino_build_7448\\partitions.csv"
        1 file(s) copied.
```
but link failed with 
`.../appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot open linker script file esp32.project.ld: No such file or directory
`
Download zip of github https://github.com/espressif/arduino-esp32 and seached for 
esp32.project.ld but not found.

removed ESP32 package from Arduino IDE ,  **AND**,  removed all esp32 package zips from staging/packages (including xtensa-esp32... )  to force a full re-download
Re-installed ESP32 1.0.4  (took much longer this time, downloaded esp32 and xtensa)

Old platform.txt still installed, copied platform.txt from github as above
got same error missing esp32.project.ld

atanisoft, 2020-02-05:
How are you installing v1.0.4?

drmpf, 2020-02-05:
Through the Arduino IDE board Manager  
Open tools->board->board Manager
enter ESP32 into search bar
click install button for ESP32 1.0.4

lbernstone, 2020-02-05:
Use [this version](https://github.com/espressif/arduino-esp32/blob/9ef3e2d2a6b79c951f894dba1dd71c811317ca6e/platform.txt) of platform.txt instead of the latest.

drmpf, 2020-02-05:
That version give the error

exec: ".....\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/esptool/esptool.exe": file does not exist
Error compiling for board SparkFun ESP32 Thing.

drmpf, 2020-04-05:
An actual fix would be appreciated, rather then just closing and ignoring.

closed 2020-06-18 by stale[bot]