espressif/arduino-esp32 #3704: rtc ds 3231 starts giving wrong time after some time with esp32 wrover-b using freertos

barkur24shubham, 2020-02-03:
### Hardware:
Board:							ESP32 wrover-b
IDE name:					Arduino IDE 1.8.11
Flash Frequency:					80Mhz
PSRAM enabled:                                          no
Upload Speed:					921600
Computer OS:                                           Windows 10

hi,
I m using three sensors on i2c and rtc ds3231 ,to divide tasks I m using freertos .using esp32 in AP mode, and also in station mode so it sends weather information to my website and a webpage by AP mode, but my rtc starts showing wrong time after some time which is out of time format.

Sorry for my mistakes i m very new with this things and this is my first post.

Thanks and regards.

//below is code(not complete code)

#include "soc/timer_group_struct.h"
#include "soc/timer_group_reg.h"
#include <WebServer.h>
#include <WiFiAP.h>
#include "RTClib.h"

String  Time()
{ 
String datetime="";  
delay(500);  
now = rtc.now();  

if(now.month()<10)
{MM="0"+String(now.month());}
else{MM=String(now.month());}
if(now.day()<10)

{DD="0"+String(now.day());}
else{DD=String(now.day());}
YY=String(now.year()%100);

if(now.hour()<10)
{hh="0"+String((now.hour()));}
else{hh=String((now.hour()));}
if(now.minute()<10)
{mm="0"+String((now.minute()));}
else{mm=String((now.minute()));}

if(now.second()<10)
{ss="0"+String(now.second());}
else{ss=String(now.second());}
  
  
datetime=String(YY+MM+DD+hh+mm+ss); 

SerialMon.println(datetime);
delay(500);

time1=hh+":"+mm+":"+ss;

return datetime;
}

 void setup()
{

     pinMode(MODEM_PWKEY, OUTPUT);
    pinMode(MODEM_RST, OUTPUT);
    pinMode(MODEM_POWER_ON, OUTPUT);
  //pinMode(34,INPUT);
  //Wire.begin(21,22); 
  SerialMon.begin(115200);
  SerialAT.begin(115200, SERIAL_8N1, MODEM_RX, MODEM_TX);
  
  delay(3000);
  SerialMon.println("REBOOT");
  //Wire.setClockStretchLimit(1000);
  pinMode(12,OUTPUT);
  pinMode(13,OUTPUT);
    pinMode(smoke,INPUT);
 
  WiFi.softAP(ssid, password);
  IPAddress myIP = WiFi.softAPIP();
  SerialMon.print("AP IP address: ");
  SerialMon.println(myIP);
  server.begin();

  SerialMon.println("Server started");

  modempower(true);
 
  
 
  // Initialize the sensor (it is important to get calibration values stored on the device).
 
  if (pressure.begin())
    SerialMon.println("BMP180 init success");
  else
  {
    SerialMon.println("BMP180 init fail\n\n");
    while(1); // Pause forever.
  }

  SerialMon.println("above html in setup");
   iaqcore.begin();

    delay(500);
    if (! rtc.begin()) 
  {
    SerialMon.println("Couldn't find RTC Module");
    while (1);
  }
  delay(1000);

 if (rtc.lostPower()) 
  {
    SerialMon.println("RTC lost power, lets set the time!");
    rtc.adjust(DateTime(F(__DATE__), F(__TIME__)));
  }
  rtc.adjust(DateTime(F(__DATE__), F(__TIME__)));
  
//Time();
Serial.println(datetime1);

   xTaskCreatePinnedToCore(
                    Task1code,   /* Task function. */
                    "Task1",     /* name of task. */
                    100000,       /* Stack size of task */
                    NULL,        /* parameter of the task */
                    1,           /* priority of the task */
                    &Task1,      /* Task handle to keep track of created task */
                    0);          /* pin task to core 0 */                  
  delay(500); 

   xTaskCreatePinnedToCore(
                    Task2code,   /* Task function. */
                    "Task2",     /* name of task. */
                    10000,       /* Stack size of task */
                    NULL,        /* parameter of the task */
                    2,           /* priority of the task */
                    &Task2,      /* Task handle to keep track of created task */
                    1);          /* pin task to core 0 */                  
  delay(500); 

 

  
  
}

void Task1code( void * pvParameters ){
  Serial.print("Task1 running on core ");
  Serial.println(xPortGetCoreID());

  for(;;){
   
     

SerialMon.println(mod);  
if(mod==1)
{

 
  digitalWrite(12, HIGH);
  digitalWrite(13, LOW);
  modempower(0);
  flag1=false;
    Serial.println("in wifi mode");
    output27State = "Wifi";
    wifi_mode();
    
   
}
else if(mod==0)
{
  
 digitalWrite(13, HIGH);
 digitalWrite(12, LOW); 
 output27State = "GSM";
 
  flag1=false;
   Serial.println("in gsm mode ");   
  gsm_mode();
}
}
 }

void Task2code( void * pvParameters ){
  Serial.print("Task2 running on core ");
  Serial.println(xPortGetCoreID());

  for(;;){
  while(WiFi.softAPgetStationNum()>0)
  {page();}
  
}
}
void loop()
{

TIMERG0.wdt_wprotect=TIMG_WDT_WKEY_VALUE;
TIMERG0.wdt_feed=1;
TIMERG0.wdt_wprotect=0;

}

stickbreaker, 2020-02-03:
@barkur24shubham `Wire()` is single threaded, It can only be used by one thread at a time. 

I need to know exactly what "wrong time" means before I could have any chance of guessing what your problem is.

Chuck.

barkur24shubham, 2020-02-03:
in the beginning it show right date and time but after some time it starts
showing like 165:165:80 ,and even after restarting it gives wrong time
(still date isn't wrong ),some time it again gives right time by itself.

closed 2020-04-17 by stale[bot]

thomasfredericks, 2021-06-19:
I am having the same problem. Di you solve it @barkur24shubham ?