espressif/arduino-esp32 #3707: printDiag still reporting wrong mode

philbowles, 2020-02-03:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.4
IDE name:						Arduino IDE 1.8.10
Flash Frequency:					80Mhz
PSRAM enabled:                                     NO
Upload Speed:					921600
Computer OS:                                        Windows 10

```cpp
#include<WiFi.h>
void setup() {
  Serial.begin(115200);
  WiFi.begin();
  WiFi.disconnect(true,true);
  Serial.printf("STA mode? %s AP mode? %s [%d]\n",WiFi.getMode() & WIFI_STA ? "true":"false",WiFi.getMode() & WIFI_AP ? "true":"false",WiFi.getMode());
  WiFi.printDiag(Serial);
  WiFi.mode(WIFI_OFF);
  Serial.printf("STA mode? %s AP mode? %s [%d]\n",WiFi.getMode() & WIFI_STA ? "true":"false",WiFi.getMode() & WIFI_AP ? "true":"false",WiFi.getMode());
  WiFi.printDiag(Serial);
}

void loop() {
}
```

```cpp
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[E][WiFiSTA.cpp:219] begin(): connect failed!
STA mode? false AP mode? false [0]
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 3 - STA_STOP
Mode: STA
Channel: 1
SSID (0): 
Passphrase (0): 
BSSID set: 0
STA mode? false AP mode? false [0]
Mode: STA
Channel: 1
SSID (0): 
Passphrase (0): 
BSSID set: 0
```
getMode() returns 0, yet printDiag reports STA - this is just plain WRONG! Do getMode and printDiag retrieve the REAL value of mode from different places? What is going on here - it has been happening for a long time #1306

atanisoft, 2020-02-03:
WiFi.getMode() returns an enum and not a bitmask. Zero is the enum value for station. If you test for equality rather than bit mask it should work as you expect.

philbowles, 2020-02-03:
"WiFi.getMode() returns an enum and not a bitmask" I wish I knew what that meant, but please don't bother trying to explain it. The code uses the same technique as the library itself uses, so let's skip that part of the discussion. The bitmask test works perfectly - the problem is that printDiag reports a value when getMode() returns 0  for the same thing.

Testing for equality of e.g. STA doesn't work when more than 1 bit is set as in when its STA+AP - & with the bitmask is the only way to check an individual state e.g just STA or just AP.

While that may be an interesting learning experience fro some, its irrelevant to the problem.

atanisoft, 2020-02-03:
> I wish I knew what that meant, but please don't bother trying to explain it.

https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/esp32/esp_wifi_types.h#L28-L34

> The code uses the same technique as the library itself uses, so let's skip that part of the discussion.

The library does *NOT* use bitmasks: https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFi.cpp#L50-L60

> the problem is that printDiag reports a value when getMode() returns 0 for the same thing.

getMode() and printDiag() share one common call, esp_wifi_get_mode. But there are additional checks in getMode() that will result in the value of WIFI_MODE_NULL (zero) being returned, those being related to wifi not initialized. Otherwise the two should be returning the same values.

> Testing for equality of e.g. STA doesn't work when more than 1 bit is set as in when its STA+AP - & with the bitmask is the only way to check an individual state e.g just STA or just AP.

It works just fine to test for equality since the wifi_mode_t enum defines a type for STA+AP.

philbowles, 2020-02-03:
"please don't bother trying to explain it" as your comments are not helpful as well as being plain wrong, yet you persist anyway. I was trying to be polite . "Otherwise the two should be returning the same values" EXACTLY! They don't. So unless you know why, all else it smoke and mirrors.

As for bitmasks you are simply talking nonsense - please stop it, it is helping nobody - here's just one snippet of the dozens of times is is used. If you don't know the answer to the problem, please don't comment. And if you don't know the difference between a bitmask and an enum, _certainly_ don't comment.
```cpp
/**
 * will force a disconnect an then start reconnecting to AP
 * @return ok
 */
bool WiFiSTAClass::reconnect()
{
    if(WiFi.getMode() & WIFI_MODE_STA) {
        if(esp_wifi_disconnect() == ESP_OK) {
            return esp_wifi_connect() == ESP_OK;
        }
    }
    return false;
}

/**
 * Disconnect from the network
 * @param wifioff
 * @return  one value of wl_status_t enum
 */
bool WiFiSTAClass::disconnect(bool wifioff, bool eraseap)
{
    wifi_config_t conf;

    if(WiFi.getMode() & WIFI_MODE_STA){
        if(eraseap){
            memset(&conf, 0, sizeof(wifi_config_t));
            if(esp_wifi_set_config(WIFI_IF_STA, &conf)){

```

atanisoft, 2020-02-03:
@philbowles I won't get into a debate about who is being helpful or bitmasks further, suffice to say we are both right and both wrong.

However, WIFI_MODE_NULL is *INVALID* as far as esp_wifi_set_mode() is concerned and will silently be ignored/dropped by the API. This is why printDiag() is showing STA as it is the last mode that was set via esp_wifi_set_mode().

WiFi.getMode() will return WIFI_MODE_NULL (0) if you call disconnect(true, true) since it shuts off the WiFi driver, you can see this [here](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiGeneric.cpp#L493-L495). These two reasons are why you see different results.

closed 2020-04-17 by stale[bot]