espressif/arduino-esp32 #3708: Can´t abort file upload

Skyseb, 2020-02-03:
HI.

Using the OTAWebUpdater Sample. 

`/*handling uploading firmware file */
  server.on("/update", HTTP_POST, []() {
    server.sendHeader("Connection", "close");
    server.send(200, "text/plain", (Update.hasError()) ? "FAIL" : "OK");
    ESP.restart();
  }, []() {
    HTTPUpload& upload = server.upload();
    if (upload.status == UPLOAD_FILE_START) {
      Serial.printf("Update: %s\n", upload.filename.c_str());
      if (!Update.begin(UPDATE_SIZE_UNKNOWN)) { //start with max available size
        Update.printError(Serial);
      }
    } else if (upload.status == UPLOAD_FILE_WRITE) {
      /* flashing firmware to ESP*/
      if (Update.write(upload.buf, upload.currentSize) != upload.currentSize) {
        Update.printError(Serial);
      }
    } else if (upload.status == UPLOAD_FILE_END) {
      if (Update.end(true)) { //true to set the size to the current progress
        Serial.printf("Update Success: %u\nRebooting...\n", upload.totalSize);
      } else {
        Update.printError(Serial);
      }
    }
  });
`

in: (upload.status == UPLOAD_FILE_WRITE)

i need abort upload after 500k... and i cant do it.

if we send a 25 mb file.. its send.. ...and there is no way to stop it.. 

y try:

   WifiOta::wserver.sendHeader("Location", String("/"), true);   
   WifiOta::wserver.send ( 302, "text/plain", "");
and not work.

can U help me!

Skyseb, 2020-02-04:
new info..

the behavior changes according to the browser that is used.

for example: 

using the chrome from the cell phone the upload works, but when it detects that the size is exceeded the redirect does not take effect.

however, if firefox is used the upload does not even begin.

Now I am checking the behavior from different browsers on the PC.

I'm going to clean the code a little so I can post it here.

closed 2020-04-18 by stale[bot]

benjaminnestler, 2023-12-30:
Are there any news to this topic?

I want to implement the same behavior as the thread creator writes. 

I knew the file size the user wants to upload and I want to avoid this, but if the upload is started, I don't see the possibility to abort the upload.