espressif/arduino-esp32 #3710: ESP32-interrupt-wdt Timeout Core 0 paniced

yogeshwaran774, 2020-02-04:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.1
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
I am facing Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0) in my ESP32 board which has Decawave dwm1000 module. 

whenever decawave gets an signal it will raise an interrupt and esp32 detects that and gets data from the decawave via SPI. The interrupt is attached in arduino core 1. It works properly for sometime and it crashes with the above mentioned error.

if 5 or 6 dwm1000 transmitter transmit packets at the interval of 300 ms once and my receiver dwm1000 will receive it. But if two many packets are coming at the same time it was crashing and the error cause is because of (Interrupt wdt timeout on CPU0) but my interrupt is attached in default arduino core(core 1) then how the debug message gives error in core 0. And i want to understand how the panic handler work and whay is meant by Interrupt wdt timeout  and when will it happens?

### Debug Messages:
```
Core 0 register dump:
PC      : 0x40090066  PS      : 0x00060034  A0      : 0x8009088c  A1      : 0x3ffbe2f0  
A2      : 0x3ffbeed0  A3      : 0x0000cdcd  A4      : 0xb33fffff  A5      : 0x00000001  
A6      : 0x00060021  A7      : 0x0000abab  A8      : 0x0000abab  A9      : 0x3ffbe2f0  
A10     : 0x00000003  A11     : 0x00060023  A12     : 0x00060021  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000001b  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Core 0 was running in ISR context:
EPC1    : 0x401da966  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40090066

Backtrace: 0x40090066:0x3ffbe2f0 0x40090889:0x3ffbe320 0x4008e9eb:0x3ffbe340 0x4009176d:0x3ffbe360 0x40081b96:0x3ffbe370 0x401da963:0x3ffbc120 0x4015c3d6:0x3ffbc140 0x40090855:0x3ffbc160 0x4008e8fd:0x3ffbc180

Core 1 register dump:
PC      : 0x40091aa0  PS      : 0x00060434  A0      : 0x80090be7  A1      : 0x3ffbe810  
A2      : 0x3ffe6ee8  A3      : 0x3ffba5c0  A4      : 0x00000001  A5      : 0x00000001  
A6      : 0x00060423  A7      : 0x00000000  A8      : 0x3ffba5c0  A9      : 0x3ffba5c0  
A10     : 0x00000018  A11     : 0x00000018  A12     : 0x00000001  A13     : 0x00000001  
A14     : 0x00060421  A15     : 0x00000000  SAR     : 0x0000001f  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x4000c428  LEND    : 0x4000c42d  LCOUNT  : 0x00000000  

Backtrace: 0x40091aa0:0x3ffbe810 0x40090be4:0x3ffbe830 0x4008f617:0x3ffbe850 0x400db099:0x3ffbe890 0x400d52e9:0x3ffbe8b0 0x400d56d7:0x3ffbe8e0 0x400d5751:0x3ffbe910 0x400d600b:0x3ffbe930 0x4008132d:0x3ffbe950 0x40081b8d:0x3ffbe970 0x400db1bd:0x3ffcf060 0x400d531a:0x3ffcf080 0x400d594c:0x3ffcf0a0 0x400d5be9:0x3ffcf0d0 0x400d5f9b:0x3ffcf0f0 0x400d299f:0x3ffcf110 0x400d2aa8:0x3ffcf130 0x400d4638:0x3ffcf150 0x400d4681:0x3ffcf170 0x400d4d89:0x3ffcf190 0x400d50a2:0x3ffcf1b0 0x400dc3f9:0x3ffcf1e0 0x4008e8fd:0x3ffcf200

```
###ESP-DECODER:
```
PC: 0x40090066: vTaskExitCritical at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/portmux_impl.inc.h line 163
EXCVADDR: 0x00000000

Decoding stack results
0x40090066: vTaskExitCritical at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/portmux_impl.inc.h line 163
0x40090889: xTaskIncrementTick at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/tasks.c line 2568
0x4008e9eb: vPortCPUReleaseMutex at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/portmux_impl.inc.h line 161
0x4015c3d6: esp_perip_clk_init at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/soc/esp32/include/soc/dport_access.h line 172
0x40090855: xTaskIncrementTick at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/tasks.c line 2534
0x4008e8fd: xPortStartScheduler at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 276

```

lbernstone, 2020-02-04:
Interrupt service routines (ISR) must be very short (<300us), so that you do not miss important "normal" processing.  You should only be assinging a variable or two, and raising a semaphore- your actual processing should be handled outside the ISR.

yogeshwaran774, 2020-02-05:
yeah i understood your point and i am not doing big task inside the interrupt. detaching and setting a bool is happening inside the ISR, then also interrupt was attached in arduino default core which is core 1. If you look at my debug it printed core 0 panic'd, how does core 0 panics with cause of ISR in core 1.

yogeshwaran774, 2020-02-05:
i observed one more thing when this issue was happened, basically i am creating a task which control led and along with that in core 1 continuously ISR executing. The task which is running in core 1 will it cause an issue because the task is not getting enough time to run.

closed 2020-04-19 by stale[bot]