espressif/arduino-esp32 #3711: SmartConfig makes the ESP crash when receiving the WiFi informations for connection

christophepersoz, 2020-02-04:
Hello,

### Description:
I'm trying to make smartConfig running but I just got the board crash when the SSID/pass are provided to the board.

I tried the sketch post by @mickeypop about SmartConfig [https://github.com/espressif/arduino-esp32/issues/1100#issuecomment-392397250](https://github.com/espressif/arduino-esp32/issues/1100#issuecomment-392397250), as well as the one provided as example by Espressif (example/WiFi/WiFiSmartConfig/WiFiSmartConfig.ino). My board configuration is based on ESP32 v1.0.4, Arduino IDE 1.8.10 and the latest version available of Arduino-esp32 library - all is running on macOS Mojave 10.14.6.
The location of the library is in sketches folder/hardware/espressif/esp32. To be sure of this I deleted it and reinstalled it by following the procedure indicated here: [Installation instructions for Mac OS](https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/mac.md)

### Hardware:
Board:							ESP32 Wemos Lolin 32 and ESPDuino 32
Core Installation version:			1.0.4
IDE name:						Arduino IDE? 1.8.10 
Flash Frequency:					80Mhz
PSRAM enabled:                                    yes
Upload Speed:					115200
Computer OS:                                        Mac OSX 10.14.6

### Sketch
```c++
#include "WiFi.h"

void setup() {
  Serial.begin(115200);

  //Init WiFi as Station, start SmartConfig
  WiFi.mode(WIFI_AP_STA);
  WiFi.beginSmartConfig();

  //Wait for SmartConfig packet from mobile
  Serial.println("Waiting for SmartConfig.");
  while (!WiFi.smartConfigDone()) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("SmartConfig received.");

  //Wait for WiFi to connect to AP
  Serial.println("Waiting for WiFi");
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("WiFi Connected.");

  Serial.print("IP Address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

I check that my WiFi router is well configured in 2.4GHz as required by Espressif phone app to send the datas over the network to the ESP. I use ESPtouch app on iPhone to send the SSID/Pass to the ESP32 board. 

I tried all of this on 3 different ESP board, 2 WEMOS Lolin EPS-32 and one EPS-DUINO32. 

The sketch compile is OK, as the upload, but the connection does not occurs and makes the board crash.
I tried different method on `WiFi.mode();` but neither `WIFI_AP_STA` or `WIFI_STA` seems changing anything.

Does anybody have encounter this, as I'm pretty sure that smartconfig should work as well as the code provided by @mickeypop.

### Debug Messages:
For the moment I can't get the Debug Level on those boards. But I'm working to find a better debug trace than the Serial.

Here the traces I'm getting on serial output for the code provided by @mickeypop,
```
WiFi Setup -- 
Entering SmartConfig
.........................................
........................................
........................................
........................................
........................................
..............................E (138395) wifi: wifi ipc: failed to post wifi task
assertion "res == coreID || res == portMUX_FREE_VAL" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/portmux_impl.inc.h", line 105, function: vPortCPUAcquireMutexIntsDisabledInternal
abort() was called at PC 0x40102a43 on core 0

Backtrace: 0x4008c434:0x3ffb5060 0x4008c665:0x3ffb5080 0x40102a43:0x3ffb50a0 0x40089b95:0x3ffb50d0 0x400888b4:0x3ffb5100 0x400889ed:0x3ffb5140 0x4008506d:0x3ffb5160 0x40085161:0x3ffb5190 0x4010b01e:0x3ffb51b0 0x4010e28d:0x3ffb54c0 0x40084376:0x3ffb54f0 0x400e369d:0x3ffb5540 0x400fd399:0x3ffb5570 0x40100853:0x3ffb55a0 0x40101e59:0x3ffb5610 0x400fd8e5:0x3ffb5670 0x4013992a:0x3ffb5690 0x4008f5cd:0x3ffb56b0 0x40088b7d:0x3ffb56f0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
```

And here the traces on Serial monitor for the sketch `WiFiSmartConfig.ino` which is also crashing as soon as the data provided by ESPtouch are received:

```
Waiting for SmartConfig.
...............E (7859) wifi: wifi ipc: failed to post wifi task
CORRUPT HEAP: Bad head at 0x3ffbbcb4. Expected 0xabba1234 got 0x3ffbdaf4
assertion "head != NULL" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c", line 214, function: multi_heap_free
abort() was called at PC 0x401012bf on core 0

Backtrace: 0x4008c434:0x3ffb5450 0x4008c665:0x3ffb5470 0x401012bf:0x3ffb5490 0x4008c0a9:0x3ffb54c0 0x40084b12:0x3ffb54e0 0x40084f19:0x3ffb5500 0x4000bec7:0x3ffb5520 0x400e1fd1:0x3ffb5540 0x400fbc11:0x3ffb5570 0x400ff0cb:0x3ffb55a0 0x401006d5:0x3ffb5610 0x400fc15d:0x3ffb5670 0x401380f2:0x3ffb5690 0x4008f5cd:0x3ffb56b0 0x40088b7d:0x3ffb56f0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
```

It is not the same error, but both seems related to `wifi ipc`
Does that ring a bell to anybody?

Thanks

mickeypop, 2020-02-04:
did yo ever get EspExceptionDecoder working?

like sketches, the file name and the folder name have to be the same or it will not be indexed

ie;
   **YourSketchFolder/tools/EspExceptionDecoder/**    and put **EspExceptionDecoder.jar** in that folder for arduino to index it and put it in the menu

christophepersoz, 2020-02-04:
Yes, I finally got it running! I hope it could help.

#### Trace copied onto EspExceptionDecoder
```
.............................................................................................................................................................................E (90445) wifi: wifi ipc: failed to post wifi task
CORRUPT HEAP: Bad head at 0x3ffbbcb4. Expected 0xabba1234 got 0x3ffbd38c
assertion "head != NULL" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c", line 214, function: multi_heap_free
abort() was called at PC 0x401012bf on core 0

Backtrace: 0x4008c434:0x3ffb5450 0x4008c665:0x3ffb5470 0x401012bf:0x3ffb5490 0x4008c0a9:0x3ffb54c0 0x40084b12:0x3ffb54e0 0x40084f19:0x3ffb5500 0x4000bec7:0x3ffb5520 0x400e1fd1:0x3ffb5540 0x400fbc11:0x3ffb5570 0x400ff0cb:0x3ffb55a0 0x401006d5:0x3ffb5610 0x400fc15d:0x3ffb5670 0x401380f2:0x3ffb5690 0x4008f5cd:0x3ffb56b0 0x40088b7d:0x3ffb56f0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
```

#### Decoding stack results
```
0x4008c434: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 155
0x4008c665: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 170
0x401012bf: __assert_func at ../../../.././newlib/libc/stdlib/assert.c line 63
0x4008c0a9: multi_heap_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c line 214
0x40084b12: heap_caps_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c line 268
0x40084f19: _free_r at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/syscalls.c line 42
0x40088b7d: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```

mickeypop, 2020-02-04:
thanks for the decoded post

it is showing a corrupted heap.  I saw this one before.
when **nodeMCU-32s** first came out i saw this and switching to **ESP32 Dev Module** as the target board fixed it for me.

this is in the boards.txt option under **ESP32 Dev Module**
**esp32.menu.PSRAM.enabled.build.defines=-DBOARD_HAS_PSRAM -mfix-esp32-psram-cache-issue**
WeMos does not have this param

it is the only board definition with the **-mfix-esp32-psram-cache-issue** param

i believe this may help

christophepersoz, 2020-02-05:
Thanks for the tips @mickeypop.

I dig a bit in the board.txt and add few options on the **Wemos Lolin 32** to experiments; the changes I made are based on the **ESP32 Dev Module** as suggested.

#### Board.txt changes under the lolin32 definition
```
lolin32.menu.PSRAM.disabled=Disabled
lolin32.menu.PSRAM.disabled.build.defines=
lolin32.menu.PSRAM.enabled=Enabled
lolin32.menu.PSRAM.enabled.build.defines=-DBOARD_HAS_PSRAM -mfix-esp32-psram-cache-issue

lolin32.menu.DebugLevel.none=None
lolin32.menu.DebugLevel.none.build.code_debug=0
lolin32.menu.DebugLevel.error=Error
lolin32.menu.DebugLevel.error.build.code_debug=1
lolin32.menu.DebugLevel.warn=Warn
lolin32.menu.DebugLevel.warn.build.code_debug=2
lolin32.menu.DebugLevel.info=Info
lolin32.menu.DebugLevel.info.build.code_debug=3
lolin32.menu.DebugLevel.debug=Debug
lolin32.menu.DebugLevel.debug.build.code_debug=4
lolin32.menu.DebugLevel.verbose=Verbose
lolin32.menu.DebugLevel.verbose.build.code_debug=5
```

Then I compiled the WiFiSmartConfig.ino sketch, and submit the SSID/Pass to the ESP32. It crash again, with and without PSRAM, that doesn't change anything. It just helped me to have a better understanding as I could activate the Debug Level to "Debug". The problem remains with the 3 boards I'm running the SmartConfig sketch.

#### Serial trace after submitting the SSID/Pass
```
xets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
[W][esp32-hal-psram.c:30] psramInit(): PSRAM init failed!
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
Waiting for SmartConfig.
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 15 - AP_STACONNECTED
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 15 - AP_STACONNECTED
........[D][WiFiSTA.cpp:738] _smartConfigCallback(): Status: FIND_CHANNEL
..............[D][WiFiSTA.cpp:738] _smartConfigCallback(): Status: GETTING_SSID_PSWD
[D][WiFiSTA.cpp:742] _smartConfigCallback(): Type: ESPTOUCH
E (11506) wifi: wifi ipc: failed to post wifi task
CORRUPT HEAP: Bad head at 0x3ffbbcb4. Expected 0xabba1234 got 0x3ffbdaf8
assertion "head != NULL" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c", line 214, function: multi_heap_free
abort() was called at PC 0x401049fb on core 0

Backtrace: 0x4008e8b0:0x3ffb5450 0x4008eae1:0x3ffb5470 0x401049fb:0x3ffb5490 0x4008e525:0x3ffb54c0 0x4008726e:0x3ffb54e0 0x40087675:0x3ffb5500 0x4000bec7:0x3ffb5520 0x400e56d5:0x3ffb5540 0x400ff34d:0x3ffb5570 0x40102807:0x3ffb55a0 0x40103e11:0x3ffb5610 0x400ff899:0x3ffb5670 0x4013b8f6:0x3ffb5690 0x40091a41:0x3ffb56b0 0x4008aff9:0x3ffb56f0

Rebooting...
```

atanisoft, 2020-02-05:
Based on the output, that board does not likely have PSRAM so the additional parameters for that are not necessary.

christophepersoz, 2020-02-05:
Indeed, the PSRAM can't be initialised on boot.

lbernstone, 2020-02-05:
How do you have the app configured on your phone?  Perhaps the ssid or password has an invalid character in it?

christophepersoz, 2020-02-05:
@lbernstone, the ESPtouch app as well as ioT Smartconfig app does not have any setup, you just enter your SSID and Password and then you submit it. But, your idea of checking the password of the SSID was a good lead. I changed the password which was containing an _
I removed it, and check, and the connection was successfully established with the 2 sketches. So I roll back to the old password with the _ char to see if that was the cause, and the connection occurs correctly. Maybe there was something wrong on the router and changing the password make it right and working normally. Thanks a lot, I haven't thought to this a second!

@mickeypop, once received for the first time the SSID and Pass, unplug and restart the board make it connect to the network normally. Perfect! Thanks for sharing your code! 😉

mickeypop, 2020-02-05:
glad it's working now

kemaldede, 2020-02-18:
> @lbernstone, the ESPtouch app as well as ioT Smartconfig app does not have any setup, you just enter your SSID and Password and then you submit it. But, your idea of checking the password of the [...]

I have the same problem. Unfortunately it didn't fix when i changed the ssid name and password of my router. Always stuck in ......... session

christophepersoz, 2020-02-18:
@kemaldede, Have you checked that your router isn’t set in 5GHz mode? SmartConfig, and maybe ESP isn’t working in 5GHz WiFi mode.

atanisoft, 2020-02-18:
ESP32 only supports 2.4Ghz WiFi bands.

kemaldede, 2020-02-18:
thank you for fast response. I checked my router. The smart config example with esp-idf example code is working,but not with arduino

closed 2020-05-02 by stale[bot]

kundannakalank, 2021-02-17:
[image]

Please help me I don't know how to solve this problem...

kundannakalank, 2021-02-17:
> @lbernstone, the ESPtouch app as well as ioT Smartconfig app does not have any setup, you just enter your SSID and Password and then you submit it. But, your idea of checking the password of the [...]

I m facing same problem please help me how to solve it

mickeypop, 2021-02-17:
@kundannakalank
without any code or circuit, it looks like you are using the mutex handler and causing a timing out.
This usually caused by setting a lock, calling a function, forgetting to release the lock and calling something that needed the lock freed to use it.

this state tracking is especially needed if you are using both cores and using a shared resource like I2C, SPI etc...
  
i like to use mutex when working with multiple I2C devices.
ie;  in every case, with mutex when you set a lock you need to release the lock when done for other functions to access the buss or it will timeout and call the panic.c lib to escape.
Simply using a global bool var to keep the cores from bumping into each other is a good start
 
the panic.c is a kernel timeout.   
Somewhere your code is likely tying up a resource needed elsewhere.

can you post your code?