espressif/arduino-esp32 #3712: SPIFFS free space

joseitor, 2020-02-04:
¿Is there any method that return the space available in SPIFFS? If not, I think it must be included, due it's important in cycle data systems, where for example you save in SPIFFS the data of the device every hour to have an file with the historical data, and wanted that once the SPIFFS memory get full it start to overwrite the oldest data to always have data of for example the last year.

lbernstone, 2020-02-04:
https://github.com/espressif/arduino-esp32/blob/master/libraries/SPIFFS/src/SPIFFS.h#L28-L29

closed 2020-04-18 by stale[bot]

duncanhames, 2020-07-17:
For anyone who would like an explicit example:

`  int tBytes = SPIFFS.totalBytes();
  int uBytes = SPIFFS.usedBytes();`

rtek1000, 2024-03-19:
> For anyone who would like an explicit example:
> 
> ` int tBytes = SPIFFS.totalBytes(); int uBytes = SPIFFS.usedBytes();`

`uint32_t SPIFFS_freeBytes = (SPIFFS.totalBytes() - SPIFFS.usedBytes());`

Note: FatFS has a proper freeBytes()

`FFat.freeBytes();`

Ref.: https://mischianti.org/esp32-integrated-ffat-fat-exfat-filesystem-6/