espressif/arduino-esp32 #3716: ESP.getPsramSize() returnes incorrect PSRAM size for Wroover-B. Error or fake module?

mrdc, 2020-02-05:
## Hardware:
Board:							TTGO ESP32 Mini32
Core Installation/update date:			1.0.4 Oct 2019
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                         yes
Upload Speed:						115200
Computer OS:                                            Windows 7

### Description:
Hello, `ESP.getPsramSize()` returnes incorrect PSRAM size for the board based on ESP32 Wroover-B: it shows 4194252 (~4MB) but it should be 8388608 (8MB).
Is it correct behaviour or my module is not Wroover-B, but the regular Wroover?

```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
  Serial.begin(115200);
  Serial.printf("PSRAM Total heap %d, PSRAM Free Heap %d\n",ESP.getPsramSize(),ESP.getFreePsram());
}

void loop() {
}
```

atanisoft, 2020-02-05:
From a very quick search I found https://www.tindie.com/products/ttgo/lilygor-ttgo-mini32-esp32-wrover-b-module/ which shows the board has 4MB flash and 4MB PSRAM... So it appears the function is working as intended.

mrdc, 2020-02-05:
> 
> 
> From a very quick search I found https://www.tindie.com/products/ttgo/lilygor-ttgo-mini32-esp32-wrover-b-module/ which shows the board has 4MB flash and 4MB PSRAM... So it appears the function is working as intended.

Based on info from Espressif, ESP32 Wrover-B modules have 8MB PSRAM.
Here is a photo of the dev board I'm using - it definately says it's Wrover-B...

[image]

Dev board manufacturer also declares that it has Wrover-B with 8MB PSRAM.

[image]

atanisoft, 2020-02-05:
This should likely be routed to https://github.com/espressif/esp-idf as ESP.getPsramSize() simply calls:
```
uint32_t EspClass::getPsramSize(void)
{
    multi_heap_info_t info;
    heap_caps_get_info(&info, MALLOC_CAP_SPIRAM);
    return info.total_free_bytes + info.total_allocated_bytes;
}
```

lbernstone, 2020-02-05:
I'm not sure v3.2 had code for 8MB PSRAM.  Can you try with the v4.0 branch?

chegewara, 2020-02-05:
I think that only on boot esp-idf can recognize size of PSRAM bigger than 4MB. 
I have only m5stack fire to tests at the moment and it has only 32Mbits SPIRAM.

mrdc, 2020-02-10:
Everything is fine - this himem code https://github.com/TridentTD/arduino_esp32_himem helps:

```
15:34:08.934 -> PSRAM total size     : 8388608 
15:34:08.934 -> ----------------------------------------
15:34:08.934 -> PSRAM first 4MB size : 3932108 
15:34:08.934 -> PSRAM first 4MB free : 3932108 
15:34:08.934 -> PSRAM HI-MEM    size : 4456448 
15:34:08.934 -> PSRAM HI-MEM    free : 4456448 
```

closed 2020-02-10 by mrdc