espressif/arduino-esp32 #3723: UART BaudRate - Is it 110 possible?

allexotecnologia, 2020-02-07:
Hey guys.

I'm trying to use one of the serial ports (1) to read a monodirectional serial stream coming from a energy meter. The thing is the baud of the meter comes at 110. 

Is it possible from the esp32 using the uarts to read at 110? If not, is there a solution that i can implement to do it?

Thanks

lbernstone, 2020-02-07:
Please follow the [issue template](https://github.com/espressif/arduino-esp32/blob/master/docs/ISSUE_TEMPLATE.md).

allexotecnologia, 2020-02-07:
> Please follow the [issue template](https://github.com/espressif/arduino-esp32/blob/master/docs/ISSUE_TEMPLATE.md).

Sorry, @lbernstone ! Already edited to help the understanding of the problem. Thanks in advance.

closed 2020-04-24 by stale[bot]