espressif/arduino-esp32 #3724: WiFiTelnetToSerial not working as described

willfaulds, 2020-02-07:
### Hardware:
Board:							Adafruit Feather HUZZAH32
Core Installation version:			1.8.11?
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    ?
Upload Speed:					921600
Computer OS:                                       Mac OSX

### Description:

The example does not work as I think it should.

I cannot get any telnet->serial comms (or vice versa) but a direct serverClients[0].write() works...

Can someone help me update the sketch to work or help me understand correct usage of the sketch? Many thanks

I.e
`serverClients[0].write("a")`
added at the end of the main loop confirms the telnet client is working an able to receive data.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

As per [example](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiTelnetToSerial/WiFiTelnetToSerial.ino)

### Debug Messages:
```
17:31:38.708 -> ⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮`⸮
17:31:38.969 -> Connecting
17:31:38.969 -> [I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: lynaguewifi
17:31:38.969 -> [I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: ssid_from_AP_2
17:31:38.969 -> [I][WiFiMulti.cpp:84] addAP(): [WIFI][APlistAdd] add SSID: ssid_from_AP_3
17:31:39.074 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
17:31:39.074 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
17:31:40.805 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 1 - SCAN_DONE
17:31:40.805 -> [I][WiFiMulti.cpp:114] run(): [WIFI] scan done
17:31:40.805 -> [I][WiFiMulti.cpp:119] run(): [WIFI] 2 networks found
17:31:40.805 -> [D][WiFiMulti.cpp:149] run():  --->   0: [1][7C:8B:CA:CB:32:80] lynaguewifi (-53) *
17:31:40.805 -> [D][WiFiMulti.cpp:151] run():        1: [11][F4:06:8D:41:A1:88] Barn(shop) (-64) *
17:31:40.805 -> [I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: 7C:8B:CA:CB:32:80 SSID: lynaguewifi Channal: 1 (-53)
17:31:41.228 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 4 - STA_CONNECTED
17:31:41.264 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
17:31:41.264 -> [D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.5.233, MASK: 255.255.255.0, GW: 192.168.5.1
17:31:41.264 -> [I][WiFiMulti.cpp:174] run(): [WIFI] Connecting done.
17:31:41.264 -> [D][WiFiMulti.cpp:175] run(): [WIFI] SSID: lynaguewifi
17:31:41.264 -> [D][WiFiMulti.cpp:176] run(): [WIFI] IP: 192.168.5.233
17:31:41.264 -> [D][WiFiMulti.cpp:177] run(): [WIFI] MAC: 7C:8B:CA:CB:32:80
17:31:41.264 -> [D][WiFiMulti.cpp:178] run(): [WIFI] Channel: 1
17:31:41.264 -> 192.168.5.233
17:31:41.264 -> Ready! Use 'telnet 192.168.5.233 23' to connect
17:32:09.473 -> New client: 0 192.168.5.38
17:33:43.681 -> [D][WiFiClient.cpp:509] connected(): Disconnected: RES: 0, ERR: 128
17:33:50.481 -> New client: 0 192.168.5.217

```

MarkyAD, 2020-02-07:
What pins did you use for Serial2? You can set up whatever pins you like by adding to setup(). Be aware that certain pins have special meanings and may or may not be usable.
```c
#define RX_PIN 25
#define TX_PIN 26

void setup() {
  Serial2.begin(9600, SERIAL_8N1, RX_PIN, TX_PIN);
}
```

willfaulds, 2020-02-07:
Ah I did wonder if something like that might be the case. Any idea what the defaults are or even better a “standard” board I could compare to?

The adafruit feather circuits are here
https://learn.adafruit.com/adafruit-huzzah32-esp32-feather/downloads

All the best,

WIll

MarkyAD, 2020-02-07:
Serial1 RX=9 TX=10
Serial2 RX=16 TX=17

I'd stay away from 16/17 as they're typically used for flash and psram. Either use Serial1 or change the pins to something useable as above.

willfaulds, 2020-02-07:
I might have fundementally missed what this example does... Would you mind putting it in dumdum terms?

I thought Serial2 was being used as a buffer and hence it wouldn’t matter what pins. If all I wanted to do was bridge Serial to telnet do I need Serial2? I actually only want to achieve a one way Serial->telnet so when deploying new code OTA I can have a basic level of debugging. Tried using Bluetooth examples only to find none work with apple devices...

All the best,

WIll

MarkyAD, 2020-02-07:
Serial2 is just another hardware UART. Since Serial itself is used for debugging i/o, one normally attaches other devices to the other UARTs. If all you want to do is redirect Serial to telnet, then just change the Serial.print() to your own function... something like this is simple:
```c
// write to one client
void termclient_write(uint8_t client, char *s) {
  termClients[client].write(s, strlen(s));
}

// write to one client with cr/lf
void termclient_writeln(uint8_t client, char *s) {
  char *crlf = "\r\n";
  termclient_write(client, s);
  termclient_write(client, crlf);
}
```
Trickier is capturing boot messages that occur before the telnet server starts. For that you can code a boot message buffer that is then displayed when the client connects.

MarkyAD, 2020-02-07:
Anyway, theres nothing wrong with the example as written, it just has a specific purpose that you can adapt to your own implementation. Please close if this solves the issue.

willfaulds, 2020-02-11:
I am testing today and will close add notes as appropriate.

MarkyAD, 2020-02-11:
As I said, the example works just fine as it is. You just have a diffent intent for what you expect from it. There is no core issue here, please close.

willfaulds, 2020-02-19:
Closing now (sorry for the delay)

As a note to anyone researching I found what I was looking for in the following library
https://github.com/JoaoLopesF/RemoteDebug

closed 2020-02-19 by willfaulds