espressif/arduino-esp32 #3725: Serial1 lockup in Begin() method

dflogeras, 2020-02-07:
### Hardware:
Board:							ESP32 Dev Module (WROOM-32D)
Core Installation version:			ed220bd0420236e9ff95888f7459f0eb9c97a570
IDE name:						Arduino IDE 1.8.7 (Gentoo)
Flash Frequency:					80MHz
PSRAM enabled:                                    no
Upload Speed:					921600
Computer OS:                                        Linux

### Description:
I've boiled a more complicated system down to a minimal test.  Effectively I must open Serial1 at a couple different baud rates to determine what sensor (if any) is connected to a MAX3227 RS232 level converter.

In order to reproduce, take a WROOM-32D devkit, and ground both pins 18 and 19.  For some reason this is required (and it is what my MAX3227 does if the sensor is not present/not powered).

Next, build this sketch.  I have deviated from defaults in the following two ways:
- 80 MHz WiFi/BT (This absolutely matters).
- Minimal SPIFFS with OTA

When you run the following it will never get to the Serial.println( "two" ) line.

If you comment out the #include <BLEServer.h>, it will get further, but never print "four".

If you then comment out the delay(100) it works as expected, entering the Loop() portion and printing hello.

If you run it at 240MHz it works as expected (with or without the above commented lines).

When hung, I caught it in openocd/JLInk and I get the following backtrace:

(gdb) info thread
[New Thread 1073464980]
  Id   Target Id         Frame
  9    Thread 1073464980 (Name: IDLE1) 0x400e9682 in esp_pm_impl_waiti () at /ho
me/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/compo
nents/esp32/pm_esp32.c:492
* 8    Thread 2 (Name: loopTask, Interrupted @CPU1) 0x4000bff0 in ?? ()
  7    Thread 1073446484 (Name: ipc0) 0x40087d90 in xQueueGenericReceive (xQueue
=0x3ffb7df4, pvBuffer=0x0, xTicksToWait=4294967295, xJustPeeking=0) at /home/run
ner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/
freertos/queue.c:1592
  6    Thread 1073445228 (Name: esp_timer) 0x40087d90 in xQueueGenericReceive (x
Queue=0x3ffaff9c, pvBuffer=0x0, xTicksToWait=4294967295, xJustPeeking=0) at /hom
e/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/compon
ents/freertos/queue.c:1592
  5    Thread 1073455812 (Name: ipc1) 0x40087d90 in xQueueGenericReceive (xQueue
=0x3ffb9e54, pvBuffer=0x0, xTicksToWait=4294967295, xJustPeeking=0) at /home/run
ner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/
freertos/queue.c:1592
  4    Thread 1073467672 (Name: Tmr Svc) 0x40089f1c in prvProcessTimerOrBlockTas
k (xNextExpireTime=<optimized out>, xListWasEmpty=<optimized out>) at /home/runn
er/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/f
reertos/timers.c:589
  3    Thread 1073469608 (Name: loopTask, State: Running @CPU1) 0x40080f03 in _u
art_isr (arg=0x0) at /home/esp32/Arduino/hardware/espressif/esp32/cores/esp32/es
p32-hal-uart.c:73
  2    Thread 1073463568 (Name: IDLE0, State: Running @CPU0) 0x400e9682 in esp_p
m_impl_waiti () at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib
-builder/esp-idf/components/esp32/pm_esp32.c:492

(Switching to thread 8, the setup() thread):

(gdb) thread 8
[Switching to thread 8 (Thread 2)]
#0  0x4000bff0 in ?? ()
(gdb) bt
#0  0x4000bff0 in ?? ()
#1  0x40088dbc in vTaskExitCritical (mux=<optimized out>) at /home/runner/work/e
sp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/t
asks.c:4274
#2  0x400d51be in esp_intr_alloc_intrstatus (source=<optimized out>, flags=1038,
 intrstatusreg=<optimized out>, intrstatusmask=<optimized out>, handler=<optimiz
ed out>, arg=0x0, ret_handle=ret_handle@entry=0x3ffbdbd8 <_uart_bus_array+36>) a
t /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/
components/esp32/intr_alloc.c:666
#3  0x400d5214 in esp_intr_alloc (source=35, flags=1024, handler=0x40080f00 <_ua
rt_isr>, arg=0x0, ret_handle=0x3ffbdbd8 <_uart_bus_array+36>) at /home/runner/wo
rk/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/
intr_alloc.c:686
#4  0x400d1a18 in uartEnableInterrupt (uart=0x3ffbdbc8 <_uart_bus_array+20>) at
/home/esp32/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal-uart.c:110
#5  0x400d1b06 in uartAttachRx (uart=0x3ffbdbc8 <_uart_bus_array+20>, rxPin=18 '
\022', inverted=false) at /home/esp32/Arduino/hardware/espressif/esp32/cores/esp
32/esp32-hal-uart.c:151
#6  0x400d1fdc in uartBegin (uart_nr=<optimized out>, baudrate=<optimized out>,
config=<optimized out>, rxPin=<optimized out>, txPin=19 '\023', queueLen=<optimi
zed out>, inverted=inverted@entry=false) at /home/esp32/Arduino/hardware/espress
if/esp32/cores/esp32/esp32-hal-uart.c:219
#7  0x400d0f7d in HardwareSerial::begin (this=0x3ffbfe08 <Serial1>, baud=460800,
 config=134217756, rxPin=<optimized out>, txPin=<optimized out>, invert=false, t
imeout_ms=timeout_ms@entry=20000) at /home/esp32/Arduino/hardware/espressif/esp3
2/cores/esp32/HardwareSerial.cpp:55
#8  0x400d0d45 in setup () at /home/esp32/Arduino/crash/crash.ino:42
#9  0x400d14f5 in loopTask (pvParameters=0x0) at /home/esp32/Arduino/hardware/es
pressif/esp32/cores/esp32/main.cpp:14
#10 0x40087f3c in vPortTaskWrapper (pxCode=0x400d14ec <loopTask(void*)>, pvParam
eters=0x0) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-buil
der/esp-idf/components/freertos/port.c:143

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

#include <BLEServer.h>

namespace bsp {
  struct pin {
    enum value {
      serial1_rx = 18,
      serial1_tx = 19,
    };
  };
}
const unsigned int ser1_baudrate( 460800 );
const unsigned int ser1_fallback_baudrate( 9600 );

void setup() {

  Serial.begin( 115200 );

  delay( 3000 );

  Serial.println( "one" );
  Serial1.begin( ser1_baudrate,
                 SERIAL_8N1,
                 bsp::pin::serial1_rx, bsp::pin::serial1_tx );
  Serial1.write( 0x0 );
  while( Serial1.available() ) Serial1.read();
  Serial.println( "two" );
  
  Serial1.end();
  Serial1.begin( ser1_fallback_baudrate,
                 SERIAL_8N1,
                 bsp::pin::serial1_rx, bsp::pin::serial1_tx );
  Serial1.write( 0x0 );
  while( Serial1.available() ) Serial1.read();

  delay( 100 );
  Serial.println( "three" );
  Serial1.begin( ser1_baudrate,
                 SERIAL_8N1,
                 bsp::pin::serial1_rx, bsp::pin::serial1_tx );
  Serial.println( "four" );
  Serial1.write( 0x0 );
  while( Serial1.available() ) Serial1.read();

  
  
}

void loop() {
  delay( 100 );
  Serial.println( "hello" );
  
}
```

### Debug Messages:
```
Nothing comes out other than my Serial.println() in the sketch provided. 
```

lyusupov, 2020-02-13:
For me It looks similar to https://github.com/espressif/arduino-esp32/issues/3563 .

dflogeras, 2020-02-14:
@lyusupov Thanks for the pointer!  Hopefully the backtrace and minimal program I added will help if/when someone can look deeper at the issue.

For the time being I'll try using updateBaudRate() to workaround as suggested in #3563

closed 2020-04-28 by stale[bot]