espressif/arduino-esp32 #3734: ESP32 won't communicate with LAN870

Captain-Bavaria, 2020-02-12:
What i tried to do:

I wired my system as suggested by [sautters blog](https://sautter.com/blog/ethernet-on-esp32-using-lan8720/).
the i used the example code and adjusted my pins according to the other issue posts i've found on here.
for me pin 14 as PHY_POWER seemed to get the best results
So my System is wired like this:

GPIO14 - PHY_POWER : NC - Osc. Enable - 4k7 Pulldown
GPIO22 - EMAC_TXD1 : TX1
GPIO19 - EMAC_TXD0 : TX0
GPIO21 - EMAC_TX_EN : TX_EN
GPIO26 - EMAC_RXD1 : RX1
GPIO25 - EMAC_RXD0 : RX0
GPIO27 - EMAC_RX_DV : CRS
GPIO00 - EMAC_TX_CLK : nINT/REFCLK (50MHz) - 4k7 Pullup
GPIO23 - SMI_MDC : MDC
GPIO18 - SMI_MDIO : MDIO
GND : GND
3V3 : VCC

My Setup
Note: GPIO0 and GPIO17 were not available so i did this hotglue contraption the pin which is not connected is GPIO17 becuase i realised that i can use any GPIO as PHY_POWER 
The rest is very similar to again sautters blog
[image]

### Hardware:
Board:						      nodeMCu32

IDE name:					      Platform.io
Flash Frequency:				      ?40Mhz?
PSRAM enabled:                                     ?no? ?yes?
Upload Speed:					      115200?
Computer OS:                                        Windows 10

### Description:
I do get the clock siganl working and PHY_POWER is low during boot, so it gets enabled just after the boot is finished which is also expalined to be very important by sautter.

strangely all the other pins are either low or high so no signals are send which i dont understand because the most common problem seems to be the clock not working correctly which in my opinin it does.
I did already read all the similar posts regarding my problem but i am still stuck
pleas dont reference me to existing problems because they did not provide a solution for me or at least i was not able to get it to work.

PHY_POWER during boot

[image]

Serial output
[image]

Clook after boot is finished
[image]

### Sketch:  (leave the backquotes for [code formatting](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/ETH_LAN8720_internal_clock/ETH_LAN8720_internal_clock.ino))
```cpp

/*
This sketch shows how to configure different external or internal clock sources for the Ethernet PHY
*/

#include <ETH.h>

/* 
   * ETH_CLOCK_GPIO0_IN   - default: external clock from crystal oscillator
   * ETH_CLOCK_GPIO0_OUT  - 50MHz clock from internal APLL output on GPIO0 - possibly an inverter is needed for LAN8720
   * ETH_CLOCK_GPIO16_OUT - 50MHz clock from internal APLL output on GPIO16 - possibly an inverter is needed for LAN8720
   * ETH_CLOCK_GPIO17_OUT - 50MHz clock from internal APLL inverted output on GPIO17 - tested with LAN8720
*/
#ifdef ETH_CLK_MODE
#undef ETH_CLK_MODE
#endif
#define ETH_CLK_MODE ETH_CLOCK_GPIO0_IN

// Pin# of the enable signal for the external crystal oscillator (-1 to disable for internal APLL source)
#define ETH_POWER_PIN   14

// Type of the Ethernet PHY (LAN8720 or TLK110)
#define ETH_TYPE        ETH_PHY_LAN8720

// I²C-address of Ethernet PHY (0 or 1 for LAN8720, 31 for TLK110)
#define ETH_ADDR        1

// Pin# of the I²C clock signal for the Ethernet PHY
#define ETH_MDC_PIN     23

// Pin# of the I²C IO signal for the Ethernet PHY
#define ETH_MDIO_PIN    18

static bool eth_connected = false;

void WiFiEvent(WiFiEvent_t event) {
  switch (event) {
    case SYSTEM_EVENT_ETH_START:
      Serial.println("ETH Started");
      //set eth hostname here
      ETH.setHostname("esp32-ethernet");
      break;
    case SYSTEM_EVENT_ETH_CONNECTED:
      Serial.println("ETH Connected");
      break;
    case SYSTEM_EVENT_ETH_GOT_IP:
      Serial.print("ETH MAC: ");
      Serial.print(ETH.macAddress());
      Serial.print(", IPv4: ");
      Serial.print(ETH.localIP());
      if (ETH.fullDuplex()) {
        Serial.print(", FULL_DUPLEX");
      }
      Serial.print(", ");
      Serial.print(ETH.linkSpeed());
      Serial.println("Mbps");
      eth_connected = true;
      break;
    case SYSTEM_EVENT_ETH_DISCONNECTED:
      Serial.println("ETH Disconnected");
      eth_connected = false;
      break;
    case SYSTEM_EVENT_ETH_STOP:
      Serial.println("ETH Stopped");
      eth_connected = false;
      break;
    default:
      break;
  }
}

void testClient(const char * host, uint16_t port) {
  Serial.print("\nconnecting to ");
  Serial.println(host);

  WiFiClient client;
  if (!client.connect(host, port)) {
    Serial.println("connection failed");
    return;
  }
  client.printf("GET / HTTP/1.1\r\nHost: %s\r\n\r\n", host);
  while (client.connected() && !client.available());
  while (client.available()) {
    Serial.write(client.read());
  }

  Serial.println("closing connection\n");
  client.stop();
}

void setup() {

  Serial.begin(115200);
  WiFi.onEvent(WiFiEvent);
  ETH.begin(ETH_ADDR, ETH_POWER_PIN, ETH_MDC_PIN, ETH_MDIO_PIN, ETH_TYPE, ETH_CLK_MODE);
}

void loop() {
  if (eth_connected) {
    testClient("google.com", 80);
  }
  delay(10000);
}
```

Captain-Bavaria, 2020-02-12:
I also do get this Serial output sometimes after several re-uploads of my code to the esp whithout changing anything on the hardware or code.
I have not found any similar outputs on the Internet yet, but i guess that it is still the same problem as the output above.

[image]

Captain-Bavaria, 2020-02-12:
I did get these Signals from MDC and MDIO. 
However they only appeared very schortley after pressing the reset button on my ESP-Board. 

I also tried measuring the respons from the LAN8720 PHY on RX0 and RX1 but i was not able to record any signal. So my conclusion is that the LAN is not responding.
While taking my measurements the external Clock signal, which the ESP was receiving on GPIO0 stayed constant at 50MHz. 

Does anybody know a solution or a way to tackel this problem ??

recorded MDC-signal 
[image]

recorded MDIO-signal
[image]

MarkyAD, 2020-02-12:
Looks like some noise and ringing on both the last 2 shots (nice Tektronix scope). How I got it to work was by making a "shield" type setup with the esp on one board and the 8720 on another, connected by pin headers. That way you can use the shortest runs possible of silicone or teflon wire to wire it up, which really improved the reliability.

Captain-Bavaria, 2020-02-17:
```cpp
void setup() {
  Serial.begin(115200);
  WiFi.onEvent(WiFiEvent);

Serial.println("test");
  
    ETH.begin(ETH_ADDR, ETH_POWER_PIN, ETH_MDC_PIN, ETH_MDIO_PIN, ETH_TYPE, ETH_CLK_MODE); 
  
Serial.println("test11");
} 

void loop() {
  if (eth_connected) {
    testClient("google.com", 80);
  }
  delay(1000);

Serial.println("test12");

}
```

I added these three **Serial.println** to see where my code is stuck and it only prints:

 **test** 
**E (2149) emac: Reset EMAC Timeout**

which means it doesnt reach **loop()** nor does it finish **ETH.begin()** .
This is very strange.
Does anyone know a soulution because this seems to be a Software Problem.

vk3-hca, 2020-02-25:
I am running the same demo code as you untouched except for a couple print statements, everything is set up identical, I also have changed the header pins on the lan pcb. Wiring,  checks out pin to pin on the board, no shorts etc. Also on protoboard with wire wrap wire.

Using PlatformIO/arduino on a ESP32-dev wroom board. I found out the board was not 38 pin, so also I picked up the GPIO0at the reset button. The second board I have is 38 pin. Different pinouts are a pain in the arse! It would been better if the extra connections were on the extra pins. 

I have not changed from phy-pwr from the default GPIO 17 
I suspect the wires a might be a bit long, 12cm on the clock and about 15 on the rest, but without the clock enable it does not matter  I have another devboard which is a 38 pin version I propose to put together with the eth board directly under the ESP32 for shortest lines.

I have 2 new lan boards both do the same thing. 
The ESP32 runs other demos fine.
I think we have very similar setups on the bench.

And my setup gives exactly the same  error messages out of boot as yours. Connected to the network, possibly somewhat prematurely it as turns out, both leds on the cat 5 socket are working, orange is steady green is flickering.

But I have it come out of boot with no phy_pwh high on gpio17, thus no clock, thus no activity on the lines. In the end I decided the pin must be faulty? But in code it can be set high and the clock is ok on gpio 0.

So in setup I start serial, print "setup" set the pin high and then ethernet begin, it makes no difference other than the clock signal.

In the main just enable 1 second on/ off the pin and print "on" /"off". The pin is OK.

Here is the code and the result.  It shows the loop is running but is slow the HIGH is 1 second, the LOW is about 5 seconds. RTos must be doing something between loop cycles?

Looks to me you put a lot work into it, I have wasted days, I installed the tool chain  a few weeks ago, the Sautter blog is dated 2107, other comments I have seen of it just working fine are similar era.

I will make another proto board with the 38pin devkit and see what it does. 

I cant explain the difference in activity, but I am wondering about my toolchain installation?

Have you succeeded as yet?

Code and output follows
  
void setup()
{
 // pinMode(17, OUTPUT);
  Serial.begin(115200);
   //digitalWrite(17, HIGH); 
 Serial.println("Setup");
  WiFi.onEvent(WiFiEvent);
  ETH.begin();
  
}

void loop()
{
 //digitalWrite(17, HIGH); 
 Serial.println("HIGH");
delay(1000); 
 //digitalWrite(17, LOW); 
 Serial.println("LOW");
delay(1000); 
 // digitalWrite(_eth_phy_power_pin, enable);
  if (eth_connected) {
    testClient("google.com", 80);
  }
  delay(10000);
}
/*

--- Miniterm on COM14  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
Setup
E (1039) emac: Timed out waiting for PHY register 0x2 to have value 0x0007(mask 0xffff). Current value 0x0000
E (2040) emac: Timed out waiting for PHY register 0x3 to have value 0xc0f0(mask 0xfff0). Current value 0x0000
E (2041) emac: Initialise PHY device Timeout
[E][ETH.cpp:101] begin(): esp_eth_enable error: -1
LOW
HIGH
LOW
HIGH
LOW

vk3-hca, 2020-02-25:
Small update,
I loaded your code in defining gpio pin 17 for phy-pwr and sem result as you but now the pin goes high?
Still no apparent activity on the other lines but the clock,

vk3-hca, 2020-02-25:
Another Update,
I have just assembled another wirewrap proto with the 38pin ESP32 dev board and the network socket.

I compiled your code with phy-pwr defined on GPIO 17 with the edition of 2 print statements in setup and loop and it works. 
IP and gateway from the DHCP server and the DNS from a local DNS server specified by DHCP so the networking DHCP seems 100% complete.

This is what comes back:
Looking for advanced Serial Monitor with UI? Check http://bit.ly/pio-advanced-monitor
--- Miniterm on COM14  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
Setup
Loop
ETH Started
ETH Connected
ETH MAC: 24:6F:28:25:E4:9B, IPv4: 192.168.2.44, FULL_DUPLEX, 100Mbps
Loop

connecting to google.com
HTTP/1.1 301 Moved Permanently
Location: http://www.google.com/
Content-Type: text/html; charset=UTF-8
Date: Tue, 25 Feb 2020 14:25:55 GMT
Expires: Thu, 26 Mar 2020 14:25:55 GMT
Cache-Control: public, max-age=2592000
Server: gws
Content-Length: 219
X-Xss-Protection: 0
X-Frame-Options: SAMEORIGIN
X-Cnection: close

<HTML><HEAD><meta http-equiv="content-type" content="text/html;charset=utf-8">
<TITLE>301 Moved</TITLE></HEAD><BODY>
<H1>301 Moved</H1>
The document has moved
<A HREF="http://www.google.com/">here</A>.
</BODY></HTML>
closing connection

Loop

SO it would seem both our problems are not the code or tool chain.

My next step will be make a header adapter board to sit the smaller ESP32 dev in to the 38pin socket. and see what it does with the shorter leads.

Its either wiring or something to do with the smaller 30pin ESP32 dev board?

This  shows  the difference in the 2 boards. Sorry about the mess and bad lighting.

[image]
 
Wiring under the module to etg card underneath.

[image]

Network side of the new board with eth card.

[image]

If it would help, I can give you the compiled hex.

I will be away for a few days so the adapter header will have to wait till I comeback.

vk3-hca, 2020-02-25:
Also weird, It does not seem to start with out a serial connection on the USB at least? Power it on 5v and no ping, plug in the usb, same, open the terminal via the usb and it starts to ping. 

Must be something to do with the serial flow control?

vk3-hca, 2020-02-28:
The 30 pin dev board mounted on the 30 to 38 pin adapter works fine in the same socket as the 38 pin version dev-board so there is no issue with the smaller board, proving its just the wiring that stopped my first board.

To add the gpio 0 to the 30 pin board I removed the pin strip plastic from the fist 2 pins an added a longer bit with one pin that lines in with the mounting hole enabling a 1 cm wire from the boot button, along with easy module removal.

[image]

The wiring length required appears to so short it makes the LAN8720 a less favorable option than just adding a Wizchip W5500 on the spi to achieve a practical board layout.

The separate device might simplify issues it like me you want eth and wifi to be independent with separate ips running and different clients at the same time.

Hope fully this exercise helps others who wire these LAN8720 modules following https://sautter.com/blog/ethernet-on-esp32-using-lan8720/ . It all works but he is not joking about wire length!

Captain-Bavaria, 2020-03-02:
Since i was on vacation last week i did not look into this problem. However i did find a soulution to my problem which is way simpler than i assumed.
I simply  misinterpreted the datasheets to identify the GPIO0 which is not wired to a pin on my board.
In order to get to the pin i hat to solder a wire to my chip and there the mistake happened, i connected the wrong pin.
Unfortunatley i got so fixed on the idea that the problem was way more complicated because i read all those issue threds here that i did not check my asumtion on GPIO0 again i simply stumbeld across the right data sheet and then noticed my obvious and dumb mistake. 
Now all code works fine.
my tipp to everyone trying this is to get proper hardwere where all the pins you need are easy to access

Captain-Bavaria, 2020-03-02:
@vk3-hca 

>  you, want eth and wifi to be independent with separate IPs running and different clients at the same time. I am yet to confirm but this does not appear to be so easy to realise with the inbuilt ESP32 EMAC.

yes this is true ESP-IDF only supports an DHCP server running on the EPS32 EMAC (a DHCP server advises a Ip-adress to every client which is connected) so if you are trying to transmit data over wifi and LAN between two devices WITHOUT using a router  on the LAN side is semes very complicated to acive.
Unfortunatley this is what i am trying to acive.

sauttefk, 2020-03-02:
> It all works but he is not joking about wire length!

Indeed I'm not kidding. This are high frequency signals, where impedance matters. Also ground-bounce and decoupling capacitors become a real issue.

vk3-hca, 2020-03-03:
Glad you have it running.
Thanks for the comment on the dhcp, I am only new to this chip the last couple of days so there is a lot to find out. If DHCP is the problem maybe it can be done in my case, because I use static IPs and avoid DHCP like the plague. Possibly just code in another DHCP client for the Eth? The wifi is already very good, leave it be.  

Each to his own but to be honest I just see a great chip that has lots of every resource but with an existing IO pin bottleneck loose about 10 primary IO pins when I could have been using a shared SPI and required 1 chip select pin with a whiz 5500 for the sake of a few extra bucks and I now am contemplating putting a STM103 beside it as an IO break out. Easy to say in hindsight.

Captain-Bavaria, 2020-03-03:
I am trying static IP right now it does get assigned properly but i can't ping it from my Pc to which i also assigned a static IP address.
```cpp

void set_ipadress()
{
	tcpip_adapter_ip_info_t info;
	memset(&info, 0, sizeof(info));
	IP4_ADDR(&info.ip, 192, 168, 1, 4);
	//IP4_ADDR(&info.gw, 192, 168, 1, 5);
	IP4_ADDR(&info.netmask, 255, 255, 255, 0);
	
    esp_err_t ret = 0x5000;
    
    ret = tcpip_adapter_dhcpc_stop(TCPIP_ADAPTER_IF_ETH); // ret=0x5000 -> tcpip_adapter_invalid_params, very old esp-idf didn't implementated this yet.
    ESP_LOGI(TAG, "dhcp client stop RESULT: %d", ret);
	
    ESP_ERROR_CHECK(tcpip_adapter_set_ip_info(TCPIP_ADAPTER_IF_ETH, &info));
    
}
```

i used this  [code](https://github.com/espressif/esp-iot-solution/tree/master/examples/eth2wifi) as my base and added my **set_ipadress** function to app_main()

```cpp
void app_main()
{
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        ESP_ERROR_CHECK( nvs_flash_erase() );
        ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK( ret );
  
    tcpip_adapter_init();

    eth_queue_handle = xQueueCreate(CONFIG_DMA_RX_BUF_NUM,  sizeof(tcpip_adapter_eth_input_t));
    xTaskCreate(eth_task, "eth_task", 2048, NULL, (tskIDLE_PRIORITY + 2), NULL);
    
    esp_event_loop_init(event_handler, NULL);

    initialise_ethernet();

    initialise_wifi();

    set_ipadress();
}
```
Maybe this does help you.

closed 2020-03-09 by Captain-Bavaria