espressif/arduino-esp32 #3736: [HardwareSerial] Lolin D32 Pro Serial Problem

JAICHANGPARK, 2020-02-13:
### Hardware:
Board:							LOLIN D32 PRO
Core Installation version:			1.0.4
IDE name:						Arduino IDE
Flash Frequency:					80Mhz?
PSRAM enabled:                                      no
Upload Speed:					115200
Computer OS:                                         Windows 10

### Description:
Now I'm using LOLIN D32 PRO module for esp32 .
And I tried to use HardwareSerial for using communication with slave device.
1. Can I use other IO PIn with Serial2? But it's not works well... 
2. If I used IO12, IO13 for HardwareSerial, the slave device was been stuck .. 

### Slave Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp
  void setup() {
  // put your setup code here, to run once:
  pinMode(5, OUTPUT);
  digitalWrite(5, LOW);
//  Serial.begin(9600);
  Serial2.begin(9600, SERIAL_8N1, 12, 13);
  while (Serial2.available()) {
    int tmp = Serial2.read();
    if (tmp == 0x80) {
      Serial2.write(0x80);
      Serial2.write(0x80);
      return;
    }
  }
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Master Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp
void setup() {
  // put your setup code here, to run once:
  Serial2.begin(9600, SERIAL_8N1, 12, 13);
  while (Serial2.available()) {
    Serial2.write(0x80);
  }
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

stickbreaker, 2020-02-13:
Do you have the TFT LCD display plugged in? 
Pin 12 is used by the display.
pin out from [WeMos LoLin D32 Pro](https://docs.wemos.cc/en/latest/_static/files/sch_d32_pro_v2.0.0.pdf)
[image]

The Only uncommitted pins are:
IO13
IO25
IO25

Use of I00,I02, IO12 or I015 will affect booting and uploading code to the ESP32.  Review [ESP32 Boot Strapping PIns](https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection) before you use them.

Chuck.

JAICHANGPARK, 2020-02-13:
@stickbreaker Thank you for comment.

No, I didn't use TFT Display. 
I try to change TX,RX pin to 13, 14. It works fine 
And I used `Serial2.flush();`

closed 2020-02-13 by JAICHANGPARK