espressif/arduino-esp32 #3739: ESP-IDF Release v4.0

joseitor, 2020-02-13:
It has just be release the new version 4.0. Are gonna be the idf of arduino-esp32 updated to it?

Suraj-singh999898, 2020-02-14:
Anyone
waiting for the answer.

atanisoft, 2020-02-14:
IDF v4 is currently only supported on the https://github.com/espressif/arduino-esp32/tree/idf-release/v4.0 branch and is not yet generally available via the Arduino IDE and it fails entirely in PlatformIO IDE.

Vzor-, 2020-02-22:
Is there an ETA on that becoming stable? I use PlatformIO ATM.

atanisoft, 2020-02-22:
There is no eta for PIO support.

schlaegerz, 2020-04-01:
Are there instructions for using the 4.0 branch with Arduino IDE. I would like to be able help with the testing of it, but I can't figure out an easy way to test it, or when to expect it to stabilize.

lbernstone, 2020-04-02:
If you are comfortable using git, follow the [development instructions](https://github.com/espressif/arduino-esp32#installation-instructions) and then checkout the idf-release/v4.0 branch.

atanisoft, 2020-04-02:
PIO now supports IDF v4 in the latest release. You can use arduino-esp32 with it as well if you follow their examples

schlaegerz, 2020-04-02:
There seems to still be a pretty bad problem with it, unless I somehow missed a setup step (building with VSCode, installed it via git)

Most of my code works well, except for wifi related parts, specifically this:

```
#include <HTTPClient.h>

void loop(){

}
void setup(){
	Serial.begin(115200);

    delay(3000);
	HTTPClient http;
	http.begin("www.google.com");
}
```

Leads to a linker error:
```
C:\Users\schla\Documents\Arduino\hardware\espressif\esp32/tools/sdk/lib\liblwip.a(dns.o):(.literal.dns_compare_name+0x4): undefined reference to `__locale_ctype_ptr'

C:\Users\schla\Documents\Arduino\hardware\espressif\esp32/tools/sdk/lib\liblwip.a(dns.o): In function `dns_compare_name':

/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/dns.c:428: undefined reference to `__locale_ctype_ptr'

/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/dns.c:428: undefined reference to `__locale_ctype_ptr'

C:\Users\schla\Documents\Arduino\hardware\espressif\esp32/tools/sdk/lib\liblwip.a(ip4_addr.o): In function `ip4addr_aton':

/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/ipv4/ip4_addr.c:160: undefined reference to `__locale_ctype_ptr'

/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/ipv4/ip4_addr.c:175: undefined reference to `__locale_ctype_ptr'

C:\Users\schla\Documents\Arduino\hardware\espressif\esp32/tools/sdk/lib\liblwip.a(ip4_addr.o):/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/ipv4/ip4_addr.c:178: more undefined references to `__locale_ctype_ptr' follow

collect2.exe: error: ld returned 1 exit status

```
It seems somehow related to this file 
https://github.com/espressif/esp-idf/blob/master/components/esp_rom/esp32s2/ld/esp32s2.rom.newlib-funcs.ld

But I don't understand what is happening.

lbernstone, 2020-04-02:
The branch is only designed for the Arduino IDE.  Things will be broken with it until it is actually released.  The code upstream has changed.  You will have to wait if you can't fix it yourself.

schlaegerz, 2020-04-02:
Ok understood, it wouldn't compile in Arduino IDE either it was making me go crazy for the last day. It was caused by the latest commit on that branch. 

I was able to actually get it working by checking out the commit before

git checkout 60f20a7869f44eda895a187c2d2376eefebc2314

closed 2020-06-16 by stale[bot]