espressif/arduino-esp32 #3740: No CaptivePortal page comes up on ESP32 Dev. board

williamesp2015, 2020-02-13:
Board:							ESP32 Dev Module
Core Installation version:			Latest
IDE name:						Platform.io Arduino
Flash Frequency:					40Mhz?
Computer OS:                                        Windows 10? And iPhone
I searched to find the resone for failing captive Portal to popup on the Chrome/Internet Explorer /iphone/Adnroid Tablet but I couldn't solve. The code is:
#ifdef ESP8266
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
ESP8266WebServer server(80);
#else
#include <WiFi.h>
#include <WiFiClient.h>
#include <WebServer.h>
#include <ESPmDNS.h>
#include <DNSServer.h>
WebServer server(80);
DNSServer dnsServer;
#endif
const byte DNS_PORT = 53;
IPAddress apIP(8, 8, 4, 4);
//IPAddress apIP(192, 168, 4, 4);

String responseHTML = ""
  "<!DOCTYPE html><html><head><title>CaptivePortal</title></head><body>"
  "<h1>Hello World!</h1><p>This is a captive portal example. All requests will "
  "be redirected here.</p></body></html>";
  void handleRoot() {
  //  digitalWrite(led, 1);
  #ifdef ESP8266
    server.send(200, "text/plain", "hello from esp8266!");
  #else
    server.send(200, "text/plain", "hello from esp32!");
  #endif
  //  digitalWrite(led, 0);
  }
  void handleNotFound(){
//  digitalWrite(led, 1);
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET)?"GET":"POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";
  for (uint8_t i=0; i<server.args(); i++){
    message += " " + server.argName(i) + ": " + server.arg(i) + "\n";
  }
  server.send(404, "text/plain", message);
//  digitalWrite(led, 0);
}
void setup() {
  Serial.begin(115200);
  delay(2000);
  WiFi.disconnect();   //added to start with the wifi off, avoid crashing
  delay(500);
  WiFi.mode(WIFI_OFF); //added to start with the wifi off, avoid crashing
  WiFi.mode(WIFI_AP);
  delay(100);
  WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));
  WiFi.softAP("CaptivePortal");
  delay(2000); // VERY IMPORTANT
  #ifdef ESP8266
    if (MDNS.begin("esp8266")) {
  #else
    if (MDNS.begin("esp32")) {
   #endif
    Serial.println("MDNS responder started");
  }
  server.on("/", handleRoot);
  server.on("/inline", [](){
    server.send(200, "text/plain", "this works as well");
  });
  server.onNotFound(handleNotFound);
  server.begin();
  Serial.println("HTTP server started");
}
void loop(void){
  server.handleClient();
    delay(1);
}

mntolia, 2020-02-18:
There are 3 ways captive portal detectors will get triggered. The easiest is an HTTP redirect for all requests and a DNS server. I have generally found that using both the implementations triggers the captive portal detector in all devices. 

Make the server.onNotFound() redirect to the captive portal.

Make sure the DNS resolver resolves to the said IP. BTW 8.8.4.4 is the IP address for google DNS. 

Check this example for DNS redirection:
https://github.com/espressif/arduino-esp32/blob/master/libraries/DNSServer/examples/CaptivePortal/CaptivePortal.ino

G6EJD, 2020-04-10:
The example you listed @mntolia does not work, there is an issue in the Dnsserver library

mntolia, 2020-04-11:
> The example you listed @mntolia does not work, there is an issue in the Dnsserver library
I am currently using it in a project. It works fine for me. I didn't implement that example exactly though.

G6EJD, 2020-04-11:
How did it differ?

mntolia, 2020-04-11:
I am using this web server library:
https://github.com/me-no-dev/ESPAsyncWebServer

This is my setup:
```
    dnsServer.setErrorReplyCode(DNSReplyCode::NoError);
    dnsServer.start(53, "*", apIP);
```
On the AsyncWebServer I redirected all 404s to the defined IP address.
The captive portals pop up on every device I tried(MacOS, iPad, Android, Windows)

G6EJD, 2020-04-11:
Yes, that one works well, but the in-built variant / dnsserver doesn’t so it’s still an issue.

gluehbirnenkopf, 2020-08-23:
I agree that this is not a captive portal example and as the other issue comments say this is 

A) poorly named
B) if the name stays the functionality should work

closed 2020-09-30 by me-no-dev

mariano22, 2021-08-31:
Maybe this comment would help to others.
I tried the code firstly posted and it worked on PC but no completly good in Android (version 10 kernal 4.4.205-perf+). It doesn't trigger the "Touch it  to log in the network" message.

[image]
[image]

By adding a handler for the error case on server was solved:

`server.onNotFound(handleRoot);`

mariano22, 2021-08-31:
Maybe this comment would help to others.
I tried the code firstly posted and it worked on PC but no completly good in Android (version 10 kernal 4.4.205-perf+). It doesn't trigger the "Touch it to log in the network" message.

By adding a handler for the error case on server was solved:

`server.onNotFound(handleRoot);`

PR #4080 (2020-09-30): Update CaptivePortal.ino
Illegal SSID used for SoftAP. Fixes #3740