espressif/arduino-esp32 #3745: Arduino IDE 1.8.12 & ESP32 : fatal error: bits/cpu_defines.h: No such file or directory

MichelBen, 2020-02-14:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			14/02/2020
IDE name:							Arduino IDE 1.8.112
Flash Frequency:					
PSRAM enabled:                                          
Upload Speed:						
Computer OS:                                            Windows 10

### Description:
I've just 
- installed Arduino IDE 1.8012
- create a subdirectory "portable ...."
- add in preference "https://arduino.esp8266.com/stable/package_esp8266com_index.json,https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json"
- add ESP32 with Board manager
- choose "ESP32 Dev Module" as board
- restarted Arduino IDE

When I try to compile an empty sketch 
```
void setup() {
  // put your setup code here, to run once:
}
void loop() {
  // put your main code here, to run repeatedly:
}
```
I have this error

**fatal error: bits/cpu_defines.h: No such file or directory**

I've solves I had to copy
 portable\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\ **include\c++\5.2.0\xtensa-esp32-elf\bits**
to
 portable\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\ **include\c++\5.2.0\bits**

I hope 

- this will help other maker
- installation scrip will be corrected

**_Michel_**

closed 2020-04-28 by stale[bot]

MolecularEngine, 2022-08-11:
Is this a file location?

Mike-Morrell, 2022-09-20:
Thanks. Had same problem with 8.4.0 and this did the trick.

fjmsouza, 2023-09-06:
Thanks, worked fine to 8.4.0 too

mikrocoder, 2023-12-12:
Hi,

I was also affected by the problem. The basic problem is the path length. That is exactly the problem. The folder depth of the ESP32 core is very deep and then there is the own structure. Then it crashes.
I had tried it with IDE 1.8.19 and IDE 2.2.1 under Windows 10 and 11. The installed IDE 2.2.1 does not cause any problems. The portable 1.8.19 causes problems from a reproducible path length. Because this varies depending on the user, the problem occurs rather rarely and can also affect different files regarding "bits" folders.

After unpacking you have the folder "**arduino-1.8.19**".
I usually put all portable 1.8.19 in the path
``C:\Arduino IDE Portable\Name\arduino-1.8.19\...``

In this case, this was now labeled
``C:\Arduino IDE Portable\NanoESP32\arduino-1.8.19\...``
too long. Shortened to
``C:\Arduino IDE Portab\NanoESP32\arduino-1.8.19\...``
everything is fine again. With one more character in front of it, it's over. Spaces are no problem.

If I were to leave the full path length, I would have to copy all files from one bits folder to the other bits folder. Copying just one file was not enough.
from:
``\arduino-1.8.19\portable\packages\esp32\tools\xtensa-esp32s3-elf-gcc\esp-2021r2-patch5-8.4.0\xtensa-esp32s3-elf\include\c++\8.4.0\xtensa-esp32s3-elf\no-rtti\bits``
to:
``\arduino-1.8.19\portable\packages\esp32\tools\xtensa-esp32s3-elf-gcc\esp-2021r2-patch5-8.4.0\xtensa-esp32s3-elf\include\c++\8.4.0\bits ``