espressif/arduino-esp32 #3747: ESP32 I2C LIDARLite issue

SkeeterOG, 2020-02-17:
### Description:

I've been attempting to run my LIDARLite v3 with my ESP32 via I2C connections (Arduino IDE).  So far I've been completely unsuccessful.  I've had this Lidar running on multiple other boardsvia I2C (Arduino Uno, Nano, Adafruit Feather M0), and so far the ESP32 is the only one I can't work with.  I use the standard example given with the Lidar (below), and modify from there typically.  I attempted to follow the guidance on issue https://github.com/espressif/arduino-esp32/issues/2949 but was not able to make it work.  I also followed the guidance here
 https://www.youtube.com/watch?v=qoc7N6Yg5jk  (3 mins)
Both of these solutions touched on the same issue: the Wire.begin() function causes some problems with ESP32.  I attempted to comment out the Wire.begin() in the LIDARLite.cpp file and add my own into the main arduino code but this didn't work.  Other solutions I tried were adding Wire.begin(21,22,XxX) where XxX is the I2C bus speed, changing it from either 100K or 400K; nothing.  The LIDARLite v3 has internal pull-up resistors, but I also added 3K pull-up resistors at one point thinking that this might help (it did not).  The output is typically a number representing the lidar's distance measurement, but so far the only output I get is "> nack" which represents a problem with the lidar (not specific, I know), which is fairly common when editing code or switching lidar wires around.  I've also tried switching out the ESP32 boards, and doing basic examples with them to make sure that they're working (they are).
Help me Obi-Wan Kenobi, you're my only hope!  (Although I suspect @stickbreaker will be helpful as well)

### Hardware:
Board: DOIT ESP32 DEVKIT V1
Core Installation version: 1.0.2
IDE name: Arduino IDE 1.8.9
Flash Frequency: 80 MHz
PSRAM enabled: NO
Upload Speed: 115200
Computer OS: Windows 10

### Sketch: 

/*------------------------------------------------------------------------------

  LIDARLite Arduino Library
  v3/GetDistanceI2c

  This example shows how to initialize, configure, and read distance from a
  LIDAR-Lite connected over the I2C interface.

  Connections:
  LIDAR-Lite 5 Vdc (red) to Arduino 5v
  LIDAR-Lite I2C SCL (green) to Arduino SCL
  LIDAR-Lite I2C SDA (blue) to Arduino SDA
  LIDAR-Lite Ground (black) to Arduino GND

  (Capacitor recommended to mitigate inrush current when device is enabled)
  680uF capacitor (+) to Arduino 5v
  680uF capacitor (-) to Arduino GND

  See the Operation Manual for wiring diagrams 
  and more information:
  http://static.garmin.com/pumac/LIDAR_Lite_v3_Operation_Manual_and_Technical_Specifications.pdf

------------------------------------------------------------------------------*/

#include <Wire.h>
#include <LIDARLite.h>

LIDARLite myLidarLite;

void setup()
{
  Serial.begin(115200); // Initialize serial connection to display distance readings

  /*
    begin(int configuration, bool fasti2c, char lidarliteAddress)

    Starts the sensor and I2C.

    Parameters
    ----------------------------------------------------------------------------
    configuration: Default 0. Selects one of several preset configurations.
    fasti2c: Default 100 kHz. I2C base frequency.
      If true I2C frequency is set to 400kHz.
    lidarliteAddress: Default 0x62. Fill in new address here if changed. See
      operating manual for instructions.
  */
  myLidarLite.begin(0, true); // Set configuration to default and I2C to 400 kHz

  /*
    configure(int configuration, char lidarliteAddress)

    Selects one of several preset configurations.

    Parameters
    ----------------------------------------------------------------------------
    configuration:  Default 0.
      0: Default mode, balanced performance.
      1: Short range, high speed. Uses 0x1d maximum acquisition count.
      2: Default range, higher speed short range. Turns on quick termination
          detection for faster measurements at short range (with decreased
          accuracy)
      3: Maximum range. Uses 0xff maximum acquisition count.
      4: High sensitivity detection. Overrides default valid measurement detection
          algorithm, and uses a threshold value for high sensitivity and noise.
      5: Low sensitivity detection. Overrides default valid measurement detection
          algorithm, and uses a threshold value for low sensitivity and noise.
    lidarliteAddress: Default 0x62. Fill in new address here if changed. See
      operating manual for instructions.
  */
  myLidarLite.configure(0); // Change this number to try out alternate configurations
}

void loop()
{
  /*
    distance(bool biasCorrection, char lidarliteAddress)

    Take a distance measurement and read the result.

    Parameters
    ----------------------------------------------------------------------------
    biasCorrection: Default true. Take aquisition with receiver bias
      correction. If set to false measurements will be faster. Receiver bias
      correction must be performed periodically. (e.g. 1 out of every 100
      readings).
    lidarliteAddress: Default 0x62. Fill in new address here if changed. See
      operating manual for instructions.
  */

  // Take a measurement with receiver bias correction and print to serial terminal
  Serial.println(myLidarLite.distance());

  // Take 99 measurements without receiver bias correction and print to serial terminal
  for(int i = 0; i < 99; i++)
  {
    Serial.println(myLidarLite.distance(false));
  }
}

### Debug Messages:
```
> nack
> nack
> nack
> nack

```
[image]

stickbreaker, 2020-02-17:
@SkeeterOG Does the LIRDARlite have the pullups to 5v?

connect your pullups on the ESP32 side to 3.3v use 3.3k ohm.

Diagnostic ideas:
* try this [i2c Scan sketch](https://github.com/stickbreaker/arduino-esp32/tree/master/libraries/Wire/examples/i2c_scan)
* Enable Core debugging.  Set CORE DEBUG LEVEL to INFO
* Enable debug buffers in `esp32-hal-i2c.c`. 
uncomment line 45:
`//#define ENABLE_I2C_DEBUG_BUFFER`
On Windows 7, installed with BoardManager this file is here
`C:\Users\{userName}\AppData\Local\Arduino15\packages\esp32\hardware\esp32\{version}\cores\esp32`
* rerun your tests and posts the debug log.

Chuck.

SkeeterOG, 2020-02-17:
It works now.
You know how you'll be working on something and it won't work, but the second you bring it to a professional it suddenly works fine?  That's how I feel here.  I uncommented line 45, saved it, tested it, it works.  I then re-commented line 45, saved it, and it still works.  I am mystified as to what I was doing wrong, as I've been working on this problem for two weeks. 
I'm very sorry to have wasted your time, but thank you for whatever magical powers you infused into my project.

closed 2020-02-17 by SkeeterOG

abedin17, 2024-02-22:
@SkeeterOG : is it possible to share  your updated code. I am having the same issue. I am using esp32 s3 dev module or esp32 wroom DA module

SkeeterOG, 2024-02-23:
@abedin17 I do not have access to that software anymore unfortunately.  I don't remember that I did anything software related to fix it though.  I think it was a voltage supply issue, or the lidar and ESP32 were not sharing the same ground.