espressif/arduino-esp32 #3751: cannot use WiFi.begin() after ETH.begin()

aliyehia92, 2020-02-19:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.4
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                     no
Upload Speed:					921600
Computer OS:                                        Windows 10

### Description:
I have an LAN8720 hooked up to an ESP32 exactly as described here https://sautter.com/blog/ethernet-on-esp32-using-lan8720/

If I'm using either the WiFi or ETH interface on its own it connects perfectly fine. However, if I use ETH.begin() followed by WiFi.begin() , 3 times out of 5 the WiFi will never connect and will remain stuck in the while loop. 

The sketch is really simple:

### Sketch: 
```cpp

#include <Arduino.h>

void setup() {
	Serial.begin(115200);
        ETH.begin(PHY1, 17, 23, 18, ETH_PHY_LAN8720, ETH_CLOCK_GPIO0_IN);
	WiFi.begin("SSID", "PASSWORD");

	while (WiFi.status() != WL_CONNECTED) {
		delay(500);
		Serial.print(".");
	}

	Serial.print("WIFI CONNECTED WITH IP ");
	Serial.println(WiFi.localIP());
}

void loop() {

}
```

### Debug Messages:
```
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 21 - ETH_STOP
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY 
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 4 - STA_CONNECTED 
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 15 - 4WAY_HANDSHAKE_TIMEOUT 
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 23 - ETH_DISCONNECTED 
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 25 - MAX 
[D][WiFiGeneric.cpp:417] _eventCallback(): ETH IP: 192.168.16.19, MASK: 255.255.255.0, GW: 192.168.16.1 
```

Captain-Bavaria, 2020-02-19:
Could you describe your setup with a bit more detail because i tried your code and it doesnt work at all i only get this error: **E (2143) emac: Reset EMAC Timeout**
i also got the hardware setup from  https://sautter.com/blog/ethernet-on-esp32-using-lan8720/

aliyehia92, 2020-02-19:
When you say that the code doesn't work at all do you mean that it doesn't compile or that it doesn't cause the same issues for you?

btw I forgot the two includes below in my above code, apologies for that:

```
#include <ETH.h>
#include <WiFi.h>
```

Captain-Bavaria, 2020-02-19:
It doesn't cause your described issue in fact its even worse because in your example it at least connects to the ETH interface. 
Which i have not achieved yet that's why i wanted to kow some more deatils about your setup.

aliyehia92, 2020-02-19:
I wish there were more details but it's really as simple as I described it. I followed the steps on the sautter site and just like that I got the required hardware and I was able to upload any sketch that uses ETH successfully. I haven't even experimented a lot with it, just when I decided to start both interfaces it started causing issues. Also it's worth noting that if I put Eth.begin() after the while loop ( after WiFi connects successfully) the issue goes away. And I know that it's not a power issue because usually if it's power-related it reboots or throws some exception.

closed 2020-05-03 by stale[bot]