espressif/arduino-esp32 #3754: [E][ssl_client.cpp:87] start_ssl_client(): Connect to Server failed!

simo89ng, 2020-02-21:
Hi,
Some time i have this error:

 [E][ssl_client.cpp:87] start_ssl_client(): Connect to Server failed!
 [E][WiFiClientSecure.cpp:132] connect(): start_ssl_client: -1
How can i fix?
Many thanks

simo89ng, 2020-02-26:
Im using the example /examples/WiFiClientSecure/WiFiClientSecure.ino code that call my server, sometime return a ssl connection fail, one time every 100 fine connection. why? 
thankyou

closed 2020-05-10 by stale[bot]

DAVIDBETAN2000, 2020-09-14:
este problema ocurre en vs code pero no en arduino, he colocado librerias de arduino en vs code y sigue el mismo problema con esp32

n1lby73, 2023-07-05:
> este problema ocurre en vs code pero no en arduino, he colocado librerias de arduino en vs code y sigue el mismo problema con esp32

got any idea about this

[  2546][E][ssl_client.cpp:37] _handle_error(): [start_ssl_client():273]: (-9984) X509 - Certificate verification failed, e.g. CRL, CA or signature check failed
[  2550][E][WiFiClientSecure.cpp:144] connect(): start_ssl_client: -9984

this error persist despite that certificate is correct