espressif/arduino-esp32 #3760: Compilation error: conversion from 'const ip_addr_t {aka const ip_addr}' to non-scalar type 'ip_addr_t {aka ip_addr}' requested

GianCann, 2020-02-22:
What you are trying to do? **Compile my esp32-arduino library with sdkconfig custom value (i need more of 3 BLE Max Connection)**
Describe your system: PC with Ubuntu desktop
Describe what is failing.

I try to compiling esp32-arduino library following this guide:
https://github.com/espressif/arduino-esp32/blob/master/docs/lib_builder.md

I get this error:

/home/gianluca/esp32-arduino-lib-builder/components/arduino/libraries/WiFi/src/WiFiSTA.cpp: In member function 'IPAddress WiFiSTAClass::dnsIP(uint8_t)':
/home/gianluca/esp32-arduino-lib-builder/components/arduino/libraries/WiFi/src/WiFiSTA.cpp:491:37: error: conversion from 'const ip_addr_t {aka const ip_addr}' to non-scalar type 'ip_addr_t {aka ip_addr}' requested **
ip_addr_t dns_ip = dns_getserver(dns_no);
^
/home/gianluca/esp32-arduino-lib-builder/components/arduino/libraries/WiFi/src/ETH.cpp: In member function 'IPAddress ETHClass::dnsIP(uint8_t)':
/home/gianluca/esp32-arduino-lib-builder/components/arduino/libraries/WiFi/src/ETH.cpp:196:37: error: conversion from 'const ip_addr_t {aka const ip_addr}' to non-scalar type 'ip_addr_t {aka ip_addr}' requested**
ip_addr_t dns_ip = dns_getserver(dns_no);

Where is the problem?

ivanporro, 2020-02-25:
Ciao :-)

Try this:

In file: 
/home/..../components/arduino/libraries/WiFi/src/WiFiSTA.cpp:491
`
IPAddress WiFiSTAClass::dnsIP(uint8_t dns_no)
{
    if(WiFiGenericClass::getMode() == WIFI_MODE_NULL){
        return IPAddress();
    }
    // ip_addr_t dns_ip = dns_getserver(dns_no);
    // return IPAddress(dns_ip.u_addr.ip4.addr);
    const ip_addr_t * dns_ip = dns_getserver(dns_no);
    return IPAddress(dns_ip->u_addr.ip4.addr);    
}
`
In file:
/home/..../components/arduino/libraries/WiFi/src/ETH.cpp:196

`IPAddress ETHClass::dnsIP(uint8_t dns_no)
{
    //ip_addr_t dns_ip = dns_getserver(dns_no);
    //return IPAddress(dns_ip.u_addr.ip4.addr);
    const ip_addr_t * dns_ip = dns_getserver(dns_no);
    return IPAddress(dns_ip->u_addr.ip4.addr);
}
`

This fix is from version 4.0 of the arduino-esp32 found here:
https://github.com/espressif/arduino-esp32/blob/idf-release/v4.0/libraries/WiFi/src/WiFiSTA.cpp

If you get the error "Please configure IDF framework to include mbedTLS -> Enable pre-shared-key ciphersuites and activate at least the path in menuconfig is:

component config > mbedTLS > TLS key exchange method > enable pre-shared keys & enable psk

Ivan

closed 2020-05-10 by stale[bot]

copercini, 2020-06-16:
@me-no-dev I think this fix should be included in master, because it generates a build error when using arduino as component of IDF 3.3.2 without it