espressif/arduino-esp32 #3761: PSRAM not being used

Roydon, 2020-02-22:
### Hardware:
Board:							ESP32-CAM
Core Installation version:			1.0.4
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                      yes
Upload Speed:					115200
Computer OS:                                          Mac OSX

### Description:
I am trying to capture an image on ESP32-CAM and do base64 encoding. Not printing output of ```base64::encode(fb->buf, fb->len)```
PSRAM is getting detected but its not getting used.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```
#include "WiFi.h"
#include "esp_camera.h"
#include "esp_timer.h"
#include "img_converters.h"
#include "Arduino.h"
#include "soc/soc.h"           // Disable brownour problems
#include "soc/rtc_cntl_reg.h"  // Disable brownour problems
#include "driver/rtc_io.h"
#include <StringArray.h>

#include <PubSubClient.h>
#include <base64.h>
#include <libb64/cencode.h>

// Replace with your network credentials
const char* ssid = "###";
const char* password = "$$$";

// Add your MQTT Broker IP address, example:
const char* mqtt_server = "###.cloudmqtt.com";
const int mqtt_port = 11073;
const char* mqtt_user = "###";
const char* mqtt_password = "###";

#define SLEEP_DELAY 10000 //Delay ofr 10 Sec
#define FILE_PHOTO "/photo.jpg"

// OV2640 camera module pins (CAMERA_MODEL_AI_THINKER)
#define PWDN_GPIO_NUM     32
#define RESET_GPIO_NUM    -1
#define XCLK_GPIO_NUM      0
#define SIOD_GPIO_NUM     26
#define SIOC_GPIO_NUM     27
#define Y9_GPIO_NUM       35
#define Y8_GPIO_NUM       34
#define Y7_GPIO_NUM       39
#define Y6_GPIO_NUM       36
#define Y5_GPIO_NUM       21
#define Y4_GPIO_NUM       19
#define Y3_GPIO_NUM       18
#define Y2_GPIO_NUM        5
#define VSYNC_GPIO_NUM    25
#define HREF_GPIO_NUM     23
#define PCLK_GPIO_NUM     22

WiFiClient mqttClient;
PubSubClient client(mqttClient);

const int LED_BUILTIN = 4;

void setup_camera() {
      // Turn-off the 'brownout detector'
      WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0);

      // OV2640 camera module
      camera_config_t config;
      config.ledc_channel = LEDC_CHANNEL_0;
      config.ledc_timer = LEDC_TIMER_0;
      config.pin_d0 = Y2_GPIO_NUM;
      config.pin_d1 = Y3_GPIO_NUM;
      config.pin_d2 = Y4_GPIO_NUM;
      config.pin_d3 = Y5_GPIO_NUM;
      config.pin_d4 = Y6_GPIO_NUM;
      config.pin_d5 = Y7_GPIO_NUM;
      config.pin_d6 = Y8_GPIO_NUM;
      config.pin_d7 = Y9_GPIO_NUM;
      config.pin_xclk = XCLK_GPIO_NUM;
      config.pin_pclk = PCLK_GPIO_NUM;
      config.pin_vsync = VSYNC_GPIO_NUM;
      config.pin_href = HREF_GPIO_NUM;
      config.pin_sscb_sda = SIOD_GPIO_NUM;
      config.pin_sscb_scl = SIOC_GPIO_NUM;
      config.pin_pwdn = PWDN_GPIO_NUM;
      config.pin_reset = RESET_GPIO_NUM;
      config.xclk_freq_hz = 20000000;
      config.pixel_format = PIXFORMAT_JPEG;

      if (psramFound()) {
        config.frame_size = FRAMESIZE_UXGA;
        config.jpeg_quality = 10;
        config.fb_count = 2;
      } else {
        config.frame_size = FRAMESIZE_SVGA;
        config.jpeg_quality = 12;
        config.fb_count = 1;
      }
      // Camera init
      esp_err_t err = esp_camera_init(&config);
      if (err != ESP_OK) {
        Serial.printf("Camera init failed with error 0x%x", err);
        ESP.restart();
      }
  }

// Capture Photo and Save it to SPIFFS
void capturePhoto( void ) {
  camera_fb_t * fb = NULL; // pointer
  bool ok = 0; // Boolean indicating if the picture has been taken correctly

  // Take a photo with the camera
  Serial.println("Taking a photo...");

  fb = esp_camera_fb_get();
  if (!fb) {
    Serial.println("Camera capture failed");
    return;
  }

  Serial.print("Heap Size : ");
  Serial.println(ESP.getFreeHeap());
  //Serial.println(fb->format);
  Serial.print("fb len : ");
  Serial.println(fb->len);
  Serial.print("base64 encode expected len : ");
  Serial.println(base64_encode_expected_len(fb->len) + 1);
  String base64image = base64::encode(fb->buf, fb->len);
  Serial.print("base64 image : ");
  Serial.println(base64image);
  esp_camera_fb_return(fb);
}

void setup_wifi() {
  delay(10);
  // We start by connecting to a WiFi network
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.print("IP address : ");
  Serial.println(WiFi.localIP());
}

void reconnect() {
  while (!client.connected()) {
    Serial.print("Attempting MQTT connection...");
    if (client.connect("ESP32Client", mqtt_user, mqtt_password)) {
      Serial.println("connected");
    } else {
      Serial.print("failed, rc=");
      Serial.print(client.state());
      Serial.println(" try again in 5 seconds");
      delay(5000);
    }
  }
}

void setup() {
  Serial.begin(115200);
  byte* psdRamBuffer = (byte*)ps_malloc(500000);
  setup_camera();
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
  setup_wifi();
  client.setServer(mqtt_server, mqtt_port);
}

// the loop function runs over and over again forever
void loop() {
  Serial.println("PSRAM found: " + String(psramFound()));
  
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
  capturePhoto();
  if (!client.connected()) {
    reconnect();
  }
  client.loop();
  delay(SLEEP_DELAY);
}
```

### Debug Messages:
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
[D][esp32-hal-psram.c:47] psramInit(): PSRAM enabled

Connecting to ###
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.1.247, MASK: 255.255.255.0, GW: 192.168.1.1
.
WiFi connected
IP address : 192.168.1.247
PSRAM found: 1
Taking a photo...
Heap Size : 187408
fb len : 104596
base64 encode expected len : 139465
base64 image : -FAIL-
Attempting MQTT connection...connected
```

closed 2020-05-07 by stale[bot]

martinius96, 2020-05-25:
@Roydon Hello there, were you sucessful? 
I am working on similar project where I need to use PSRAM for capture photo that I want to send via WebClient to server.

Roydon, 2020-05-25:
Was not able to convert entire image to base64 and publish it to MQTT, workaround I is by breaking base64 encoded image into parts and publishing separately to mqtt with past sequence no so it can be reconstructed at receiving side.

iwas108, 2022-04-02:
any update on how to use PSRAM to publish base64 image?

VojtechBartoska, 2022-04-04:
@iwas108 Do you still face this issue on 2.0.3-RC1 version?

iwas108, 2022-04-20:
> @iwas108 Do you still face this issue on 2.0.3-RC1 version?

Its clear, i have done and able to send upto 512KB of base64 image from AI Thinker ESP32-Cam (4MB PSRAM) via MQTT. This is special case due to IoT infrastructure that is only accept communication via MQTT. Here is my final code:

```
void myTask()
{
  if(tb.connected())
  {
    camera_config_t cam;
    cam.ledc_channel = LEDC_CHANNEL_0;
    cam.ledc_timer = LEDC_TIMER_0;
    cam.pin_d0 = Y2_GPIO_NUM;
    cam.pin_d1 = Y3_GPIO_NUM;
    cam.pin_d2 = Y4_GPIO_NUM;
    cam.pin_d3 = Y5_GPIO_NUM;
    cam.pin_d4 = Y6_GPIO_NUM;
    cam.pin_d5 = Y7_GPIO_NUM;
    cam.pin_d6 = Y8_GPIO_NUM;
    cam.pin_d7 = Y9_GPIO_NUM;
    cam.pin_xclk = XCLK_GPIO_NUM;
    cam.pin_pclk = PCLK_GPIO_NUM;
    cam.pin_vsync = VSYNC_GPIO_NUM;
    cam.pin_href = HREF_GPIO_NUM;
    cam.pin_sscb_sda = SIOD_GPIO_NUM;
    cam.pin_sscb_scl = SIOC_GPIO_NUM;
    cam.pin_pwdn = PWDN_GPIO_NUM;
    cam.pin_reset = RESET_GPIO_NUM;
    cam.xclk_freq_hz = 20000000;
    cam.pixel_format = PIXFORMAT_JPEG;
    cam.fb_count = 2;
    cam.frame_size = (framesize_t)mySettings.frameSize;
    cam.jpeg_quality = mySettings.jpegQuality;

    sprintf_P(logBuff, PSTR("Taking snap (jpegQuality: %d, frameSize: %d), please wait."), cam.jpeg_quality, (int)cam.frame_size);
    recordLog(1, PSTR(__FILE__), __LINE__, PSTR(__func__));

    // camera init
    esp_err_t err = esp_camera_init(&cam);
    if (err != ESP_OK) {
      sprintf_P(logBuff, PSTR("Camera init failed with error 0x%x"), err);
      recordLog(1, PSTR(__FILE__), __LINE__, PSTR(__func__));
      esp_camera_deinit();
    }
    delay(5000);
    camera_fb_t * fb = NULL;
    fb = esp_camera_fb_get();

    if (!fb) {
      sprintf_P(logBuff, PSTR("Camera capture failed."));
      recordLog(1, PSTR(__FILE__), __LINE__, PSTR(__func__));
    }
    else
    {
      size_t size = base64_encode_expected_len(fb->len) + 1;
      char * buffer = (char *) ps_malloc(size);
      if(buffer) {
        base64_encodestate _state;
        base64_init_encodestate(&_state);
        int len = base64_encode_block((const char *) &fb->buf[0], fb->len, &buffer[0], &_state);
        len = base64_encode_blockend((buffer + len), &_state);
      }
      esp_camera_fb_return(fb);
      esp_camera_deinit();

      SpiRamJsonDocument doc(size + 64);
      doc["b64"] = buffer;
      free(buffer);
      doc["w"] = fb->width;
      doc["h"] = fb->height;
      doc["f"] = fb->format;

      size_t finalDataSize = measureJson(doc);
      byte * data = (byte *) ps_malloc(finalDataSize);
      serializeJson(doc, data, finalDataSize);
      doc.clear();

      sprintf_P(logBuff, PSTR("Snap size %d.  Memory: %d/%d - %d. Sending now..."), finalDataSize, heap_caps_get_free_size(MALLOC_CAP_32BIT), ESP.getPsramSize(), ESP.getFreeHeap());
      recordLog(1, PSTR(__FILE__), __LINE__, PSTR(__func__));

      const char* topic = "v1/devices/me/telemetry";
      publishMqtt(topic, data, finalDataSize);
      free(data);
    }
  }
}

void publishMqtt(const char* channel, uint8_t *data, uint32_t len) {
  unsigned long start_ts = millis();

  tb.beginPublish(channel, len, false);

  size_t res;
  uint32_t offset = 0;
  uint32_t to_write = len;
  uint32_t buf_len;
  do {
    buf_len = to_write;
    if (buf_len > mySettings.tempBuffSize){
      buf_len = mySettings.tempBuffSize;
    }
    res = tb.write(data+offset, buf_len);
    offset += buf_len;
    to_write -= buf_len;
  } while (res == buf_len && to_write > 0);

  tb.endPublish();

  sprintf_P(logBuff, PSTR("Finished: (size %d bytes, %d bytes written in %ld ms)\n"), len, len-to_write, millis()-start_ts);
  recordLog(1, PSTR(__FILE__), __LINE__, PSTR(__func__));
}
```

I'm using PubSubClient as MQTT library, and do a little modification on max header size from uint16_t to uint_32t. For now it working as expected.