espressif/arduino-esp32 #3762: GPIO16 and 17

kriswiner, 2020-02-23:
I am using the ESP32 PICO D4 in a custom board design based on the ESP32Eye design, among others, which all work fine. But I have not been able to make use of GPIO16 and 17. I just realized that none of the board variants (including ESP32 Dev Module that I have been using) lists GPIO16 or 17 in the pins_arduino.h file. So no wonder my attempts to blink and led with these pins has never worked.

Is there someway I could add these pins to the pins_arduino.h list without messing up the other pin functions? And why aren't these pins typically included in most f not all of the ESP32 dev board variants?

stickbreaker, 2020-02-23:
GPIO16 and 17 are standard pins, UNLESSf you have PSRAM. The WROVER base ESP32 use those pins, so like the GPIO6..11 they are off limits.

Chuck.

lbernstone, 2020-02-23:
Pins 16 and 17 are used by the internal flash in Pico chips.
https://dl.espressif.com/dl/schematics/esp32-pico-kit-v4_schematic.pdf

kriswiner, 2020-02-23:
Thanks, that explains why they are not included in the pin list...

G6EJD, 2020-04-10:
@kriswinder, should you not close this issue now you know what the problem was?

kriswiner, 2020-04-10:
OK

closed 2020-04-10 by kriswiner

mkeyno, 2022-04-11:
I use wroom and both pins are free, however, when I use simple lrdc or timer library, they work fine, but if use extra timer they are not working
Does anybody knows which peripheral or API affect this pin functionality

mkeyno, 2022-04-11:
I use wroom and both pins are free, however, when I use simple lrdc or timer library, they work fine, but if use extra timer they are not working
Does anybody know which peripheral or API affects this pin functionality?