espressif/arduino-esp32 #3765: Embed littleFS

laercionit, 2020-02-25:
Hello .. Good night.
I've been designing an aquarium controller for a few years.
I currently use ESP8266, the code has already been ported and adapted to run on ESP32. Fantastic speed and differences between MCUs.

I'm still using SPIFFS in ESP32, because I didn't see anything native to LITTLEFS. By my studies it is more reliable and I have been using it for some time on ESP8266.

Any chance of LITTLEFS being incorporated into the CORE ARDUINO ESP32?

beegee-tokyo, 2020-02-27:
Not sure if anybody works on that. 
You can try the [esp32_littleflash](https://github.com/lllucius/esp32_littleflash)
But I didn't try that yet.

laercionit, 2020-02-27:
LittleFS is much more robust than SPIFFS, does not lose formatting due to power failure.
Currently it is already incorporated in ESP8266, as it is something far superior and I see that the ESP32 CORE has advanced, I found it coherent to know if there were plans to develop native Data Upload by IDE and even partition with this type of configuration.

zekageri, 2020-03-19:
It would be cool if someone would port it to esp32. SPIFFS is limited if we use 16mb esp32s. I have to use a 7mb ota 7mb app and 1mb flash because the slowness of the spiffs. :(

9H5G, 2020-05-29:
It would be very welcome - not least for code portability reasons. :)

sblantipodi, 2020-06-14:
since SPIFFS has been deprecated on ESP8266, LittleFS should be integrated in ESP32.
there is no reason to deprecate something on ESP8266 and leave it live on ESP32.

lbernstone, 2020-06-14:
While it would be nice to have a port, littleFS is not supported in esp-idf, so is unlikely to become canon here, and it is not part of Arduino compatibility (the purpose of this project).
The beauty of the Arduino model is that 3rd party libraries can come and go without putting the onus of maintenance on the chip manufacturers.

sblantipodi, 2020-06-14:
and what is the sense of deprecating SPIFFS on ESP8266 and not on ESP32?

lbernstone, 2020-06-14:
You will need to ask them.  It may be deprecated on esp-idf eventually and replaced with LittleFS.  At that point, it will be trivial to add to arduino.  The two projects are not welded together.

me-no-dev, 2020-06-15:
wait... why should we care that 8266 crew deprecated SPIFFS and somehow follow them? I am really confused by such way of thinking. They did not bother to ask or discuss this with us. SPIFFS is supported in IDF and is supported here. LittleFS has nothing to do with it.

sblantipodi, 2020-06-15:
are esp8266 and esp32 two separate products?
aren't both espressif products? why do they have different support?
thanks

me-no-dev, 2020-06-15:
8266 Arduino is community supported. Everything else is Espressif. So, 8266 has and does it's own things when we talk Arduino. They do not sync with us and we do not with them. They run NonOS and we run FreeRTOS.

lorol, 2020-06-17:
Hi, I managed to make it as a lib, please see this and comment your thoughts:

https://github.com/lorol/arduino-esp32littlefs-plugin
https://github.com/lorol/LITTLEFS

Thank you

luc-github, 2020-06-17:
@lorol so littleFS is slower than SPIFFS per your tests ?

lorol, 2020-06-17:
@luc-github  Writing to LITTLEFS  is 4 times faster. Reading is slower. (on my simple test)

luc-github, 2020-06-17:
Sorry I just noted the last number, the written ones
I will give a try - thank you

luc-github, 2020-06-17:
@lorol I have tested with latest ESP32 git - I used the sample and adding time test and size display
it works very well - thank you 
```
.WiFi connected
IP address: 
192.168.1.30
Contacting Time Server

Now is : 2020-06-17 20:01:42

Listing directory: /
  FILE: /hello.txt	SIZE: 6  LAST WRITE: 2020-06-17 20:00:53
  FILE: /hello2.txt	SIZE: 6  LAST WRITE: 2020-06-17 20:00:38
Writing file: /hello.txt
- file written
Appending to file: /hello.txt
- message appended
Reading file: /hello.txt
- read from file:
Hello World!
Renaming file /hello.txt to /foo.txt
- file renamed
Reading file: /foo.txt
- read from file:
Hello World!
Testing file I/O with /test.txt
- writing................................................................
 - 1048576 bytes written in 14731 ms
- reading................................................................
- 1048576 bytes read in 885 ms
Writing file: /hello.txt
- file written
Listing directory: /
  FILE: /foo.txt	SIZE: 14  LAST WRITE: 2020-06-17 20:01:42
  FILE: /hello.txt	SIZE: 6  LAST WRITE: 2020-06-17 20:01:58
  FILE: /hello2.txt	SIZE: 6  LAST WRITE: 2020-06-17 20:00:38
  FILE: /test.txt	SIZE: 1048576  LAST WRITE: 2020-06-17 20:01:42
Deleting file: /foo.txt
- file deleted
Deleting file: /test.txt
- file deleted
Total: 1507328, Used: 8192
Test complete

```

lorol, 2020-06-17:
Yes, you can compare the "feel" of running to SPIFFS.
LITTLEFS looks faster overall. At least I like to be this way ... after so much time spent to make it working  :)

luc-github, 2020-06-17:
FYI to compare speed with ESP32 FAT also
```
Testing file I/O with /test.txt
- writing................................................................
 - 1048576 bytes written in 13057 ms
- reading................................................................
- 1048576 bytes read in 276 ms
Free space:     409600
```

lorol, 2020-06-17:
@luc-github  
See if your tests give similar results like these: https://github.com/joltwallet/esp_littlefs#performance

luc-github, 2020-06-17:
the partition was not same size, I will redo tomorrow to compare properly

lorol, 2020-06-17:
I don't expect FAT to be beaten on speed, anyway.

sblantipodi, 2020-06-17:
once the job is finished and all works ok, 
why don't you open a pull request to ask espressif to insert this patch in the official SDK?

lorol, 2020-06-17:
@ because I never made a PR yet :)

lorol, 2020-06-17:
@sblantipodi  here you go: https://github.com/espressif/arduino-esp32/pull/4096

sblantipodi, 2020-06-20:
@lorol congrats man

closed 2020-09-02 by stale[bot]