espressif/arduino-esp32 #3771: HTTP Client setConnectTimeout not working with https

spyder0069, 2020-02-26:
I am trying to post to a https page and if the device cannot reach that page (server down scenario) it halts for 18 seconds.  The .setConnectTimeout  does not seem to make any difference.  However, if I change the url to a http (still blocked though) I immediately get a error code -5 and the code continues.  In my application holding for 18 seconds is too long.

For me the easy way to recreate this was to use my routers parental controls and just block the device from reaching the outside world.  

As long as its not block my device does work properly and post to my https url.  Any ideas on this?

spyder0069, 2020-02-27:
I am a newb but I am chasing through to see where the problem goes.  At this point I have tracked it down to the ssl_client.cpp file.  When it makes this call
`if (lwip_connect(ssl_client->socket, (struct sockaddr *)&serv_addr, sizeof(serv_addr)) == 0) {`
That is where it hangs for the 18 seconds when it can't connect.  The timeout variable is being set and is valid up to that point.  Something in this call is where it hangs and it doesn't look like it passes any sort of timeout to lwip_connect.

closed 2020-05-11 by stale[bot]

mtbspace, 2020-09-04:
Problem still exists for me.  If I set the firewall to block my client, the client will take 18 seconds to timeout, even when I have set to 5 seconds with ```setConnectTimeout(5000)```

mtbspace, 2020-09-04:
Problem still exists for me. If I set the firewall to block my client, the client will take 18 seconds to timeout, even when I have set to 5 seconds with setConnectTimeout(5000)

kimble4, 2021-04-12:
I'm also seeing this.