espressif/arduino-esp32 #3772: Rebuild library.... and now?

GianCann, 2020-02-26:
What you are trying to do? Rebuild the ESP32-arduino libraries with different sdkconfig parameters to elevate the maximum number of BLE connection (from 3 to 9)
Describe your system( Hardware, computer, O/S, core version, environment). Ubuntu / Windows 10

I managed to recompile the libraries, following these instructions:
https://github.com/espressif/arduino-esp32/blob/master/docs/lib_builder.md

I rebuild the library on Ubuntu.

Now, I have in the folder /esp32-lib-builder/out the new libraries and the bootloaders.

I using Arduino, on Windows 10, with the default (1.0.4) ESP32 core library.
How should I proceed to use the libraries that I have compiled?

lbernstone, 2020-02-26:
The script automates the process of building arduino as an ESP-IDF component. Once it is complete, you can cherry pick the needed libraries from out/tools/sdk/lib, or run `tools/copy-to-arduino.sh` to copy the entire built system.

GianCann, 2020-02-26:
Thank you @lbernstone  for reply.
I can't use the script, because I compiled the libraries on Ubuntu, and I want to use them in the Arduino environment, on Windows.
When you say "you can choose the necessary libraries" you mean that I can take the * .a files and replace them with those in the folder packages in "C: \ Users \ (username) \ AppData \ Local \ Arduino15 \ packages \ esp32 .... "?

sorry for my "newbie", but i need only to have more of 3 BLE server device connected to one ESP32, and seem so complicated to do...

lbernstone, 2020-02-27:
> When you say "you can choose the necessary libraries" you mean that I can take the * .a files and replace them with those in the folder packages in "C: \ Users \ (username) \ AppData \ Local \ Arduino15 \ packages \ esp32 .... "?
Yes. I prefer to do this with a git repo, but you can replace the files from board manager.

closed 2020-05-11 by stale[bot]

brandonros, 2021-04-10:
I changed one line in `/Users/brandonros/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/sdkconfig`

How can I force it to "recompile" with the extra library included?

Trying to solve

```
sketch/bignumber.ino.cpp.o:(.literal._Z5setupv+0x38): undefined reference to `mbedtls_ripemd160_init'
sketch/bignumber.ino.cpp.o:(.literal._Z5setupv+0x3c): undefined reference to `mbedtls_ripemd160_starts_ret'
sketch/bignumber.ino.cpp.o:(.literal._Z5setupv+0x40): undefined reference to `mbedtls_ripemd160_update_ret'
sketch/bignumber.ino.cpp.o:(.literal._Z5setupv+0x44): undefined reference to `mbedtls_ripemd160_finish_ret'
sketch/bignumber.ino.cpp.o: In function `setup()':
/Users/brandonros/Documents/Arduino/bignumber/bignumber.ino:24: undefined reference to `mbedtls_ripemd160_init'
/Users/brandonros/Documents/Arduino/bignumber/bignumber.ino:27: undefined reference to `mbedtls_ripemd160_starts_ret'
/Users/brandonros/Documents/Arduino/bignumber/bignumber.ino:29: undefined reference to `mbedtls_ripemd160_update_ret'
/Users/brandonros/Documents/Arduino/bignumber/bignumber.ino:30: undefined reference to `mbedtls_ripemd160_finish_ret'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board ESP32 Dev Module.
```