espressif/arduino-esp32 #3773: ESP32 EVB won't boot when powered by 5V. Boot loop ?

GeorgeFlorian, 2020-02-27:
Board: ESP32-EVB-EA Rev.H
Platform: PlatformIO Core 4.2.1

[image]

The circuit above is composed of a 220V to 12V PSU and a 12V to 5V switching power supply The measured output is ~5V.

I've connected the ESP to a 5V power source. If I boot up the ESP without connecting it to a micro-USB it won't boot up unless I hit RESET.

If I'm also connecting a micro-USB to the ESP it will log some gibberish and then it will work normally.

```
��␛��␃�'�$$'��␇�␛'''␄�$$$�␄$ ␃#�$����␄␄#��$$$$�$�␄$d ␃#�$���␛␜␀␄#��$$$��$�␄l$ ␃����'�␛'��␟��'��␛'8|�␃␃#�����␀����<�;␛�''8␂;;'#$�␄$���|�#␄␄␄$#���␜␛$␛$␛␛�␛� ␃���'�␛'��␟��'��␛'8<�␃␃#�����␀�����<�;␒�''8␃;;'#$�␄$���<�#␄�␄$#�␛���;␛���Ó ␃␃����'�␂$␄$#��'''�␎�␛''�␄�'���␜␛'##␇$␃Ã��'�␃�␄$#��'''�␇�␛''�␄�'���␜␛'##␇$����'�␛'��␟��'��␛'8<�␃␂#�����␀�����<�;␛�''8␃;;n#$�␄$���<�#␌$�$#�␛����␓$$␇�$$$�␇��ؐ␇$����'�␛'��␟��'��␛'8<�␃␃#�$�␜�␃�����<�;␛�''8␃;;o#$�␄������'g␟␀�␄$␄$#�$$�␃�␛�$�<�� ␃␃����'�␃�␄$#��'''�␇�␛''�␄�'���␜␛'##␇$␃����'�␃␄�$#��'''�␇�␛o'�␄�'���␜␛'##␇$����'�␛'��␟��'��␛'8<�␃␃#�����␀�����<�;␛�''8␃;;'#$�␄$���<�b␄$�$#�␛����␛$$␇�$$$�␇�ې�␇$����'�␛'��␟��'��␛'8<�␃␃#�$�␜�␃��ܞ�<�;␛�''8␃;;'#$�␄������g'␟␀�␄$␄$#�$$�␃�␛�$�<��`␃␃����'�␃$�$#��'''�␇�␛''�␄�'���␜␛'##␇$␃����'�␃��$#��'''�␇�␛''�␄�'��␜␛'##␇$����'�␛'��␟��'��␛'8<�␃␃#�����␀�ۜ��<�;␛�''8␃{;'#$�␄$���<�#␄$�$#�␛����␛$$␇�$$$�␇Û��␇$����'�␛'��␟��'��␛'8<�␃␃#�$�␜�␃�ۜ��<�;␛�''8␃{;'#$�␄������''␟␀�␄$␄$#�d$�␃�␛�$�<Ó ␃␃����'�␃��$#��'''�␇�␛o'�␄�'���␜␛g##␇$␃����g�␃␄␄$#��'''�␇�␛''�␄�'���␜␛'##␇$����'�␛'��␟��'��␛'8<�␃␃#�����␀�����<�;␛�''8␃;;'#$�␄$���<�#␄$�$#�␛����␛$d␇�$$$�␇����␇$����'�␛'��␟��'��␛'8<�␃␃#�$�␜�␃�����<�;␛�''8␃;;'#$�␄������''␟␀�␄$␄$#�$$�␃�␛�$�<�� ␃␃����'�␃$␄$#��'''�␇�␒''�␄�'���␜␛'##␇$␃��ے'�␃$␄␄$#��'''�␇�␛''�␄�'���␜␛'##␇$���'�␛'��␟��'��␛'8<�␃␃#�����␀�����<�;␓�''8␃;;'#$�␄$���<�#␄$�$c�␛����␛$$␇�d$$�␇����␇$����'�␛'��␟��'��␛'8<�␃␃#�$�␜�␃�����<�;␛�''8␃;;'#$�␄������''␟␀�␄$␄$#�$$�␃�␛�$�<�� ␃␃����'�␃$$␄$#��'''�␇�␛''�␄�g���␜␛'##␇$␃����'�␃$�␄$#��'''�␇�␛''�␄�'���␜␛o##␇$����'�␛o��␟��'��␛'8<�␃␃#�����␀�����<�;␛�''8␃;;'#$�␄$���<�#␄$�$#�␛����␛$$␇�$$$�␇����␇$����'�␛o��␟��'��␛'8<�␃␃#�$�␜�␃�����<�;␛�''8␃;;'#$�␄������''␟␀�␄$␄$#�$$�␃�␛�$�<�� ␃␃����'�␃$�␄$#��''n�␇�␛''�␄�'���␜␛'##␇$␃����'�␃$␄�$#��'''�␇�␛''�␄�'���␜␛g##␇$����'�␛'��␟��'��␛'8<�␃␃c�����␀�����|�;␛�''8␃;;'b$�␄$���<�#␄l�$#�␛����␛$$␇�$$$�␇����␏$����'�␛'��␟��'��␛'8<�␃␃c�$�␜�␃�����|�;␛�''8␃;;'c$�␄������''␟␀�␄$␄$#�$$�␃�␛�$�<�� ␃␃����'�␃$$�$#��'''�␇�␛''�␄�'���␜␛'##␇$␃����'�␃$��$#��'''�␇�␛''�␄�'���␜␛'##␇$�Ó�'�␛'��␟��'�␛'8<�␃␃#����␀�����<�;␛�''8␃;;'#$�␄d���<�#␄$�$#�␛����␛$$␇�$l$�␇����␇$����'�␛'��␟��'�␛'8<�␃␃#�$�␜�␃�����<�;␛�''8␃;;'#$�␄������''␟␀�␄$␄$#�$$�␃�␛�$�<�� ␃␃����'�␃$��$#��'o'�␇�␛''�␄�g���␜␛'##␇$␃����'�␃$␄␄$#��'''�␇�␛''�␄�'���␜␛o##␇$����'�␛'��␟��'��␛'8<�␃␃#�����␀�����<�r␛�''8␃;;'#$�␄$���<�#␄$�$#�␛����␛$$␇�$$$�␇����␇$����'�␛'��␟��'��␛'8<�␃␃#�$�␜�␃�����<�r␛�''8␃;;'#$�␄������''␟␀�␄$␄$#�$$�␃�␛�$�<�� ␃␃����'�␃$$␄$#��''n�␇�␛''�␄�'���␜␛'##␇$␃����'�␃�␄␄$#��'''�␇�␛''�␄�'���␜␛'##␎$ $$#��'�8�␃�''�$;;�;r␃������'��;␃�$����'<'␃$␄␄�<�␛''�$�␛␛''###␇$␇�''�<?�''��␃���p�'��;�␛#�␃$''$���;�'g�<␀$$ ␃���'�␛'��␟��'��␛'8<�␃␃#�����␀�����<�;␛�''8␃;;'#$�␄$���<�#␄$�$#�␛����␛$$␇�l$$�␇����␇$����'�␛'��␟��'��␛'8<�␃␃#�$�␜�␃�����<�;␛�''8␃;;'#$�␄������''␟␀�␄$␄$#�$$�␃�␒�$�<�� ␃����'�␛'��␟��'��␛'8<�␃␃#�����␀�����<�;␛�''8␃;;'#$�␄$���<�#␄$�$#�␛����␛$$␇�$$$�␇����␇$����'�␛'��␟��'��␛'8<�␃␃#�$�␜�␃�����<�;␛�''8␃;;'#$�␄������''␟␀�␄$␄$#�$$�␃�␛�$�<�� ␃����'�␛'��␟��'��␛'8<�␃␃#�����␀�����<�;␛�''8␃;;'#$�␄l���<�#␄$�$#�␛����␛$$␇�$$$�␇����␇$����'�␛'��␟��'��␛'8<�␃␃#�$�␜�␃�����<�;␛�''8␃;;'#$�␄������''␟␀�␄d␄$#�$$�␃�␛�$�<�� ␃
rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
State of Induction PINS: 
INPUT_1 (GPIO 15): 1
INPUT_2 (GPIO 17): 1
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
```

If I power up the ESP using ONLY the micro-USB it still outputs the gibberish in the beginning but then it works as it should be.

Does it have something to do with the ESP32 running on 5V ? It looks like it.

My other ESP EVB work, both Rev F and Rev H. Only this particular Rev H doesn't boot.  
Did I do something to the board ? **Is it in some sort of boot loop or something ?**

beegee-tokyo, 2020-02-27:
Just to clarify, you are hopefully **NOT** powering the ESP32 directly with 5V? If you do, you are very lucky that it is still alive!!!! The ESP32 is a 3.3V device.

I would check the 5V from the PSU, specially during power up. And as well the 3.3V on the ESP32. Maybe you need additional capacitors on the power lines.

How is the EN pin of the ESP32 connected? It might go HIGH too fast, before the voltage is stable. Maybe a capacitor there would be helpful as well to delay the enabling of the ESP32.

GeorgeFlorian, 2020-02-27:
If I shouldn't power it up by 5V why the hell does the ESP EVB has a 5V
input ?

beegee-tokyo, 2020-02-27:
The dev boards and breakout boards have 3.3V regulators. On your picture it is not clear if you have an ESP32 on your own PCB or if you have a breakout or dev board on it.

GeorgeFlorian, 2020-02-27:
> The dev boards and breakout boards have 3.3V regulators. On your picture it is not clear if you have an bare ESP32 on your own PCB or if you have a breakout or dev board on it.

[image]

What is unclear in this photo ? I have mentioned that I am using an ESP32-EVB-EA Rev.H and in the right side of the picture is an ESP32-EVB-EA powerd via the 5V input. 

Anyway, this devboard most definitely has 3V3 regulators. But the issue is with the boot sequence.

MarkyAD, 2020-02-27:
@GeorgeFlorian Nobody wants to help a prick. Remember, this is an issue tracker meant to track problems with the core. It's not a help forum. If you can't get your board to work, look in the mirror first.

sansillusion, 2020-02-27:
I am tempted to think that the stepdown might be creating too much ripples in the voltage. Can you try another 5V supply ? Mabe a phone charger with at least 1A of current.

sansillusion, 2020-02-27:
Can't see your picture in high res because of spam website you used to host it.

MarkyAD, 2020-02-27:
I wonder why he chose 2 gpio's with reserved meanings? gpio15 is normally used to silence boot messages when pulled low. gpio 17 has special meaning to both flash ram and psram on certain boards. Great choices!

atanisoft, 2020-02-27:
> gpio15 is normally used to silence boot messages when pulled low. gpio 17 has special meaning to both flash ram and psram on certain boards.

GPIO 15 is used on this board for an SD slot, it will cause problems if the SD slot is used likely.
GPIO 17 is exposed on UEXT1 on this board, it appears safe to use.

This board has a lot of extras on board so the GPIO pins that are free to use are very limited overall if you intend on using the on-board functionality.

GeorgeFlorian, 2020-02-27:
> @GeorgeFlorian Nobody wants to help a prick. Remember, this is an issue tracker meant to track problems with the core. It's not a help forum. If you can't get your board to work, look in the mirror first.

You're saying prick, but I'm saying that I am confused about what is wrong about my explanations and image.
Also, I am posting here because on this repository I have found a lot of help in the past.

> Just to clarify, you are hopefully **NOT** powering the ESP32 directly with 5V? If you do, you are very lucky that it is still alive!!!! The ESP32 is a 3.3V device.
> 
> I would check the 5V from the PSU, specially during power up. And as well the 3.3V on the ESP32. Maybe you need additional capacitors on the power lines.
> 
> How is the EN pin of the ESP32 connected? It might go HIGH too fast, before the voltage is stable. Maybe a capacitor there would be helpful as well to delay the enabling of the ESP32.
> I wonder why he chose 2 gpio's with reserved meanings? gpio15 is normally used to silence boot messages when pulled low. gpio 17 has special meaning to both flash ram and psram on certain boards. Great choices!

So here is where the confusion comes from. I sincerely thought that specifying the ESP board would suffice and help avoid these confusions.

The board is an OLIMEX ESP32-EVB-EA. This is [its schematics](https://github.com/OLIMEX/ESP32-EVB/blob/master/HARDWARE/REV-H/ESP32-EVB_Rev_H.pdf).  
I have consulted with an OLIMEX moderator on which pins are safe to use and for this particular project I use GPIO 13,15,16 and 17. But this is beside the point because I have disconnected everything from the pins.

> I am tempted to think that the stepdown might be creating too much ripples in the voltage. Can you try another 5V supply ? Mabe a phone charger with at least 1A of current.

That same Power Supply works for all of my other ESP32-EVB. As I've mentioned before, this is the only board that doesn't boot from 5V. 

>If you can't get your board to work, look in the mirror first.

If this is my mistake I would like to identify it and fix it before I damage other boards.

MarkyAD, 2020-02-27:
True that people here are very helpful. I just think you'll get more support for "why doesn't my project work" type issues from somewhere like http://esp32.com. The gibberish you're seeing at boot is because it uses a different (and odd) baud rate before boot.

GeorgeFlorian, 2020-02-28:
> True that people here are very helpful. I just think you'll get more support for "why doesn't my project work" type issues from somewhere like http://esp32.com. The gibberish you're seeing at boot is because it uses a different (and odd) baud rate before boot.

My issue is not about "why doesn't my project works" but "why doesn't this ESP board boot". My project works just fine with all the other boards. Only this one has a problem and I would like to find out what problem and how to avoid it.

MarkyAD, 2020-02-28:
And I still fail to see how you relate this to a problem with the core, which is what this issue tracker is for. If it did it on all boards, then yes there might be a core problem, but it doesnt. That sort of points toward the board itself.

closed 2020-02-28 by GeorgeFlorian