espressif/arduino-esp32 #3775: Task watchdog getting triggered when opening OTA partition

dflogeras, 2020-02-27:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			1.0.4
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                          no
Upload Speed:						921600
Computer OS:                                            Linux

### Description:
We are noticing an issue on _some_ modules (so far 2/4) whereby calling esp_ota_begin() can stall the ipc0 task for too long, and cause a watchdog reset.

In case it matters, we are compiling with CPU freq 80MHz, and using the minimal SPIFFS w OTA partition scheme.

Initially, we thought that the arduino LoopTask was causing the watchdog timing out, so we moved the calls to esp_ota_get_next_update_partition() and esp_ota_begin() into their own thread to let the LoopTask run freely.  This does not solve it, as the task that watchdogs is actually ipc0 as shown:

E (13210) task_wdt: Task watchdog got triggered. The following tasks did not re:
E (13241) task_wdt:  - IDLE0 (CPU 0)                                            
E (13374) task_wdt: Tasks currently running:                                    
E (13374) task_wdt: CPU 0: ipc0                                                 
E (13374) task_wdt: CPU 1: IDLE1                                                
E (13374) task_wdt: Aborting.                                                   
abort() was called at PC 0x400dbbb7 on core 0

Putting a JTAG on and inspecting the ipc0 task, its callstack is as follows:

[Switching to thread 9 (Thread 1073446484)]
#0  0x40083df2 in spi_flash_op_block_func (arg=0x0) at /home/runner/work/esp32-a
rduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/cache_
utils.c:82
82      in /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder
/esp-idf/components/spi_flash/cache_utils.c
(gdb) where
#0  0x40083df2 in spi_flash_op_block_func (arg=0x0) at /home/runner/work/esp32-a
rduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/cache_
utils.c:82
#1  0x40082c6a in ipc_task (arg=0x0) at /home/runner/work/esp32-arduino-lib-buil
der/esp32-arduino-lib-builder/esp-idf/components/esp32/ipc.c:62
#2  0x40091270 in vPortTaskWrapper (pxCode=0x40082c08 <ipc_task>, pvParameters=0
x0) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp
-idf/components/freertos/port.c:143

Working boards seem to still take over 4 seconds to return from esp_ota_begin().  To test the theory that we have some boards that may have slower flash than the others, we built our own SDK and changed the task watchdog timer from 5s to 10s, and that works around the issue.  The misbehaving boards actually take about 5.2 seconds to call esp_ota_begin().  The problem also seems to be affected negatively when BLE is initialized (as in the minimal example below).

I'd really prefer not to have to build our own SDK from a bunch of master branches.  Is there any way to get the arduino released SDK to raise the watchdog timeout limit?  If not, is there anyway to check out _EXACTLY_ the sources used to re-create a numbered release?  I only know about the tools in https://github.com/espressif/esp32-arduino-lib-builder which don't seem to be versioned.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

#include <esp_ota_ops.h>
#include <BLEServer.h>
#include <BLEDevice.h>
#include <BLE2902.h>

BLEServer* ble_server( nullptr );
BLEService* ble_service( nullptr );
BLECharacteristic* tx_characteristic( nullptr );
BLECharacteristic* rx_characteristic( nullptr );
BLECharacteristic* proto_characteristic( nullptr );

static const std::string service_uuid( "6e400001-b5a3-f393-e0a9-e50e24dcca9e" );
static const std::string characteristic_rx_uuid( "6e400002-b5a3-f393-e0a9-e50e24dcca9e" );
static const std::string characteristic_tx_uuid( "6e400003-b5a3-f393-e0a9-e50e24dcca9e" );

static const std::string characteristic_protocol_uuid( "6e400004-b5a3-f393-e0a9-e50e24dcca9e" );

static const esp_partition_t* partition;
static esp_ota_handle_t handle;
SemaphoreHandle_t open_sem;

[... 206 lines cut ...]
[V][BLECharacteristic.cpp:198] handleGATTServerEvent(): >> handleGATTServerEvenn
[V][BLECharacteristic.cpp:646] setValue(): >> setValue: length=3, data=322e30, e
[V][BLECharacteristic.cpp:452] handleGATTServerEvent(): << handleGATTServerEvent
[V][FreeRTOS.cpp:189] take(): Semaphore taking: name: SetValue (0x3ffe0524), ow>
[V][BLECharacteristic.cpp:198] handleGATTServerEvent(): >> handleGATTServerEvenn
[V][FreeRTOS.cpp:198] take(): Semaphore taken:  name: SetValue (0x3ffe0524), ow>
[V][BLECharacteristic.cpp:452] handleGATTServerEvent(): << handleGATTServerEvent
[V][FreeRTOS.cpp:143] give(): Semaphore giving: name: SetValue (0x3ffe0524), ow>
[V][BLEServer.cpp:271] handleGATTServerEvent(): << handleGATTServerEvent        
[V][BLECharacteristic.cpp:655] setValue(): << setValue                          
[V][BLECharacteristic.cpp:477] notify(): >> notify: length: 3                   
[D][BLECharacteristic.cpp:782] onNotify(): BLECharacteristicCallbacks           
[D][BLECharacteristic.cpp:783] onNotify(): BLECharacteristicCallbacks           
[V][GeneralUtils.cpp:296] hexDump():      00 01 02 03 04 05 06 07 08 09 0a 0b 0f
[V][GeneralUtils.cpp:297] hexDump():      -- -- -- -- -- -- -- -- -- -- -- -- --
[V][GeneralUtils.cpp:323] hexDump(): 0000 32 2e 30                             0
[V][BLECharacteristic.cpp:487] notify(): << notify: No connected clients.       
[D][BLECharacteristic.cpp:794] onStatus(): BLECharacteristicCallbacks           
[D][BLECharacteristic.cpp:795] onStatus(): BLECharacteristicCallbacks           
[I][BLEDevice.cpp:569] getAdvertising(): create advertising                     
[D][BLEDevice.cpp:571] getAdvertising(): get advertising                        
[V][BLEAdvertising.cpp:181] start(): >> start: customAdvData: 0, customScanResp0
[D][BLEAdvertising.cpp:200] start(): - no services advertised                   
[V][BLEAdvertising.cpp:241] start(): << start
```

lbernstone, 2020-02-28:
Since you know exactly the method that causes the issue, why not just disable the watchdog (disableCore0WDT) while OTA begin runs?  You really should be suspending all normal activity during an OTA in any case, so turning off WDT for 4-5 seconds should be reasonably safe.

dflogeras, 2020-03-01:
I'm not a huge fan of disabling watchdogs ever; same reason I don't take my seatbelt off when I get on the road my house is on because nothing usually happens there.

We are using BLE to send the update data to the device, so there is still some async activity.  I have had the BLE stack give me guru meditations during development of this firmware.

lbernstone, 2020-03-01:
Firmware update is not driving the car, it is replacing the oil filter.  The engine should be off when you do that :smile:.  You should only need to WDT off when you apply the update (the end method).  That is where it will be moving a big chunk of data to the disk, and waiting for an ack.

dflogeras, 2020-03-01:
Well, semantics of WDT aside, is there a documented way to rebuild _exactly_ an arduino libs release, ie. 1.0.4?  The arduino-lib-builder script seems to be unversioned, and also pulls in master branches of its dependencies.

lbernstone, 2020-03-01:
If you look at the releases page, there is a tag and a commit point for each release.  Checkout that commit.

dflogeras, 2020-03-01:
I understand, but I mean in this repository

https://github.com/espressif/esp32-arduino-lib-builder/

There does not seem to be releases.  And the scripts seem to checkout unversioned subdeps.

lbernstone, 2020-03-02:
If you look in tools/config.sh there is a line for the branch (read https://github.com/espressif/arduino-esp32/blob/master/docs/lib_builder.md).  You can also use a specific commit by feeding it with an IDF_COMMIT variable:
`IDF_COMMIT=4638628873a061c36faffebe4d146d13f960076d build.sh`

closed 2020-05-15 by stale[bot]

plektra, 2021-03-04:
Was there any solutions or easier workarounds to this than to rebuild the whole SDK? I'm experiencing the same WDT issue with OTA library when calling `esp_https_ota_perform(esp_https_ota_handle_t https_ota_handle)`. The board I'm using is ESP32 DevKitC V4.

lbernstone, 2021-03-04:
```
#include <esp_task_wdt.h>

      esp_task_wdt_init(15,0);      
      if (Update.end(true)) { //true to set the size to the current progress
        log_v("Update Success: %u\nRebooting...", upload.totalSize);        
        esp_task_wdt_init(5,0);
      }

plektra, 2021-03-05:
> ```
> #include <esp_task_wdt.h>
> 
>       esp_task_wdt_init(15,0);      
>       if (Update.end(true)) { //true to set the size to the current progress
>         log_v("Update Success: %u\nRebooting...", upload.totalSize);        
>         esp_task_wdt_init(5,0);
>       }
> ```

Oh it was so obvious. :) Thanks, this works out well.

txf-, 2023-04-05:
Just as a note to anybody else that may come across this issue. This problem does not occur in all devices. For some there is some unknown difference between some WROOM32 devices, that needs a longer watchdog timeout, and plenty of other devices that manage just fine with the defaults.