espressif/arduino-esp32 #3779: Add setPins to Wire instead of overloading begin() in a breaking manner

ladyada, 2020-03-02:
hi folks - we get about 1 request every week or so for us to change our libraries so people can pass I2C pin numbers into Wire.begin()

We can't merge these PRs because official Arduino code style has `Wire.begin()` in sensor/device `begin()`:
https://github.com/arduino-libraries/Arduino_LSM6DS3/blob/master/src/LSM6DS3.cpp#L76

please add a `Wire.setPins()` or similar function that can be called at the beginning of the sketch, instead of breaking compatibility by changing the prototype of Wire.begin() - it will make both library developers and users much happier!

see: 
- https://github.com/adafruit/Adafruit_BME280_Library#62
- https://github.com/adafruit/Adafruit_MCP9808_Library/issues/23
- https://github.com/adafruit/Adafruit_BMP280_Library#38
- (and more..)

stickbreaker, 2020-03-03:
@ladyada A simple answer for this problem, call `Wire.begin(sdapin,sclpin,busspeed)` in `setup()` and your libraries use the standard `Wie.begin()` in the library.  (I assume your libraries execute a `Wire.begin()` as part of the library initialization code, multiple invocations of `Wire.begin()` are supported.) The `Wire()` code is designed to reuse the previously assigned pins and bus speed. Since the pin allocation is necessary for ESP32, it should be set. And `Setup()` is the logical place.

Chuck.

ladyada, 2020-03-03:
@stickbreaker hi, thanks for the response. that is the exact thing we will not do because official Arduino code style has Wire.begin() in sensor/device begin() :)

https://www.arduino.cc/en/Reference/WireBegin

ESP changed the Wire prototype so that it is different from every other Arduino platform. We are asking that this not be done and an alternative way to set pins is defined

stickbreaker, 2020-03-04:
@ladyada So, you want code like this?
```c++
void setup(){
#ifdef ARDUINO_ARCH_ESP32
   Wire.setpins(sda,scl);
#endif
   Wire.begin();
}
```

Chuck.

ladyada, 2020-03-04:
yeah that would best! also, capitalize P in setPins() to match arduino style :)

stickbreaker, 2020-03-04:
If you can use the `#ifdef` this example is already supported.
```c++
void setup(){
#ifdef ARDUINO_ARCH_ESP32
   Wire.begin(sda,scl);
#endif
   Wire.begin(); //redundant, but won't cause any problems.
}
```

Chuck.

ladyada, 2020-03-04:
hi, not if the code is inside existing written libraries. they need to be able to call setPins before the sensor begin

```
APDS9960 sensor = APDS9960();

void setup() {
   Wire.setPins(4, 5);
   sensor.begin(); // calls Wire.begin
}
```

Andrewiski, 2020-03-04:
@ladyada  Is this why the OLED feather displays garbage on ESP32 but same code works on a Wiced Feather?

ladyada, 2020-03-04:
@Andrewiski probably not - if you need tech support please post in the adafruit forums :)

ladyada, 2020-03-10:
@stickbreaker hi hi, checkin' in - do you have any other questions on this issue?  :)

ladyada, 2020-05-09:
@me-no-dev can y'all assign someone else who is interested in this issue? @stickbreaker has not replied in ~2 months.

ladyada, 2020-07-14:
@me-no-dev can y'all assign someone else who is interested in this issue? @stickbreaker has not replied in ~4 months

me-no-dev, 2020-07-16:
@ladyada yup. Let me see what I can do. I have no heard from Chuck if a few as well.
What I gather is that you just need a method added that would preset the pins used for `begin()`. Am I correct?

ladyada, 2020-07-16:
yes! right now we get weekly requests to have special code for ESP since they want to run `Wire.begin(4, 5)` or whatever, in the sensor `begin()` function. Instead, if we had a function that would `Wire.setPins()`, then when `Wire.begin()` is called later it would use the pins set by setPins earliler. If setPins is never called, `Wire.begin()` would use the board support package default

me-no-dev, 2020-07-17:
cool! Will do!

ladyada, 2020-07-17:
thankyouthankyouthankyou it will be incredibly helpful and let many more ESP customers use our libraries!

me-no-dev, 2020-07-18:
done :)

closed 2020-07-18 by me-no-dev

ladyada, 2020-07-18:
yaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaay this is awesome! thank you!

me-no-dev, 2020-07-18:
@ladyada please do email me if you see that there is no response. Things do get buried sometimes for various reasons :)

githufsb, 2020-10-18:
Wow!  I was just looking at how to use 9960 on a wemos esp32 board!  Thank you @ladyada and @me-no-dev!  It's so great to start a dev project and then see github comments from days ago that may solve your problem!

kamikaze2112, 2025-03-02:
For anyone here pulling their hair out like me wondering why this doesn't work with their ESP32-S3-devkit1 and the Adafruit SSD1306 library,  make sure you've got the right I2C address for your display.  The example code in the Adafruit SSD1306 library states that the 128x64 I2C address is 0x3D.  The particular 128x64 displays that I have are 0x3C by default.

[image]