espressif/arduino-esp32 #3780: Rebooting problem esp32

muznah9, 2020-03-02:
Hi all,

I tried to use esp32 with the THINGER platform it was working fine, but now it doesn’t work, and I have a rebooting problem.
Before this problem appears I was trying to change the partition table because I faced a storage space problem. ( I am working with Arduino IDE)

```
assertion "offset == dataSize" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/nvs_storage.cpp", line 438, function: esp_err_t nvs::Storage::readMultiPageBlob(uint8_t, const char*, void*, size_t)
abort() was called at PC 0x4011ca8f on core 0

Backtrace: 0x4008c284:0x3ffb52c0 0x4008c4b5:0x3ffb52e0 0x4011ca8f:0x3ffb5300 0x400fd929:0x3ffb5330 0x400fd994:0x3ffb53b0 0x400fca5a:0x3ffb5410 0x400fcf91:0x3ffb5450 0x4010ca2f:0x3ffb5470 0x4010cc07:0x3ffb54c0 0x4010ac4e:0x3ffb54e0 0x401078ae:0x3ffb5500 0x4008f3d3:0x3ffb5520 0x400889cd:0x3ffb5560

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8

-----------------------------------------------------------------------

Decoding stack results

0x4008c284: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 155
0x4008c4b5: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 170
0x4011ca8f: __assert_func at ../../../.././newlib/libc/stdlib/assert.c line 63
0x400fd929: nvs::Storage::readMultiPageBlob(unsigned char, char const*, void*, unsigned int) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/nvs_storage.cpp line 438
0x400fd994: nvs::Storage::readItem(unsigned char, nvs::ItemType, char const*, void*, unsigned int) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/nvs_storage.cpp line 455
0x400fca5a: nvs_get_str_or_blob(nvs_handle, nvs::ItemType, char const*, void*, size_t*) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/nvs_api.cpp line 515
0x400fcf91: nvs_get_blob(nvs_handle, char const*, void*, size_t*) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/nvs_api.cpp line 525
0x400889cd: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```

CODE

```
#include <ThingerESP32.h>

#define USERNAME "Pb"
#define DEVICE_ID "esp32"
#define DEVICE_CREDENTIAL "BPV@gDE1DCgI"

#define SSID "AndroidAPP"
#define SSID_PASSWORD "1a2b3cdef"

ThingerESP32 thing(USERNAME, DEVICE_ID, DEVICE_CREDENTIAL);

void setup() {

  pinMode(5, OUTPUT);

  thing.add_wifi(SSID, SSID_PASSWORD);

  // digital pin control example (i.e. turning on/off a light, a relay, configuring a parameter, etc)
  thing["led"] << digitalPin(5);

  // resource output example (i.e. reading a sensor value)
  thing["millis"] >> outputValue(millis());

}

void loop() {
  thing.handle();
}
```

lbernstone, 2020-03-02:
Your nvs seems damaged.  [Wipe the flash](https://github.com/espressif/esptool#erase-flash-erase_flash--erase-region) and try again.

muznah9, 2020-03-03:
Hi @lbernstone ,
Thank you for your replay.
I will try your solution.

muznah9, 2020-03-04:
Thank you @lbernstone. 
Your solution solved the problem.

closed 2020-03-04 by muznah9