espressif/arduino-esp32 #3781: Converting RX-TX pins for GPIO use

PrateekGoyal18, 2020-03-02:
Hello Everyone,
I have a system in which I have connected 2 push buttons to TX0 and RX0 of my ESP32. So I wanted to ask that can I make these TX0/RX0 to GPIO for reading switch input?

lbernstone, 2020-03-02:
If you don't initialize HardwareSerial(0) - aka Serial, and have logging set to None, then you should be able to use pins 1 and 3 for inputs.  If you want to use them for outputs, then you need a pull down on pin 15 to prevent the boot messages.

PrateekGoyal18, 2020-03-03:
Got it, thanks @lbernstone

closed 2020-03-03 by PrateekGoyal18

ahmetakman, 2021-08-11:
I have an HSPI connected device and pin 15 uses it as a CS pin. So if I want to use RX TX pins for GPIO is there any possibility that I pull pin 15 down always and continue to use it as a CS pin? (I mean pulling it down on software.)  Also, boot messages are not a problem for me ( I need to use them for led and buzzer) , If I do not pull down pin 15 would that be a problem? Because a friend of mine suggested that using those pins for GPIO might cause instability for the ESP32 chip.