espressif/arduino-esp32 #3784: esp32 using WiFiClientSecure and AWS api gateway client certificate, fail to connect to server

kannieWeesNie, 2020-03-03:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			?11/jul/2017?
IDE name:							Platform.io
Flash Frequency:					?40Mhz?
PSRAM enabled:                                          ?no?
Upload Speed:						115200
Computer OS:                                            Windows 10

### Description:
-using WiFiClientSecure to call AWS api gateway url (https://x.......g.execute-api.eu-west-1.amazonaws.com/v1/esp32)
-api gateway issued client certificate containing PEM-encoded public key
- fails connecting to server (see debug messages below)
- it works with the www.howsmyssl.com example

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>
#include <WiFiClientSecure.h>

const char *server = "x.......g.execute-api.eu-west-1.amazonaws.com";

void setup() {

....after connecting to wifi
client.setCertificate(api_gateway_test_root_ca); // my api gateway certificate containing PEM-encoded public key - i assume it works as in debug states Certificate verified

Serial.println("\nStarting connection to server...");
  if (!client.connect(server, 443))
    Serial.println("Connection failed!");
  else
  {
    Serial.println("Connected to server!");
	client.println("GET https://x......g.execute-api.eu-west-1.amazonaws.com/v1/esp32?param1=solly&param2=molly&param3=folly&param4=zolly");
    client.println("Host: x......g.execute-api.eu-west-1.amazonaws.com");

    client.println("Connection: close");
    client.println();

    while (client.connected())
    {
      String line = client.readStringUntil('\n');
      if (line == "\r")
      {
        Serial.println("headers received");
        break;
      }
    }
    // if there are incoming bytes available
    // from the server, read them and print them:
    while (client.available())
    {
      char c = client.read();
      Serial.write(c);
    }

    client.stop();
	
	}
}

void loop() {
}
```

### Debug Messages:
after connecting to wifi, it tries connecting to server...
Starting connection to server...
[V][ssl_client.cpp:56] start_ssl_client(): Free internal heap before TLS 265288
[V][ssl_client.cpp:58] start_ssl_client(): Starting socket
[V][ssl_client.cpp:93] start_ssl_client(): Seeding the random number generator
[V][ssl_client.cpp:102] start_ssl_client(): Setting up the SSL/TLS structure...
[I][ssl_client.cpp:156] start_ssl_client(): WARNING: Use certificates for a more secure communication!
[V][ssl_client.cpp:180] start_ssl_client(): Setting hostname for TLS session...
[V][ssl_client.cpp:195] start_ssl_client(): Performing the SSL/TLS handshake...
[V][ssl_client.cpp:216] start_ssl_client(): Verifying peer X.509 certificate...
[V][ssl_client.cpp:225] start_ssl_client(): Certificate verified.
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400ea714  PS      : 0x00060a30  A0      : 0x800e1edd  A1      : 0x3ffb1c70  
A2      : 0x3ffbaab0  A3      : 0x40084f34  A4      : 0x3ffb8214  A5      : 0x00000000
A6      : 0x00000000  A7      : 0xffffffbc  A8      : 0xfefefefe  A9      : 0x3ffb1c10  
A10     : 0x3ffb80c0  A11     : 0x3ffd7704  A12     : 0x00000000  A13     : 0x00000001  
A14     : 0x00060c20  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xfefeff26  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffc

Backtrace: 0x400ea714:0x3ffb1c70 0x400e1eda:0x3ffb1c90 0x400d30ab:0x3ffb1cb0 0x400d2811:0x3ffb1f20 0x400d295d:0x3ffb1f50 0x400d18b1:0x3ffb1f70 0x400d4413:0x3ffb1fb0 0x40088b9d:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac

Jeroen88, 2020-03-04:
According to the example you should call setCACert(), see [here](https://github.com/espressif/arduino-esp32/blob/13e02063d63146096952fc6a6ef0bc3ced5b1203/libraries/WiFiClientSecure/examples/WiFiClientSecure/WiFiClientSecure.ino#L75)

If you may decode the backtrace, you'll get more information about the failure.

kannieWeesNie, 2020-03-04:
> According to the example you should call setCACert(), see [...]

Thanks. but as far as i know, the api gateway is not a CA certificate. i have, although tried setCAcert() but this fails with certificate not verified. using setCertificate(), as per my code, verifies the certificate (see debug message) so as far as i can tell, the certificate is not the problem?

as this is all out of my depth, maybe those with more experience may find the api gateway doc helpful - see here [https://docs.aws.amazon.com/apigateway/latest/developerguide/getting-started-client-side-ssl-authentication.html](url)

Jeroen88, 2020-03-04:
The client certificate, the one that you use with your code, is to identify yourself (or at least the client) to the amazon server. But a https connection the client wants to identify the server. For this you should use setCACert(). 

I did not read ths docs, but maybe [this ](https://www.amazontrust.com/repository/AmazonRootCA1.pem)is the right root CA certificate.

Jeroen88, 2020-03-04:
I am Dutch, just curious, what does your nickname mean? :)

closed 2020-05-17 by stale[bot]