espressif/arduino-esp32 #3787: HTTPClient issues if used with both WiFi and Bluetooth enabled

palia95, 2020-03-05:
### Hardware:
Board:							ESP32 Dev Module
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:					115200
Computer OS:                                           Mac OSX

### Description:
When using HTTP Client with both BLE and WiFi Enabled, the HTTP Client doesn't work. In the sketch below, I got as HTTP Error Code -1, which clearly doesn't exits. That means that the HTTP Client doesn't work, it can't even connect to the server (of course the server is up). However, if I try to run the same code (check for OTA update HTTP Client) without using BLE (delete the BLE part, just a simple WiFI UDP client with an HTTP Client which check for OTA update), I had no issues to make the code run and the client connect to the HTTP server. It's a really weird behaviour, since I had no issues using only with WiFi Connection. Any explanation?

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include "BLEDevice.h"
#include "DHTesp.h"
//#include "esp_system.h"
#include <WiFi.h>
#include "WiFiUDP.h"
#include <HTTPClient.h>
#include <ESP32httpUpdate.h>
#include <Arduino.h>

const int FW_VERSION = 0;           // Check FW version for update

const char* ssid     = "";
const char* password = "";

//IP address to send UDP data to:
// either use the ip address of the server or 
// a network broadcast address
const char * udpAddress = "192.168.178.155";
const int udpPort = 20001;

// The remote service we wish to connect to.
static BLEUUID serviceUUID("c8e459b1-85b8-4816-99b4-3c1a34b53437");
// The characteristic of the remote service we are interested in.
static BLEUUID TempUUID("460240d0-e610-4fd5-8a95-a885a30f0660");
static BLEUUID HumUUID("460240d0-e610-4fd5-8a95-a885a30f0661");
static BLEUUID LUUID("460240d0-e610-4fd5-8a95-a885a30f0663");
static BLEUUID DelayUUID("460240d0-e610-4fd5-8a95-a885a30f0700");

//Define DHT Sensor
DHTesp dht;
int dhtPin = 18;
//int light = 39;
float nxtwkp = 60;

//Readings variable
//int L;
String temp;
String hum;
//String lig;
String DP;
String timedelay;

//The udp library class
WiFiUDP udp;

//Watch Dog Const
const int wdtTimeout = 20000;  //time in ms to trigger the watchdog
hw_timer_t *timer = NULL;

static boolean doConnect = false;
static boolean connected = false;
static boolean doScan = false;
static BLERemoteCharacteristic* pRemoteTempCharacteristic;
static BLERemoteCharacteristic* pRemoteHumCharacteristic;
static BLERemoteCharacteristic* pRemoteLCharacteristic;
static BLERemoteCharacteristic* pRemoteDelayCharacteristic;
[... 294 lines cut ...]
    // Set the characteristic's value to be the array of bytes that is actually a string.
    Serial.println("Writing new chars");
    pRemoteTempCharacteristic->writeValue(temp.c_str(), temp.length());
    pRemoteHumCharacteristic->writeValue(hum.c_str(), hum.length());
    //pRemoteLCharacteristic->writeValue(lig.c_str(), lig.length());
    std::string value = pRemoteDelayCharacteristic->readValue();
    //Serial.println(value.c_str());
    timedelay=value.c_str();
    Serial.println(timedelay);
    nxtwkp=timedelay.toFloat();
    
    //pClient->disconnect();

  }else if(doScan){
    BLEDevice::getScan()->start(0);  // this is just example to start scan after disconnect, most likely there is better way to do it in arduino
  }
  
  //Serial.println(ESP.getFreeHeap());
  Serial.println(nxtwkp);  
  ESP.deepSleep(nxtwkp*1000000); // Delay a second between loops.
} // End of loop
```

### Debug Messages from Serial Monitor:
"Firmware version check failed, got HTTP response code -1"

closed 2020-05-18 by stale[bot]

dmoskovtsov, 2020-08-17:
Experiencing a similar behaviour. Any updates on this?

palia95, 2020-08-18:
> Experiencing similar behaviour. Any updates on this?

Unfortunately not, the issue is still there, at least for me

gpkvt, 2021-05-28:
Same here. http.GET works fine, http.POST returns -1. Removing the Bluetooth Stack => http.POST works fine again.

DonatelloX, 2021-06-16:
The issue has not been solved, the only possibility is to use:
```
btStop();
[...your code HTTPClient...]
btStart();
```