espressif/arduino-esp32 #3791: ESP32 Ethernet to Ethernet Data Transfer Example

fsam6177, 2020-03-05:
### Hardware:
Board:							?ESP32 Olimex? 
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
PSRAM enabled:                                          ?no?
Upload Speed:						?115200?
Computer OS:                                            ?Windows 10? 

### Description:
How to do data transfer between ethernet without internet? I couldn't find any example. Could someone provide an example to send data like the chat server example in arduino Ethernet Library/from ethernet in esp32 to another ethernet device?

pipi61, 2020-03-06:
select wifi example
insert 
#include <ETH.h>
...
void WiFiEvent(WiFiEvent_t event) function 
(check \Arduino\hardware\espressif\esp32\libraries\WiFi\examples\ETH_LAN8720\ETH_LAN8720.ino)
....
void setup(void) 
{
....
  WiFi.onEvent(WiFiEvent);
#define ETH_TYPE        ETH_PHY_LAN8720
  #define ETH_CLK_MODE    ETH_CLOCK_GPIO0_IN  
 //ETH.begin(ETH_ADDR, ETH_POWER_PIN, ETH_MDC_PIN, ETH_MDIO_PIN, ETH_TYPE, ETH_CLK_MODE);
  ETH.begin(   1,        17,                23,        18,       ETH_TYPE, ETH_CLK_MODE); //20191111 esp32
.....
}

fsam6177, 2020-03-06:
I am really sorry..i don't get it.. How do we configure one of the ethernet as client and the other as server? Or basically how do you send data just like UART in case of ethernet?

pipi61, 2020-03-06:
check/find in your arduino libraries examples, and integrate the ethernet

fsam6177, 2020-03-06:
There is no example which does what i require. I don't want to connect to internet, so how do we get the IP address or do data transfer in a wired fashion. How do we send and read data using only ethernet ( no internet)?

pipi61, 2020-03-07:
Internet is not required to any examples. But IP address required. If use router on local network, then ipaddress request from DHCP. If not use router(and DHCP server) use fix ipaddress in examples...

pipi61, 2020-03-07:
Sorry
This is not a chat forum. This is a bug reporting and development issue.
If you have further questions, please visit the forum: https://www.esp32.com/viewforum.php?f=19

closed 2020-05-20 by stale[bot]