espressif/arduino-esp32 #3793: Reading or Writing to Flash Causes I2C Errors

CraftyMyner, 2020-03-06:
### Hardware:
PLATFORM: Espressif 32 1.11.2 > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
LIBRARY: Adafruit_MCP23017 @ 100khz

### Description:
I have an ESP32 DevKit with 4 MCP23017's on I2C and 3 MCP3008's on SPI (Not using fash SPI), I am using Adafruit's libraries for both of them. I have an issue where after reading or writing to flash via preferences the I2C bus gets stuffed up and doesn't seem to recover until a reset.
I have tried everything imaginable, faster and slower polling, adjusting I2C clock, delays between reading io and accessing flash. Mutex's to not allow both to happen at the same time. Repolling I2C device if received errors. I have tried 4.7k, 2.2k, and 1k pull up resistors.

I have the two jobs in separate freeRTOS tasks, this shouldn't matter though as i2c is only being used by one task and the flash only being used by the other task as well as the mutex preventing either from happening at the same time.

I have now removed everything from the project and focused on one MCP23017 @ address 0x20 and one preference of 20xuint16 (40bytes) to read each in their own task.

I still get I2C errors which cause the MCP23017 to return either all bits set or no bits set. I might be doing something completely wrong here but I am out of options for figuring this out.

### I2C Debug Messages (Keeps repeating same message):
```
[E][esp32-hal-i2c.c:318] i2cDumpI2c(): i2c=0x3ffbef28
[I][esp32-hal-i2c.c:319] i2cDumpI2c(): dev=0x60013000 date=0x16042000
[I][esp32-hal-i2c.c:321] i2cDumpI2c(): lock=0x3ffb853c
[I][esp32-hal-i2c.c:323] i2cDumpI2c(): num=0
[I][esp32-hal-i2c.c:324] i2cDumpI2c(): mode=1
[I][esp32-hal-i2c.c:325] i2cDumpI2c(): stage=3
[I][esp32-hal-i2c.c:326] i2cDumpI2c(): error=5
[I][esp32-hal-i2c.c:327] i2cDumpI2c(): event=0x3ffb85c0 bits=112
[I][esp32-hal-i2c.c:328] i2cDumpI2c(): intr_handle=0x3ffb85f0
[I][esp32-hal-i2c.c:329] i2cDumpI2c(): dq=0x3ffb859c
[I][esp32-hal-i2c.c:330] i2cDumpI2c(): queueCount=1
[I][esp32-hal-i2c.c:331] i2cDumpI2c(): queuePos=0
[I][esp32-hal-i2c.c:332] i2cDumpI2c(): errorByteCnt=-1
[I][esp32-hal-i2c.c:333] i2cDumpI2c(): errorQueue=0
[I][esp32-hal-i2c.c:334] i2cDumpI2c(): debugFlags=0x00000000
[I][esp32-hal-i2c.c:288] i2cDumpDqData(): [0] 7bit 20 W STOP buf@=0x3ffc1f0e, len=3, pos=3, ctrl=11111
[I][esp32-hal-i2c.c:306] i2cDumpDqData(): 0x0000: ...                              12 00 00 
[I][esp32-hal-i2c.c:346] i2cDumpInts(): 0 row   count   INTR    TX      RX      Tick 
[I][esp32-hal-i2c.c:350] i2cDumpInts(): [01]    0x0001  0x0202  0x0004  0x0000  0x00033764
[I][esp32-hal-i2c.c:350] i2cDumpInts(): [02]    0x0001  0x0020  0x0000  0x0000  0x00033764
[I][esp32-hal-i2c.c:1130] i2cProcQueue(): Bus busy, reinit
[W][esp32-hal-i2c.c:1411] i2cCheckLineState(): invalid state sda(21)=0, scl(22)=1
[D][esp32-hal-i2c.c:1419] i2cCheckLineState(): Recovered after 1 Cycles
```

stickbreaker, 2020-03-06:
@CraftyMyner :
Are you using bit 7 of either port as input? (P7A or P7B) [MicroChip Forum Error)](https://www.microchip.com/forums/m831656.aspx) 

Please post your test code I Use MCP23017, MCP23008 without any issues.

Chuck.

CraftyMyner, 2020-03-06:
I do use bit 7 for input, was not aware of this issue. However, I have now gotten it to work with direct I2C calls rather than using wire or a library. (i2c_master_start() and i2c_master_write_byte() ect.)

###Code that causes errors
```
#include <Arduino.h>

#include "Adafruit_MCP23017.h"
#include "Preferences.h"

#define kSAMPLES 20
#define kSAMPLE_KEY "samples"

TaskHandle_t readGPIOTask = NULL;
void readGPIO(void * parameter);

TaskHandle_t readPrefTask = NULL;
void readPref(void * parameter);

uint16_t sampleData[kSAMPLES];

MCP23017 io;
Preferences prefs;

void setup() {
  Serial.begin(115200);

  io.begin(0);
  prefs.begin("test");
  prefs.putBytes(kSAMPLE_KEY,sampleData,sizeof(sampleData));

  xTaskCreate(readGPIO, "ReadGPIO", 4096, NULL, 1, &readGPIOTask);
  delay(5000);
  xTaskCreate(readPref, "ReadPref", 4096, NULL, 1, &readPrefTask);
}

void loop() {}

void readGPIO(void * parameter) {
    while (true) {
      uint16_t gpio = io.readGPIOAB();
      if (gpio > 0) log_e("IO READ ERROR!");
      vTaskDelay(1);
    }
}

void readPref(void * parameter) {
    while (true) {
      uint16_t data[kSAMPLES];
      prefs.getBytes(kSAMPLE_KEY,data,sizeof(data));
      log_i("Read Pref");
      vTaskDelete(readPrefTask);
      vTaskDelay(1);
    }
}
```

stickbreaker, 2020-03-06:
I'll check out adafruits library and see if I can see any conflicts.

Chuck.

closed 2020-05-20 by stale[bot]