espressif/arduino-esp32 #3794: Proper way of restarting WiFi after WiFi.mode(WIFI_OFF);

thezenox, 2020-03-06:
I am running an Upload Task connecting to a WiFiAP (doing a scan and select the best Network) and uploading stuff. If finished uploading, wifi is turned off and the task is deleted.
If I want to upload again, the task ist created again, but I get:
`E (345591) wifi: esp_wifi_scan_start 1245 wifi not start`

So what is the proper way of turning WiFi off completly (reduce power consumption) and enabeling it again? I already tried different things but can't get it working.

Currently I am doing this:
Start WiFi (only relevant code):
```
WiFi.mode(WIFI_MODE_STA); // calls esp_wifi_set_mode(WIFI_MODE_STA); and esp_wifi_start();
WiFi.enableSTA(true);
scanResult = WiFi.scanNetworks(); // doing scan
if(scanResult > 0){
 WiFi.begin(ssid, pw); // connecting the best network I have the key for
}
```
End WiFi:
```
WiFi.disconnect(true); // calls esp_wifi_disconnect(); and WiFi.enableSTA(false);
WiFi.mode(WIFI_OFF); // calls esp_wifi_stop(); and wifiLowLevelDeinit(); in espWiFiStop();, while wifiLowLevelDeinit is doing nothing, yet https://github.com/espressif/arduino-esp32/blob/2f13a960acd1baf1599d855385c64a2716708f85/libraries/WiFi/src/WiFiGeneric.cpp#L128
```

Debug results for wifi start (only on first call, nothing if recreating the task):
```
[D][WiFiGeneric.cpp:332] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:332] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:332] _eventCallback(): Event: 1 - SCAN_DONE
```
Wifi end gives
```
[D][WiFiGeneric.cpp:332] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:348] _eventCallback(): Reason: 8 - ASSOC_LEAVE
[D][WiFiGeneric.cpp:332] _eventCallback(): Event: 3 - STA_STOP
```

When ending wifi task again because of timeout, I get:
`E (505690) wifi: esp_wifi_disconnect 1153 wifi not start`

Any Idea what is happening here and how to restart wifi again?

maribeiro, 2020-03-06:
There seems to be a bug on the WifiGeneric class
See #3522

closed 2020-05-19 by stale[bot]

NOUSHADO, 2021-03-26:
Dear  i am having facing the issue of esp32 Nod mcu going offline i have tried many  but , it was suitable your code 
i was used in expalexa  program exactly working fine  
but i am not able to add this code in blynk esp32 program code it was getting error
for this can you suggest me how to use it
any way thanking you very much

NOUSHADO, 2021-03-26:
this is the error

Arduino: 1.8.13 (Windows Store 1.8.42.0) (Windows 10), Board: "WeMos WiFi&Bluetooth Battery, 80MHz, 921600, None"

BlynkESP32googleamazone26-03-2021-1:40:1: error: expected initializer before 'WiFi'

 WiFi.mode(WIFI_MODE_STA); // calls esp_wifi_set_mode(WIFI_MODE_STA); and esp_wifi_start();

 ^

BlynkESP32googleamazone26-03-2021-1:41:1: error: 'WiFi' does not name a type

 WiFi.enableSTA(true);

 ^

BlynkESP32googleamazone26-03-2021-1:42:1: error: 'scanResult' does not name a type

 scanResult = WiFi.scanNetworks(); // doing scan

 ^

BlynkESP32googleamazone26-03-2021-1:43:1: error: expected unqualified-id before 'if'

 if(scanResult > 0){

 ^

Multiple libraries were found for "WiFi.h"

 Used: C:\Users\noush\OneDrive\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.5\libraries\WiFi

 Not used: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\WiFi

exit status 1

expected initializer before 'WiFi'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

YordanYanakiev, 2021-06-29:
static bool wifiLowLevelDeinit(){
    //deinit not working yet!
    //esp_wifi_deinit();
    return true;
}
found this inside the WiFiGeneric.cpp .. (line 148)

so.. the deinit is not going to do anything...