espressif/arduino-esp32 #3795: Interrupt speed

yo90bosses, 2020-03-06:
I'm working on a self balancing robot and it for feedback it's using encoders on the motors. I'm concerned that the encoders might be too fast for the ESP32. The motors will ran at max 15000rpm and the encoders have a resolution of 500 per rotation so I'm expecting around max 250000 interrupts per motor going to the ESP32. Since I'm also going to be sending telemetry data to my laptop and reading sensors and doing calculations, will the ESP32 be fast enough for this?
I could use a different encoder with a lower resolution of 30 per rotation but the extra resolution would be nice for my calculations.

jjbubka, 2020-03-06:
I think it is better to use the pulse counter.
https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/pcnt.html
There are some examples if you google for it.

yo90bosses, 2020-03-07:
@jjbubka as far as I understand the pulse counter, I wouldn't be able to tell the direction of rotation if I'm only counting pulses.

bertmelis, 2020-03-07:
What do you want to do during the interrupt? And what do you mean with a self balancing robot? Sort of this Boston Dynamics type?

yo90bosses, 2020-03-07:
@bertmelis not much. I just need to keep track of how many interrupts have past and what direction the encoder is turning. So basically a few digitalread calls, if statements and additions. The main calculations for speed and position can be done in the loop function.
It's one of these robots https://youtu.be/I6z26LVu5y0 but im going to add a raspberry Pi and camera for tracking of objects and position.
The ESP32 is supposed use IMU and encoder data to roughly calculate it's position from where it has driven. That will be done in the loop function not in interrupts. 
I could use one core only for encoders and another for the control if it's too much to run everything on one core.

jjbubka, 2020-03-07:
@yo90bosses, are you sure?
Take a look using pcnt in quadrature mode:
https://www.esp32.com/viewtopic.php?t=4983
BR

closed 2020-05-20 by stale[bot]